crankyoldgit/IRremoteESP8266 #522: Raw code transformed to base 32 or base  64

jimmys01, 2018-09-17:
So, in order to transfer RAW signal via MQTT for example an ESPeasy contributor has transformed the RAW data to base32 string, then it can be transmitted back via MQTT (since it has got a lot shorter). I was hopping that this functionality could be implemented directly to the library. The decoding part is already present to https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P035_IRTX.ino
The decoding part has to be transferred from the google excel sheet to Arduino code… I haven’t fount any way to do binary to base32, especially for such long codes.
Example 
uint16_t rawData[59] = {3220, 9830, 550, 1536, 556, 512, 530, 486, 556, 502, 532, 1514, 556, 512, 532, 486, 556, 504, 532, 486, 554, 512, 532, 502, 532, 504, 530, 504, 530, 1524, 554, 504, 530, 510, 532, 1528, 558, 494, 556, 1530, 556, 504, 532, 502, 532, 1554, 532, 488, 554, 504, 530, 486, 556, 512, 530, 1530, 556, 510, 532};
Gets Transformed to
 FVS000000030CPJ0CPJ6CPJ0CPG6C1J60PJ6C1J 
with pulse and blank periods of around 250.. To send this via the ESPeasy the command is IRSEND,RAW,FVS000000030CPJ0CPJ6CPJ0CPG6C1J60PJ6C1J,38,265,243

To encode the RAW code to base32 i use this google doc I have modified to handle HVAC codes as well. https://docs.google.com/spreadsheets/d/16icZJWFbHq2iIxuEYxHzKWoP4ZEtra_iegIqDCMVvG4/edit?usp=sharing

Thanks for the excelent lib by the way

crankyoldgit, 2018-09-17:
Note: This **really** isn't an issue for IRremoteESP8266 per se.

Potentially I could write a routine to add that to the output of IRrecvDumpV2, however, it's not exactly something the library uses itself. That extra output could potentially confuse other users, or maybe disable that output by default.

Perhaps a standalone python program that takes the `rawData` line into a [base32](https://en.wikipedia.org/wiki/Base32#base32hex) line would be best, but is that _better_ than the current Google Sheets solution you already have?

Alternatively, you could add/request something gets added to https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P016_IR.ino to dump base32 there as it makes logical sense given (as far as I know, this is a format only used in EasyESP etc)

FWIW, the algorithm to use is described [here](http://www.ietf.org/rfc/rfc3548.txt) and there exists [some](http://ab-initio.mit.edu/octave-Faddeeva/gnulib/lib/base32.c) C implementations to do similar to what you want.

jimmys01, 2018-09-18:
The HEX code that the IRrecvDumpV2 gives back for the RAW code cannot be used to be transmited like the NEC code can, i believe that a base32 or base64 implementation would be very usfull.

crankyoldgit, 2018-09-18:
Correct. The "Hex" code for an `UNKNOWN` is [useless for sending.](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code)

If the desire is to shorten the message to fit inside a "small" MQTT message, then _base64_ would be a better mechanism compared to base32. However, base64 won't solve your problem w.r.t. EasyESP as-is.

crankyoldgit, 2018-09-18:
I wrote some python code to handle converting rawData lines to base-N encoding. Turns out that the "base-32" encoding is not just a conversion of the uint16_t array to a base-32 string.
e.g. in `RSEND,RAW,FVS000000030CPJ0CPJ6CPJ0CPG6C1J60PJ6C1J,38,265,243`, it assumes the base-32 string is then multiplied by` 265` & `243` for pulses & spaces. The encoding makes some assumption to compress the values by assuming the values are clean multiples of some common factor.
While this is probably a valid assumption in most cases, it inherently is loss-y.

This encoding is near akin to Pronto codes in some ways *and* using a run-length encoding to indicate the number of pulses/spaces(blanks) to send.

Thus, this "encoding" is not really base-32, it's a base-32 encoding of a compression-of-sorts. i.e. Where there are '1's in the base-2 representation after the base-32 string is decoded, it sends a pulse N times for each '1'. Similarly for each run of '0's etc.

While the encoding/compression is neat and clever, it really is specific only to EasyESP, and it loses some signal fidelity, just to save space in a MQTT message. As such, I don't think this is a problem for the IRremoteESP8266 library to solve. This is clearly a problem outside of this library's scope.