crankyoldgit/IRremoteESP8266 #525: Decoding Panasonic A/C problem

pandemixxx, 2018-09-25:
Hi, I'm currently trying to sniff Panasonic AC signal with IRrecvDumpV2 and here's what I'm getting

`Timestamp : 000099.145
Encoding  : UNKNOWN
Code      : 40B6FE46 (220 bits)
Not Mitsu AC.
Library   : v2.5.0

Raw Timing[439]:
   +  3582, -  1686,    +   488, -   378,    +   488, -  1238,    +   488, -   378, 
   +   488, -   378,    +   488, -   378,    +   488, -   378,    +   488, -   378, 
   +   488, -   384,    +   488, -   378,    +   488, -   378,    +   488, -   378, 
   +   488, -   378,    +   488, -   378,    +   488, -  1242,    +   486, -   378, 
   +   488, -   384,    +   488, -   378,    +   488, -   378,    +   488, -   380, 
   +   486, -   382,    +   484, -   382,    +   484, -  1264,    +   464, -  1266, 
   +   460, -  1272,    +   462, -   378,    +   488, -   406,    +   460, -  1266, 
   +   462, -   380,    +   488, -   382,    +   484, -   388,    +   478, -   406, 
   +   462, -   410,    +   462, -   404,    +   462, -   406,    +   462, -   396, 
   +   470, -   406,    +   462, -   404,    +   462, -   406,    +   460, -   404, 
   +   462, -   410,    +   462, -   404,    +   462, -   404,    +   462, -   406, 
   +   464, -   406,    +   462, -   404,    +   462, -   406,    +   462, -   404, 
   +   462, -   410,    +   462, -   404,    +   462, -   406,    +   462, -   404, 
   +   462, -   404,    +   462, -   404,    +   462, -   406,    +   460, -   406, 
   +   462, -   410,    +   462, -   404,    +   462, -  1264,    +   484, -  1244, 
   +   486, -   382,    +   482, -   382,    +   486, -   382,    +   486, -   378, 
   +   486, -   382,    +   488, -  9924,    +  3554, -  1686,    +   488, -   378, 
   +   490, -  1240,    +   486, -   378,    +   488, -   378,    +   488, -   378, 
   +   488, -   378,    +   488, -   382,    +   484, -   386,    +   486, -   378, 
   +   488, -   382,    +   486, -   378,    +   488, -   382,    +   486, -   382, 
   +   484, -  1242,    +   486, -   380,    +   488, -   386,    +   484, -   382, 
   +   486, -   380,    +   486, -   382,    +   486, -   380,    +   486, -   380, 
   +   486, -  1242,    +   486, -  1242,    +   484, -  1248,    +   484, -   380, 
   +   488, -   382,    +   484, -  1242,    +   486, -   382,    +   484, -   382, 
   +   484, -   382,    +   484, -   382,    +   486, -   386,    +   484, -   382, 
   +   486, -   382,    +   484, -   382,    +   486, -   382,    +   486, -   380, 
   +   484, -   382,    +   486, -   382,    +   488, -   380,    +   486, -   382, 
   +   484, -   380,    +   462, -   406,    +   488, -   376,    +   484, -  1246, 
   +   482, -  1246,    +   460, -   404,    +   480, -   392,    +   484, -   386, 
   +   482, -  1244,    +   484, -   382,    +   484, -   382,    +   484, -  1242, 
   +   482, -  1244,    +   484, -   382,    +   464, -   410,    +   460, -   404, 
   +   462, -   406,    +   462, -   404,    +   462, -   404,    +   470, -   396, 
   +   462, -   406,    +   462, -   404,    +   462, -  1286,    +   460, -  1268, 
   +   458, -  1268,    +   460, -  1266,    +   460, -  1266,    +   460, -   406, 
   +   460, -  1266,    +   462, -   406,    +   460, -  1272,    +   462, -   406, 
   +   460, -   406,    +   460, -   406,    +   460, -   406,    +   462, -   404, 
   +   462, -   406,    +   460, -   406,    +   462, -   410,    +   462, -   404, 
   +   462, -   406,    +   460, -   406,    +   460, -   406,    +   462, -   404, 
   +   462, -   406,    +   460, -   406,    +   460, -   410,    +   462, -   406, 
   +   460, -  1268,    +   460, -  1266,    +   460, -   404,    +   460, -   406, 
   +   462, -   406,    +   460, -   406,    +   460, -   412,    +   456, -   410, 
   +   460, -   410,    +   438, -   428,    +   460, -   410,    +   456, -   410, 
   +   456, -  1272,    +   436, -  1288,    +   438, -   434,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   428,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   434,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   428,    +   438, -   428, 
   +   440, -   428,    +   438, -   428,    +   438, -   432,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   428,    +   438, -   428, 
   +   438, -   428,    +   438, -   430,    +   438, -  1294,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   428,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   434,    +   438, -   428, 
   +   438, -  1288,    +   438, -  1290,    +   438, -   428,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -   432,    +   438, -  1288, 
   +   438, -  1290,    +   438, -   430,    +   438, -   428,    +   438, -   428, 
   +   438, -   428,    +   438, -   428,    +   438, -  1292,    +   438

uint16_t rawData[439] = {3582, 1686,  488, 378,  488, 1238,  488, 378,  488, 378,  488, 378,  488, 378,  488, 378,  488, 384,  488, 378,  488, 378,  488, 378,  488, 378,  488, 378,  488, 1242,  486, 378,  488, 384,  488, 378,  488, 378,  488, 380,  486, 382,  484, 382,  484, 1264,  464, 1266,  460, 1272,  462, 378,  488, 406,  460, 1266,  462, 380,  488, 382,  484, 388,  478, 406,  462, 410,  462, 404,  462, 406,  462, 396,  470, 406,  462, 404,  462, 406,  460, 404,  462, 410,  462, 404,  462, 404,  462, 406,  464, 406,  462, 404,  462, 406,  462, 404,  462, 410,  462, 404,  462, 406,  462, 404,  462, 404,  462, 404,  462, 406,  460, 406,  462, 410,  462, 404,  462, 1264,  484, 1244,  486, 382,  482, 382,  486, 382,  486, 378,  486, 382,  488, 9924,  3554, 1686,  488, 378,  490, 1240,  486, 378,  488, 378,  488, 378,  488, 378,  488, 382,  484, 386,  486, 378,  488, 382,  486, 378,  488, 382,  486, 382,  484, 1242,  486, 380,  488, 386,  484, 382,  486, 380,  486, 382,  486, 380,  486, 380,  486, 1242,  486, 1242,  484, 1248,  484, 380,  488, 382,  484, 1242,  486, 382,  484, 382,  484, 382,  484, 382,  486, 386,  484, 382,  486, 382,  484, 382,  486, 382,  486, 380,  484, 382,  486, 382,  488, 380,  486, 382,  484, 380,  462, 406,  488, 376,  484, 1246,  482, 1246,  460, 404,  480, 392,  484, 386,  482, 1244,  484, 382,  484, 382,  484, 1242,  482, 1244,  484, 382,  464, 410,  460, 404,  462, 406,  462, 404,  462, 404,  470, 396,  462, 406,  462, 404,  462, 1286,  460, 1268,  458, 1268,  460, 1266,  460, 1266,  460, 406,  460, 1266,  462, 406,  460, 1272,  462, 406,  460, 406,  460, 406,  460, 406,  462, 404,  462, 406,  460, 406,  462, 410,  462, 404,  462, 406,  460, 406,  460, 406,  462, 404,  462, 406,  460, 406,  460, 410,  462, 406,  460, 1268,  460, 1266,  460, 404,  460, 406,  462, 406,  460, 406,  460, 412,  456, 410,  460, 410,  438, 428,  460, 410,  456, 410,  456, 1272,  436, 1288,  438, 434,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 434,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  440, 428,  438, 428,  438, 432,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 430,  438, 1294,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 434,  438, 428,  438, 1288,  438, 1290,  438, 428,  438, 428,  438, 428,  438, 428,  438, 432,  438, 1288,  438, 1290,  438, 430,  438, 428,  438, 428,  438, 428,  438, 428,  438, 1292,  438};  // UNKNOWN 40B6FE46

`

I'm not sure if Panasonic library included and how can I use this data, is there anything different should be done with Panasonic IR signal?

crankyoldgit, 2018-09-25:
In short, the library (as of 2.5.0) **does not** support Panasonic A/C messages. i.e. No special send/decode functions for that protocol you captured. You will have to use the `sendRaw()` method until it does. (see the example code for how)

Longer answer:
I ran the `rawData` you supplied through the analyse script. The results are below.
It looks like it has two sections of data. 8 bytes + 19 bytes (total of 27 bytes/216 bits)
Some of those bytes may be redundant, as `0x4004` is a standard preamble for the Panasonic protocol. I'll need some more data (samples) to be sure.
In theory, it should be fairly easy to add basic decoding and sending support, but if you want detailed decoding and construction of messages (i.e. setting temps/modes/etc), **you** will need to do a lot of leg work (analysis) after the basic stuff is done.

```
IRremoteESP8266/tools$ ./auto_analyse_raw_data.py -f /tmp/issue_525 
Found 439 timing entries.
Potential Mark Candidates:
[3582, 490]
Potential Space Candidates:
[9924, 1686, 1294, 434]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 3568
kHdrSpace  = 1686
kBitMark   = 465
kOneSpace  = 1262
kZeroSpace = 401
kSpaceGap = 9924

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+0100000000000100000001110010000000000000000000000000000001100000GAP(9924)
  Bits: 64
  Hex:  0x4004072000000060 (MSB first)
        0x0600000004E02002 (LSB first)
  Dec:  4612819752354578528 (MSB first)
        432345564309364738 (LSB first)
  Bin:  0b0100000000000100000001110010000000000000000000000000000001100000 (MSB first)
        0b0000011000000000000000000000000000000100111000000010000000000010 (LSB first)
kHdrMark+kHdrSpace+01000000000001000000011100100000000000000000110001001100000000011111010100000000000000000110000000000110000000000000000000000001000000000110000011000001
  Bits: 152
  Hex:  0x40040720000C4C01F5000060060000010060C1 (MSB first)
        0x83060080000060060000AF8032300004E02002 (LSB first)
  Dec:  1427598566361800644592691388272474788781383873 (MSB first)
        2921920464864293261250805234523445123210027010 (LSB first)
  Bin:  0b01000000000001000000011100100000000000000000110001001100000000011111010100000000000000000110000000000110000000000000000000000001000000000110000011000001 (MSB first)
        0b10000011000001100000000010000000000000000000000001100000000001100000000000000000101011111000000000110010001100000000000000000100111000000010000000000010 (LSB first)

Total Nr. of suspected bits: 216
```

pandemixxx, 2018-09-25:
Thank you for quick response. I'll handle it with raw data then to make it work faster.

But if any further testing or additional data needed for adding this protocol, will be happy to contribute.

crankyoldgit, 2018-09-26:
Cool. For starters. Some full descriptions of what is on the remote (i.e. it's state: Power on, cool mode, temp 25, fan auto, sleep off, ...). Several of those., plus some power on and off signals.
Sometimes the power on/off/fan swing etc are different structures, so a few of those will confirm if the single example you provided is typical or not.

huexpub, 2018-09-27:
maybe Transformation from https://github.com/ToniA/ESPEasy/tree/HeatpumpIR/lib/HeatpumpIR for  panasonic ac?

crankyoldgit, 2018-09-28:
Nice find. It looks like it might be the same protocol that you supplied. i.e. Correct structure of 8 + 19 = 27 bytes Have you confirmed if that HeatpumpIR software works for your device?

crankyoldgit, 2018-09-29:
@pandemixxx Can you please test the [panasonic_ac](https://github.com/markszabo/IRremoteESP8266/tree/panasonic_ac) branch? It should work as it seems to decode the rawData you provided.

Hat tip to @huexpub for the suggestion of code to look at. That helped immensely.

pandemixxx, 2018-09-29:
Wow guys, this is actually beyond amazing
Power on:
```
Timestamp : 000013.108
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E00400402A80310000066000008000060D (216 bits)
Mesg Desc.: Model: 4 (JKE), Power: Off, Mode: 4 (HEAT), Temp: 21C, Fan: 0 (MIN), Swing (Vertical): 1 (Full Up), Quiet: Off, Powerful: Off
Library   : v2.5.0

Raw Timing[439]:
   +  3576, -  1710,    +   464, -   402,    +   464, -  1262,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   408,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -  1262,    +   464, -   402, 
   +   464, -   406,    +   464, -   402,    +   466, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -  1264,    +   462, -  1262, 
   +   464, -  1268,    +   462, -   404,    +   462, -   402,    +   464, -  1264, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   406,    +   464, -   402,    +   466, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   406,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   466, -   406,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   408,    +   464, -   402,    +   464, -  1266,    +   462, -  1262, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   406,    +   462, -  9946,    +  3528, -  1710,    +   464, -   402, 
   +   464, -  1264,    +   464, -   404,    +   462, -   402,    +   464, -   402, 
   +   466, -   402,    +   464, -   402,    +   464, -   408,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   404, 
   +   462, -  1264,    +   464, -   402,    +   464, -   406,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -  1266,    +   460, -  1264,    +   464, -  1268,    +   464, -   402, 
   +   464, -   402,    +   464, -  1262,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   408,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   408,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   404,    +   460, -  1266,    +   462, -   408,    +   464, -   402, 
   +   464, -  1264,    +   464, -   402,    +   464, -  1264,    +   464, -   402, 
   +   464, -  1264,    +   464, -   402,    +   464, -   406,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   404,    +   464, -  1284,    +   460, -  1264, 
   +   462, -   404,    +   464, -   402,    +   464, -   402,    +   464, -  1266, 
   +   460, -  1264,    +   464, -   402,    +   464, -   406,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   404,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   408,    +   462, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   406,    +   464, -   404, 
   +   464, -  1264,    +   462, -  1264,    +   464, -   402,    +   464, -   404, 
   +   464, -   402,    +   464, -   402,    +   464, -   410,    +   462, -   402, 
   +   464, -   404,    +   462, -   402,    +   464, -   404,    +   464, -   402, 
   +   464, -  1266,    +   460, -  1264,    +   462, -   408,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   406,    +   460, -   406, 
   +   460, -   402,    +   464, -   402,    +   464, -   412,    +   460, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   402,    +   464, -   404, 
   +   462, -   402,    +   464, -   402,    +   464, -   410,    +   460, -   408, 
   +   460, -   406,    +   460, -   406,    +   462, -   402,    +   464, -   406, 
   +   460, -   406,    +   460, -   402,    +   464, -  1270,    +   462, -   402, 
   +   464, -   402,    +   464, -   404,    +   462, -   402,    +   464, -   404, 
   +   464, -   402,    +   464, -   404,    +   462, -   410,    +   462, -   402, 
   +   464, -  1264,    +   462, -  1264,    +   462, -   402,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   410,    +   460, -  1266, 
   +   462, -   402,    +   464, -  1266,    +   458, -  1266,    +   464, -   402, 
   +   464, -   402,    +   464, -   402,    +   464, -   406,    +   462

uint16_t rawData[439] = {3576, 1710,  464, 402,  464, 1262,  464, 402,  464, 402,  464, 402,  464, 402,  464, 402,  464, 408,  464, 402,  464, 402,  464, 402,  464, 402,  464, 402,  464, 1262,  464, 402,  464, 406,  464, 402,  466, 402,  464, 402,  464, 402,  464, 402,  464, 1264,  462, 1262,  464, 1268,  462, 404,  462, 402,  464, 1264,  464, 402,  464, 402,  464, 402,  464, 402,  464, 406,  464, 402,  466, 402,  464, 402,  464, 402,  464, 402,  464, 402,  464, 402,  464,  [... cut ...] 8,  488, 378,  488, 380,  486, 384,  488, 378,  488, 1264,  462, 1242,  484, 380,  486, 380,  488, 380,  486, 380,  486, 410,  460, 380,  488, 380,  486, 1242,  486, 380,  486, 1242,  486, 1266,  462, 1240,  486, 382,  488};  // PANASONIC_AC
uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x32, 0x2E, 0x80, 0xA2, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x80, 0x00, 0x06, 0x74};
```

If any further testing needed, I'm here to assist.
Thank You so much!

pandemixxx, 2018-09-29:
@crankyoldgit Yes, Didn't have access to the unit, now I do.

AC model: CS-YW9MKD
Remote model: A75C3747

Power On:
```
Timestamp : 000017.515
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E00400412C803100000660000080000610 (216 bits)
Mesg Desc.: Model: 4 (JKE), Power: On, Mode: 4 (HEAT), Temp: 22C, Fan: 0 (MIN), Swing (Vertical): 1 (Full Up), Quiet: Off, Powerful: Off
Library   : v2.5.0

Raw Timing[439]:
   +  3584, -  1684,    +   486, -   380,    +   488, -  1240,    +   488, -   376, 
   +   490, -   376,    +   488, -   380,    +   484, -   384,    +   486, -   380, 
   +   472, -   398,    +   490, -   402,    +   464, -   380,    +   464, -   402, 
   +   486, -   380,    +   464, -   426,    +   442, -  1262,    +   464, -   400, 
   +   468, -   430,    +   440, -   404,    +   464, -   402,    +   464, -   426, 
   +   440, -   402,    +   464, -   426,    +   442, -  1284,    +   442, -  1286, 
   +   440, -  1292,    +   440, -   398,    +   468, -   398,    +   466, -  1286, 
   +   442, -   426,    +   440, -   426,    +   460, -   404,    +   442, -   404, 
   +   464, -   430,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   404,    +   464, -   426,    +   440, -   426,    +   440, -   426, 
   +   442, -   430,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   442, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   422, 
   +   446, -   430,    +   442, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   442, -   430,    +   442, -   426,    +   440, -  1286,    +   442, -  1286, 
   +   440, -   426,    +   440, -   426,    +   442, -   424,    +   440, -   426, 
   +   440, -   426,    +   442, -  9970,    +  3506, -  1730,    +   440, -   426, 
   +   440, -  1286,    +   442, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   440, -   426, 
   +   440, -   426,    +   442, -   426,    +   440, -   426,    +   442, -   426, 
   +   440, -  1286,    +   442, -   426,    +   440, -   430,    +   442, -   426, 
   +   440, -   426,    +   440, -   426,    +   442, -   426,    +   440, -   426, 
   +   442, -  1284,    +   442, -  1286,    +   440, -  1290,    +   440, -   426, 
   +   440, -   426,    +   440, -  1286,    +   442, -   426,    +   440, -   426, 
   +   440, -   426,    +   442, -   426,    +   440, -   430,    +   442, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   442, -  1286, 
   +   442, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -  1286,    +   442, -   430,    +   440, -   426, 
   +   440, -   426,    +   440, -  1286,    +   440, -  1286,    +   440, -   426, 
   +   440, -  1286,    +   440, -   426,    +   440, -   430,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   442, -   426,    +   440, -   426, 
   +   442, -   426,    +   440, -   426,    +   440, -  1308,    +   438, -  1286, 
   +   440, -   426,    +   440, -   426,    +   440, -   426,    +   440, -  1284, 
   +   442, -  1288,    +   438, -   426,    +   440, -   432,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   442, -   426,    +   440, -   430,    +   440, -   426, 
   +   440, -   426,    +   442, -   426,    +   440, -   426,    +   440, -   426, 
   +   442, -   426,    +   442, -   424,    +   440, -   430,    +   442, -   426, 
   +   440, -  1288,    +   436, -  1286,    +   442, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   442, -   426, 
   +   440, -   426,    +   442, -   424,    +   442, -   426,    +   440, -   426, 
   +   440, -  1286,    +   442, -  1286,    +   440, -   432,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   442, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   442, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   426,    +   442, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   442, -   426, 
   +   440, -   426,    +   440, -   426,    +   442, -   426,    +   440, -   426, 
   +   442, -   426,    +   440, -   424,    +   442, -  1290,    +   442, -   426, 
   +   438, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   440, -   426, 
   +   442, -  1286,    +   440, -  1286,    +   440, -   426,    +   440, -   426, 
   +   440, -   426,    +   442, -   426,    +   440, -   430,    +   442, -   426, 
   +   440, -   426,    +   440, -   426,    +   442, -   426,    +   440, -  1286, 
   +   440, -   426,    +   440, -   426,    +   440, -   430,    +   438

[... 48 lines cut ...]
   +   486, -   404,    +   462, -   380,    +   486, -   408,    +   464, -   402, 
   +   464, -   380,    +   486, -   380,    +   486, -   380,    +   486, -   380, 
   +   486, -   380,    +   486, -   380,    +   486, -   408,    +   462, -   380, 
   +   486, -  1264,    +   464, -  1264,    +   464, -   376,    +   488, -   378, 
[... cut ...] , 380,  486, 402,  464, 402,  464, 408,  464, 404,  464, 1262,  464, 1264,  462, 380,  486, 404,  462, 404,  464, 380,  486, 408,  464, 1264,  464, 1264,  462, 1264,  462, 1240,  486, 382,  484, 404,  464, 402,  464, 380,  488};  // PANASONIC_AC
uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x40, 0x2C, 0x80, 0x31, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x80, 0x00, 0x06, 0x0F};
```

btw question, do I send **uint8_t state[27]** with sendPanasonic?

crankyoldgit, 2018-09-29:
Thanks for the extra data. Odd. All the earlier ones were marked as "Power: Off". These two look fine. DId you just capture them (previous ones) all with the remote in the "Off" setting? Anyway, those two seem to match your descriptions, so .. _shrug_ .. nothing to fix there.

> btw question, do I send uint8_t state[27] with sendPanasonic?

No. Use `sendPanasonicAC()` instead.
e.g. For the last message you gave:
```
uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x40, 0x2C, 0x80, 0x31, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x80, 0x00, 0x06, 0x0F};
irsend.sendPanasonicAC(state);
```

P.S. I've just added another commit to that branch, so do a `git pull`on that branch and it should include some subtle improvements.

pandemixxx, 2018-09-29:
Yes, previously I sniffed signals from remote without AC. Now I did it in front of AC, so it went "On". If I need to do more in front of AC, just tell ;)

crankyoldgit, 2018-09-30:
If you could test `sendPanasonicAC()` and confirm the AC unit acts accordingly, that'd be great.

MikkelTb, 2018-09-30:
Hi.
I'm welly interesting in this, and so happy that somebody has decode the remote control.

I have a Panasonic Multi-split aircon.
I also have the capability to test "Sleep Timer" "Timer on" and "Timer off".

And I can test it on my Aircon.

crankyoldgit, 2018-09-30:
Hi @MikkelTb Feel free to try and decode what bits/bytes of the `state[]` are used for the timer functions.

I suggest you create a google sheets spreadsheet to track what data you collect and so others (e.g. me) can see the latest data you've worked out.
Basically, you need to keep the other remote settings the same, and only modify the sleep/timer functions to see what "changes" in the `state[]` variable.
Once you've tracked down what changes for what functions, then it's try to work out what the data means.
Typically for timers, there are number of encoding for the time. Some systems just use a simple counter for "minutes since midnight", some count "how many 15 min periods there have been since midnight",
some store the "hours since midnight" in one byte, and "minutes since the start of the hour" in another.

Basically, record what's on the remote, try to change as little as possible per sample. i.e. start only on one function, and try to find the difference between one minute and the next minute in sequence, or the next hour in the sequence.

Hopefully the pattern will emerge. When you've got some `state[]` data and idea of the format, I can try to write the code to decode it and recreate it.

Oh, an important fact is also to let us know what model number it is. It could be something we haven't seen yet.

pandemixxx, 2018-09-30:
> If you could test `sendPanasonicAC()` and confirm the AC unit acts accordingly, that'd be great.

Yap, going to test it, I'll report all tested messages a little bit later today

pandemixxx, 2018-09-30:
Just tested 2 commands I posted before - Power On and Power Off

`uint8_t stateon[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x41, 0x2C, 0x80, 0x31, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x80, 0x00, 0x06, 0x10};`

`uint8_t stateoff[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x40, 0x2C, 0x80, 0x31, 0x00, 0x00, 0x06, 0x60, 0x00, 0x00, 0x80, 0x00, 0x06, 0x0F};`

Both sent with irsend.sendPanasonicAC(state) - everything seems OK, works well!

If any further testing needed, I'm here for you.

pandemixxx, 2018-09-30:
I have a question, or maybe there's some information here I couldn't find, please link if so.
Is it possible to separate on/off command from fan power, mode or temperature? Just like if I got 23 on my AC and I just want to send a signal to change mode or fan position without changing temperature data or it don't work like that and every signal I send should contain all info?

MikkelTb, 2018-09-30:
[AC Remote Info.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2431935/AC.Remote.Info.xlsx)

I have collected the follow information.
For the 1h/2h/3h..7h Sleep timer, it look like, it send a packet, there is not like the On/Off and temperatur.
Mine remote control, the On/Off is a Toggel, and a single button.

For The on and Off settings, i can not figure out, how the value for time is calculated in the byte string.

crankyoldgit, 2018-09-30:
@MikkelTb 
Yeah, the data looks weird.
For the "Sleep" ones. e.g. Ark6 (Sleep 1h) and Ark7 (Sleep 2h) there doesn't appear to be a difference in the code sent. Unless its a placebo in that there is not unit operation difference between those remote settings, then I don't know how that info is being transmitted.

For the timer data (Ark17 etc), it looks like On & Off timers are controlled by the 2nd and 3rd least significant bits of `state[13]` i.e. `(state[13] & 0b00000010)` Non-zero is the On Timer is active, 0 is it is On timer is off/cancelled. Ditto for the Off Timer, except use `0b00000100` instead.

What is the time setting resolution for those settings? i.e .Can you set the timer down to any exact minute, or is it only to the nearest 5 or 10 minute mark?

I also assume there is a Clock (current time) on the remote as well, we need to track down which bytes those are stored in and record the "time on the remote" when you capture the message.

General FYIs:
You don't need to keep/record the "Raw Timing[]:" +/- data.
You only need to keep the `uint16_t rawData[] = {...];` line **IF** it detects it as `UNKNOWN`.
The most important data from IRrecvDumpV2 for this project is just the `uint8_t state[] = {...};` line.

Oh, another [q], does your unit support Horizontal swing control via the remote?

crankyoldgit, 2018-09-30:
Looking at the time encoding, I'd suggest looking at time values of "00:00", "00:01" or "00:10" (if it does it in 10 mins steps), & "01:00", "01:10" etc. That may help making it easier to decode how the timer times are stored.
Ditto for the "clock" setting, assuming there is one on the remote.

crankyoldgit, 2018-09-30:
@pandemixxx Thanks for that. That's excellent news.

pandemixxx, 2018-10-01:
@crankyoldgit Thank you for shared info, saved a lot of time for me.

crankyoldgit, 2018-10-01:
@MikkelTb  Okay, so (24hr * 60min) / 10min = 144 units, which can easily fit in a single byte (<= 256). So the time(r)s can possibly reside in a single byte, maybe.

Odd that it doesn't transmit the remote's time with each message. That _may_ mean that the timer values are _relative_ to the time on the remote.
e.g. Remote's clock say "13:50" ... and if you set an OFF timer for 14:30. Then it may be saying _"Hey AC unit, please  set an OFF timer for in 40 minutes (4 units) time."_

Other than that, I'm out of ideas with out lots more data (inc. the clock time on the remote)

crankyoldgit, 2018-10-01:
@MikkelTb I also STRONGLY recommend using Google Sheets for the spreadsheet. It saves me having to download and convert the spreadsheet (I don't have Excel ;-) You can also allow people to comment on it too and it's always up to date in real-time. ;-)

crankyoldgit, 2018-10-02:
FYI, I've merged the pansonic branch into master. It will thus make it into the next release.

pandemixxx, 2018-10-02:
Thank you @crankyoldgit , that's great news, should I close this issue?

crankyoldgit, 2018-10-02:
FYI, v2.5.1 has been released and includes the work we've done thus far.

@pandemixxx Yeah, we should probably close it.
@MikkelTb Can please create a new issue if you get any progress on the timer stuff?

closed 2018-10-02 by crankyoldgit