crankyoldgit/IRremoteESP8266 #527: Decoding Electra A/C Problem

ShonP40, 2018-09-25:
Hi, I'm trying to read a Electra AC signal with IRrecvDumpV2 and here's what I'm getting: (My A/C remote is set to 24c 1 fan speed)

`22:43:50.848`` -> Encoding  : UNKNOWN
22:43:50.848 -> Code      : 9AD8CDB5 (106 bits)
22:43:50.848 -> Not Mitsu AC.
22:43:50.848 -> Library   : v2.5.0
22:43:50.848 -> 
Raw Timing[211]:
22:43:50.882 ->    +  9166, -  4470,    +   642, -  1632,    +   642, -  1632,    +   668, -   534, 
22:43:50.882 ->    +   666, -   534,    +   668, -   534,    +   614, -   536,    +   640, -  1636, 
22:43:50.882 ->    +   640, -  1646,    +   694, -  1662,    +   612, -  1628,    +   642, -  1666, 
22:43:50.882 ->    +   664, -   532,    +   668, -   534,    +   666, -   534,    +   666, -   532, 
22:43:50.882 ->    +   666, -  1644,    +   642, -   532,    +   640, -  1634,    +   668, -  1632, 
22:43:50.917 ->    +   642, -   538,    +   666, -  1660,    +   610, -  1666,    +   664, -  1632, 
22:43:50.917 ->    +   642, -  1672,    +   610, -   536,    +   666, -   534,    +   694, -   532, 
22:43:50.917 ->    +   666, -  1636,    +   614, -   538,    +   666, -  1632,    +   642, -   536, 
22:43:50.917 ->    +   666, -   544,    +   692, -   534,    +   640, -   558,    +   640, -   534, 
22:43:50.952 ->    +   640, -   540,    +   666, -   534,    +   638, -  1666,    +   638, -  1636, 
22:43:50.952 ->    +   640, -   550,    +   666, -   534,    +   640, -   540,    +   666, -   534, 
22:43:50.952 ->    +   640, -   540,    +   666, -   536,    +   638, -   540,    +   666, -   536, 
22:43:50.952 ->    +   638, -   550,    +   664, -   536,    +   638, -   540,    +   664, -   536, 
22:43:50.952 ->    +   638, -   540,    +   666, -   534,    +   638, -  1640,    +   664, -   536, 
22:43:50.987 ->    +   692, -   546,    +   664, -   536,    +   664, -   536,    +   664, -   536, 
22:43:50.987 ->    +   664, -   546,    +   612, -   532,    +   636, -   538,    +   664, -   536, 
22:43:50.987 ->    +   664, -   546,    +   612, -   538,    +   638, -   538,    +   638, -   538, 
22:43:50.987 ->    +   664, -   536,    +   690, -   538,    +   662, -   538,    +   664, -   538, 
22:43:50.987 ->    +   662, -   548,    +   664, -   536,    +   662, -   538,    +   662, -   562, 
22:43:51.022 ->    +   638, -   564,    +   636, -   564,    +   636, -  1668,    +   582, -   556, 
22:43:51.022 ->    +   652, -   572,    +   612, -   568,    +   636, -   564,    +   610, -   570, 
22:43:51.022 ->    +   636, -   556,    +   616, -   550,    +   656, -   566,    +   610, -   570, 
22:43:51.022 ->    +   632, -   578,    +   608, -  1640,    +   662, -   562,    +   642, -  1686, 
22:43:51.057 ->    +   582, -   570,    +   634, -   566,    +   604, -   576,    +   636, -   566, 
22:43:51.057 ->    +   610, -   578,    +   634, -  1664,    +   584, -   590,    +   660, -  1636, 
22:43:51.057 ->    +   610, -  1642,    +   664, -   590,    +   610, -   590,    +   636, -   566, 
22:43:51.057 ->    +   634, -   568,    +   686
22:43:51.057 -> 
22:43:51.057 -> uint16_t rawData[211] = {9166, 4470,  642, 1632,  642, 1632,  668, 534,  666, 534,  668, 534,  614, 536,  640, 1636,  640, 1646,  694, 1662,  612, 1628,  642, 1666,  664, 532,  668, 534,  666, 534,  666, 532,  666, 1644,  642, 532,  640, 1634,  668, 1632,  642, 538,  666, 1660,  610, 1666,  664, 1632,  642, 1672,  610, 536,  666, 534,  694, 532,  666, 1636,  614, 538,  666, 1632,  642, 536,  666, 544,  692, 534,  640, 558,  640, 534,  640, 540,  666, 534,  638, 1666,  638, 1636,  640, 550,  666, 534,  640, 540,  666, 534,  640, 540,  666, 536,  638, 540,  666, 536,  638, 550,  664, 536,  638, 540,  664, 536,  638, 540,  666, 534,  638, 1640,  664, 536,  692, 546,  664, 536,  664, 536,  664, 536,  664, 546,  612, 532,  636, 538,  664, 536,  664, 546,  612, 538,  638, 538,  638, 538,  664, 536,  690, 538,  662, 538,  664, 538,  662, 548,  664, 536,  662, 538,  662, 562,  638, 564,  636, 564,  636, 1668,  582, 556,  652, 572,  612, 568,  636, 564,  610, 570,  636, 556,  616, 550,  656, 566,  610, 570,  632, 578,  608, 1640,  662, 562,  642, 1686,  582, 570,  634, 566,  604, 576,  636, 566,  610, 578,  634, 1664,  584, 590,  660, 1636,  610, 1642,  664, 590,  610, 590,  636, 566,  634, 568,  686};  // UNKNOWN 9AD8CDB5`

crankyoldgit, 2018-09-26:
It is reported as `UNKNOWN` because it isn't yet supported.

Here is some basic analysis of the message you provided:
```
IRremoteESP8266/tools$ ./auto_analyse_raw_data.py -g -f /tmp/elec 
Found 211 timing entries.
Potential Mark Candidates:
[9166, 694]
Potential Space Candidates:
[4470, 1686, 590]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 9166
kHdrSpace  = 4470
kBitMark   = 646
kOneSpace  = 1647
kZeroSpace = 547

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11000011111000010110111100010100000001100000000000000100000000000000000000000100000000001010000010110000
  Bits: 104
  Hex:  0xC3E16F1406000400000400A0B0 (MSB first)
        0x0D050020000020006028F687C3 (LSB first)
  Dec:  15519260102815336942777674014896 (MSB first)
        1031513688850279573500600944579 (LSB first)
  Bin:  0b11000011111000010110111100010100000001100000000000000100000000000000000000000100000000001010000010110000 (MSB first)
        0b00001101000001010000000000100000000000000000000000100000000000000110000000101000111101101000011111000011 (LSB first)

Total Nr. of suspected bits: 104
```

Adding basic support for this protocol shouldn't be hard.

ShonP40, 2018-09-26:
Thank you very much.
I am going to try in like 1 hour when I get home.

ShonP40, 2018-09-26:
Can you update the IRrecvDumpV2 please?

crankyoldgit, 2018-09-26:
It shouldn't need to be updated if you are using the `electra_ac` branch: https://github.com/markszabo/IRremoteESP8266/tree/electra_ac

`IRrecvDumpV2` just needs to be recompiled against the "new" version of the library.

ShonP40, 2018-09-26:
I can't even test it, I've tried to use  `ir_Electra_test.cpp` but Arduino IDE failed to compile it.

crankyoldgit, 2018-09-26:
That's a unit test. Unless you know what you are doing, you are looking at the wrong code.
https://github.com/markszabo/IRremoteESP8266/wiki/Library-Maintainers-Guide#unit-tests

All you should need to do is download either the branch's latest zip file (e.g. https://github.com/markszabo/IRremoteESP8266/archive/electra_ac.zip) , or via:
```
# Assuming you have already 'git clone'd the repository per the main page.
cd <your_git_directory_location>/IRremoteESP8266
git fetch  # pull down all the infomation for the library
git checkout electra_ac  # Switch to the 'electra_ac' branch
git pull  # Ensures you have the latest version etc.
```

And compile the `IRrecvDumpV2.ino` example code as normal.

crankyoldgit, 2018-09-26:
For `IRsendDemo`:
```
state[13] = {0xC3, 0xE1, 0xD7, 0xF0, 0x06, 0x00, 0x04, 0x00, 0x00, 0x04, 0x00, 0xA0, 0x97};
irsend.sendElectraAC(state);
```

You can't really easily do it in the `IRserver` example code.

However, you should be able to just use the `IRMQTTServer` example code in three ways.
1) Manually via it's web page. Select the Electra AC protocol, and added in `C3E1D7F006000400000400A097`.
2) Via MQTT with a message to the `ir_server/send` topic of: `48,0xC3E1D7F006000400000400A097`
3) Via it the HTTP API. e.g. `http://<your_esp8266's_ip_address>/ir?type=48&code=C3E1D7F006000400000400A097`

That's all documented in the comments of the IRMQTTServer example.

ShonP40, 2018-09-26:
Is there a way to connect my ESP to my WiFi network with the `IRMQTTSERVER` example code?

crankyoldgit, 2018-09-26:
Yes. That's one of it's sole purposes.

Please read the documentation in the code. e.g. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L24

ShonP40, 2018-09-26:
Oh, thanks

ShonP40, 2018-09-26:
And do i need to host an MQTT server for it to send IR commands?

crankyoldgit, 2018-09-26:
No. Again. Please read the documentation in the code:
e.g. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L107

ShonP40, 2018-09-26:
Ok.
And about the `IRsendDemo` can you create a .ino file for me because i don't know where to put the `state[13] = {0xC3, 0xE1, 0xD7, 0xF0, 0x06, 0x00, 0x04, 0x00, 0x00, 0x04, 0x00, 0xA0, 0x97};
irsend.sendElectraAC(state);`

crankyoldgit, 2018-09-26:
https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

ShonP40, 2018-09-26:
Thank you very much for adding support for Electra A/C.

I can now control my A/C with my ESP.

And I’ve added the IP with the command to Siri shortcuts so I can just ask Siri to turn on or off the A/C.

closed 2018-09-26 by ShonP40

crankyoldgit, 2018-10-02:
FYI, this was included in the v2.5.1 release.