crankyoldgit/IRremoteESP8266 #531: having trouble compiling IRremoteESP8266/examples/IRMQTTServer : ELECTRA_AC was not declared in this scope

perfessor101, 2018-09-27:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._
V2.50

### Expected behavior
_What steps did you do and what should it have done?_

initialized a new project in platformio
downloaded the IRMQQTServer example
downloaded and modified the PubSubClient.h
added the WiFiManager Library
using static IP true, changed HOSTNAME, kIPaddress, KGateway and MQTT_Server addresses
IRMQTTServer hopefully compiles

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

initialized a new project in platformio
downloaded the IRMQQTServer example
downloaded and modified the PubSubClient.h
added the WiFiManager Library
using static IP true, changed HOSTNAME, kIPaddress, KGateway and MQTT_Server addresses
when compiling I'm getting the error "ELECTRA_AC was not declared in this scope"

not sure what other libraries to add or where

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

Not Yet Will head there now, sorry

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

IRMQTTServer has never worked for me before first time compiling it

### Other useful information
_More information is always welcome. Be verbose._

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <PubSubClient>
|-- <ESP8266WiFi> 1.0
|-- <WifiManager> 0.14
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <IRremoteESP8266> 2.5.0
Compiling .pioenvs/nodemcuv2/src/main.cpp.o
Compiling .pioenvs/nodemcuv2/liba8c/ESP8266WiFi/WiFiClientSecureAxTLS.cpp.o
Compiling .pioenvs/nodemcuv2/liba8c/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
Compiling .pioenvs/nodemcuv2/liba8c/ESP8266WiFi/WiFiServer.cpp.o
src/main.cpp: In function 'void parseStringAndSendAirCon(uint16_t, String)':
src/main.cpp:494:10: error: 'ELECTRA_AC' was not declared in this scope
case ELECTRA_AC:
^
src/main.cpp:495:19: error: 'kElectraAcStateLength' was not declared in this scope
stateSize = kElectraAcStateLength;
^
src/main.cpp: In function 'void handleIr()':
src/main.cpp:842:52: error: 'getUInt64fromHex' was not declared in this scope
data = getUInt64fromHex(server.arg(i).c_str());
^
src/main.cpp:851:60: error: 'sendIRCode' was not declared in this scope
sendIRCode(ir_type, data, data_str.c_str(), nbits, repeat);
^
src/main.cpp: In function 'void sendIRCode(int, uint64_t, const char*, uint16_t, uint16_t)':
src/main.cpp:1182:10: error: 'ELECTRA_AC' was not declared in this scope
case ELECTRA_AC:  // 48
^
*** [.pioenvs/nodemcuv2/src/main.cpp.o] Error 1
=================================== [ERROR] Took 20.14 seconds ===================================

crankyoldgit, 2018-09-27:
Hi,
 I think your issue maybe you are using a mix of files from different versions of the library.

The version of IRMQTTServer you have is from the `master` branch, and the library is probably from the `v2.5.0` release. I know this as Electra support was added to the `master` after v2.5.0 was released.

You solution is either `git pull` if you are tracking the master branch. or revert the `IRMQTTServer.ino` file back to what shipped with the v2.5.0 release.  e.g. https://github.com/markszabo/IRremoteESP8266/blob/v2.5.0/examples/IRMQTTServer/IRMQTTServer.ino

crankyoldgit, 2018-09-27:
Oh FYI,
Your steps:
> initialized a new project in platformio
downloaded the IRMQQTServer example
downloaded and modified the PubSubClient.h

If you `git clone` this library, there is a [platformio.ini](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/platformio.ini) file in the IRMQTTServer example directory. If you add that directory as a project, I believe it should work without doing most of those steps. This is what I do to build it.

That file tells PlatformIO to download the needed libraries and and make the needed compile-time changes for PubSub etc.
You *may* need to tweak that file to use 0.14 of the WifiManager library, as I encountered an issue compiling against the ESP Core 2_4_2 library. See #530 for details.

perfessor101, 2018-09-28:
I did a git clone and then imported the example into platformio from the "clone" and when trying to compile it I got an issue with a WPS definition ... then I upgraded the lib_deps_external line in platformio.ini from WifiManager@0.12 to WifiManager@0.14 like suggested and everything compiled nicely.  I may need to get a local copy of the lib PubSubClient to increase the MQTT_MAX_PACKET_SIZE to 512.

Copying the example files out of the "clone" in various ways was my main mistake.

Many thanks and much apologies,
Bobby

closed 2018-09-28 by perfessor101

crankyoldgit, 2018-09-28:
You won't need to change PubSubClient if you are using the supplied platformio.ini file.
It sets the packet size for you:
i.e. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/platformio.ini#L6