crankyoldgit/IRremoteESP8266 #532: IRMQTTServer - Fujitsu Codes seem to use wrong encoding.

kilakewe, 2018-09-28:
### Version/revison of the library used
v2.5.0

### Expected behavior
Put IRMQTTServer onto Esp
Use web interface to send Fujitsu AC code
Use 2nd arduino to read IR using IRrecvDumpV2
Send heat pump code and expect same code read back.

### Actual behavior
IR read back is encoded for DENON instead of Fujitsu.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
Timestamp : 002736.669
Encoding  : DENON
Code      : 2FD (14 bits)
Library   : v2.5.0

Raw Timing[63]:
   +   278, -   762,    +   276, -   772,    +   276, -   772,    +   280, -   768, 
   +   276, -   772,    +   276, -  1812,    +   276, -   772,    +   278, -  1812, 
   +   274, -  1812,    +   276, -  1810,    +   276, -  1814,    +   250, -  1836, 
   +   276, -  1812,    +   276, -   770,    +   278, -  1814,    +   278, - 43628, 
   +   276, -   772,    +   248, -   798,    +   276, -   772,    +   274, -   774, 
   +   274, -   776,    +   276, -   772,    +   276, -  1814,    +   250, -   798, 
   +   250, -   800,    +   278, -   772,    +   276, -   772,    +   250, -   798, 
   +   274, -   774,    +   248, -  1840,    +   276, -   772,    +   276

uint16_t rawData[63] = {278, 762,  276, 772,  276, 772,  280, 768,  276, 772,  276, 1812,  276, 772,  278, 1812,  274, 1812,  276, 1810,  276, 1814,  250, 1836,  276, 1812,  276, 770,  278, 1814,  278, 43628,  276, 772,  248, 798,  276, 772,  274, 774,  274, 776,  276, 772,  276, 1814,  250, 798,  250, 800,  278, 772,  276, 772,  250, 798,  274, 774,  248, 1840,  276, 772,  276};  // DENON 2FD
uint64_t data = 0x2FD;
```

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Circuit diagram and hardware used (if applicable)
Current recommended circuit 

### I have followed the steps in the [Troubleshooting Guide]
Yes

### Has this library/code previously worked as expected for you?
Yes, other types read as expected

crankyoldgit, 2018-09-28:
Can you please explain in as much detail as possible what exact steps you took to send the Fujitsu AC code?
e.g. Exactly what "code" you sent, and which HTML menu, or URL, or MQTT command use used to send it?

Also, the serial output from IRMQTTServer would be most helpful.

crankyoldgit, 2018-09-28:
Also, Can you include the "Information" section from the IRMQTTServer's page?

kilakewe, 2018-09-28:
Not too sure what's change be is now behaving as expected. Possibly a hardware issue.

closed 2018-09-28 by kilakewe

crankyoldgit, 2018-09-28:
Phew. Because I couldn't easily spot a problem in the code. ;-)