crankyoldgit/IRremoteESP8266 #533: Comparing received values with an enum

johannesmols, 2018-09-28:
### Version/revison of the library used
2.5.0

### Expected behavior

Comparing decoded results to enumeration and triggering events based on that.

### Actual behavior

Not recognizing values in enumeration.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
Timestamp : 000026.144
21:07:45.625 -> Encoding  : UNKNOWN
21:07:45.625 -> Code      : 4B50303E (56 bits)
21:07:45.625 -> Library   : v2.5.0
21:07:45.625 -> 
Raw Timing[111]:
21:07:45.658 ->    +   240, -   362,    +   240, -   366,    +   238, -   416,    +   240, -   470, 
21:07:45.658 ->    +   214, -   440,    +   214, -   416,    +   212, -   678,    +   238, -   390, 
21:07:45.658 ->    +   238, -   680,    +   214, -   494,    +   240, -   388,    +   240, -   456, 
21:07:45.658 ->    +   192, -  4972,    +   240, -   888,    +   238, -   866,    +   238, -   918, 
21:07:45.691 ->    +   238, -   890,    +   238, -   890,    +   240, -   892,    +   238, -   890, 
21:07:45.691 ->    +   238, -   864,    +   238, -  2020,    +   240, -  1996,    +   240, -  1994, 
21:07:45.691 ->    +   242, -  1992,    +   240, -  1994,    +   240, -  2022,    +   238, -  1968, 
21:07:45.691 ->    +   242, -  1994,    +   238, -   892,    +   238, -  1994,    +   240, -  1996, 
21:07:45.691 ->    +   238, -   888,    +   240, -  1998,    +   238, -   888,    +   238, -   894, 
21:07:45.724 ->    +   236, -   916,    +   238, -  1972,    +   238, -   888,    +   240, -   916, 
21:07:45.724 ->    +   240, -  1970,    +   238, -   888,    +   240, -  1998,    +   238, -  1992, 
21:07:45.724 ->    +   240, -  1996,    +   240, - 40130,    +   240, -   714,    +   176, -   890, 
21:07:45.724 ->    +   238, -   708,    +   214, -   746,    +   242, -   762,    +   238, -   758, 
21:07:45.768 ->    +   236, -   706,    +   236, -   706,    +   238, -  3152,    +   238
21:07:45.768 -> 
21:07:45.768 -> uint16_t rawData[111] = {240, 362,  240, 366,  238, 416,  240, 470,  214, 440,  214, 416,  212, 678,  238, 390,  238, 680,  214, 494,  240, 388,  240, 456,  192, 4972,  240, 888,  238, 866,  238, 918,  238, 890,  238, 890,  240, 892,  238, 890,  238, 864,  238, 2020,  240, 1996,  240, 1994,  242, 1992,  240, 1994,  240, 2022,  238, 1968,  242, 1994,  238, 892,  238, 1994,  240, 1996,  238, 888,  240, 1998,  238, 888,  238, 894,  236, 916,  238, 1972,  238, 888,  240, 916,  240, 1970,  238, 888,  240, 1998,  238, 1992,  240, 1996,  240, 40130,  240, 714,  176, 890,  238, 708,  214, 746,  242, 762,  238, 758,  236, 706,  236, 706,  238, 3152,  238};  // UNKNOWN 4B50303E
```

### Steps to reproduce the behavior
I am trying to receive values from my remote and process them using an _enum_ containing the values corresponding to the different buttons on the remote. I'm failing to properly compare the received values with my own as I can not figure out the datatype that is really being used in the `results`.

#### Example code used
```
enum IR_CODES : uint64_t
{ 
  POWER = 0xFFA25D,
  FUNC_STOP = 0xFFE21D,
  VOL_PLUS = 0xFF629D
};
```

In the function:

```
if (irrecv.decode(&results))
{
  switch (results.value)
  {
    case POWER:
      Serial.println("Power");
      break;
    case FUNC_STOP:
      Serial.println("Func Stop");
      break;
    case VOL_PLUS:
      Serial.println("Volume Plus");
      break;
    default:
      Serial.println("Unknown button");
      break;
  }
    
  irrecv.resume();  // Receive the next value
}
```

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Has this library/code previously worked as expected for you?
Yes, 2.5.0

### Other useful information
I have tried a few different integer types without success, and couldn't find the data type in the source code.

johannesmols, 2018-09-28:
Nevermind, it magically just started working now :) Sorry for the unnecessary issue!

closed 2018-09-28 by johannesmols

crankyoldgit, 2018-09-29:
For the record, `value` is a `uint64_t` per https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L90

Also, Just FYI: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code