crankyoldgit/IRremoteESP8266 #536: IR received to mqtt payload

savage872, 2018-10-01:
Love you project. I'm a newbie.

I have a nodemcu with a ir sender connected on D2 and ir receiver on d5. I've flashed my node mcu with a mix of IRMQTTServer and IRrecvDumpV2.
What I'm trying to do is publish to mqtt ir info received on d5 with the same sort of payload and with the codes that you send on D2.
Can someone help please?
I hope I've posted in the right section.

crankyoldgit, 2018-10-01:
Hi.
 Unfortunately, there isn't any example code in our library for turning received IR messages into MQTT messages. It wouldn't be hard to do, but it's not trivial or core to this library. There are plenty of projects out there that do this on the ESP8266.
For a starting point, check out our wiki page: https://github.com/markszabo/IRremoteESP8266/wiki/Related-User-Projects
That is just the tip of the iceberg. Some google searches will find plenty more.

Of course, if you are willing to write it, I'll gladly help it get merged into the library. PRs are always welcome. ;-)

savage872, 2018-10-02:
I had a look at the link and haven't found anything that would help me the matter. I could probably adapt the code if I found a working example, but writing code from scratch is not something that I can do easily.
I have tried installing https://github.com/enc-X/mqtt-ir-transceiver on my nodemcu but for some reason I couldn't get it up and running.

crankyoldgit, 2018-10-02:
I'm moderately sure that [ESPEasy](https://www.letscontrolit.com/wiki/index.php/ESPEasy) supports what you wants.
So does: https://github.com/1technophile/OpenMQTTGateway @1technophile can probably help you there. https://github.com/1technophile/OpenMQTTGateway/wiki/ESP8266-IR-Send-and-Receive
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-IR
And for some code to look at, try: https://github.com/1technophile/OpenMQTTGateway/blob/master/ZgatewayIR.ino#L92-L156

A google search lists plenty of things too: https://www.google.com.au/search?q=IR+mqtt+gateway

1technophile, 2018-10-02:
I confirm that OpenMQTTGateway support IR-->MQTT translation, don't hesitate to ask questions to [OMG forum](https://community.openmqttgateway.com/) if needed

savage872, 2018-10-02:
Thanks you both, especially @crankyoldgit. I have installed openmqttgateway and it's working fine, I just need to get used to it!

closed 2018-10-02 by savage872

crankyoldgit, 2018-10-03:
@savage872 I've also added what I think you wanted to IRMQTTServer in PR #543

rasp111, 2018-12-17:
Hi
First of all Thanks for the wonderful work done by you guys.
The code works amazingly well for me & is very consistent.
I have configured Openhab to publish an MQTT command to an ESP that is acting as MQTT IR BROKER..
Whenever I publish an MQTT message to an switch (say TV ON), the ESP publishes the relevant IR_code on the "ir_server/send" Topic. 
The ESP contains all the IR codes i need in my case stored & i call the relevant codes and publish on the "ir_server/send" Topic.
example:
IR Code to turn ON/ OFF Samsung TV is stored as -> const char* C1 = "7,E0E040BF,32";      
I have the ESP publish channel configured to "ir_server/sent" topic using:
if (String(MQTT_V1_COMMAND_TOPIC).equals(p_topic)) {
    if (payload.equals(String(ACTION_ON))) {V1();}  

char IRT[15]= "ir_server/send";

void V1() {client.publish(IRT,C1);}     // SAMSUNG TV ON/OFF

This succesfully turns on the Samsung TV.

Now that i want to move my code also into the same ESP that is doing IRMQTTSERVER, i want help to avoid client.publish(IRT,C1); and somehow use appropriate format and call the relevant function that already exists like callback(char* topic, byte* payload, unsigned int length).

I am pretty unskilled at programing & hence looking forward for some help to correctly format the data to be sent as payload while trying to use the callback() function:

void V1() {callback(..., ....);}     // SAMSUNG TV ON/OFF
 
Regards
Ravi

crankyoldgit, 2018-12-18:
Answered in #586