crankyoldgit/IRremoteESP8266 #540: Panasonic AC not decoded (but usefull info to resolve)

shyne99, 2018-10-02:
### Version/revison of the library used
2.5.1

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

```
Timestamp : 000662.276
Encoding  : UNKNOWN
Code      : 1CD756CF (220 bits)
Library   : v2.5.1

Raw Timing[439]:
   +  3512, -  1714,    +   466, -   408,    +   466, -  1280,    +   470, -   408, 
   +   466, -   412,    +   466, -   408,    +   466, -   412,    +   462, -   412, 
   +   466, -   414,    +   466, -   408,    +   466, -   412,    +   462, -   412, 
   +   466, -   412,    +   466, -   408,    +   466, -  1280,    +   466, -   412, 
   +   462, -   416,    +   462, -   412,    +   466, -   408,    +   466, -   412, 
   +   462, -   416,    +   462, -   412,    +   462, -  1282,    +   462, -  1284, 
   +   462, -  1288,    +   466, -   412,    +   462, -   412,    +   462, -  1284, 
   +   462, -   416,    +   440, -   438,    +   462, -   412,    +   462, -   412, 
   +   462, -   416,    +   466, -   412,    +   462, -   412,    +   462, -   412, 
   +   440, -   442,    +   462, -   412,    +   462, -   412,    +   460, -   418, 
   +   462, -   416,    +   462, -   412,    +   462, -   418,    +   462, -   412, 
   +   462, -   416,    +   462, -   412,    +   436, -   442,    +   462, -   412, 
   +   460, -   418,    +   462, -   416,    +   462, -   412,    +   460, -   412, 
   +   462, -   420,    +   436, -   438,    +   462, -   412,    +   462, -   416, 
   +   432, -   448,    +   436, -   438,    +   436, -  1310,    +   436, -  1310, 
   +   462, -   420,    +   432, -   442,    +   436, -   438,    +   462, -   416, 
   +   432, -   444,    +   432, - 10008,    +  3480, -  1744,    +   492, -   382, 
   +   492, -  1254,    +   492, -   386,    +   488, -   390,    +   492, -   382, 
   +   492, -   386,    +   488, -   386,    +   492, -   386,    +   492, -   386, 
   +   488, -   386,    +   488, -   386,    +   492, -   386,    +   492, -   382, 
   +   492, -  1258,    +   488, -   386,    +   488, -   390,    +   492, -   386, 
   +   488, -   386,    +   488, -   386,    +   492, -   390,    +   488, -   386, 
   +   488, -  1256,    +   488, -  1258,    +   488, -  1262,    +   488, -   390, 
   +   488, -   386,    +   488, -  1258,    +   488, -   390,    +   488, -   392, 
   +   488, -   386,    +   488, -   386,    +   488, -   394,    +   488, -   386, 
   +   488, -   386,    +   488, -   390,    +   488, -   390,    +   488, -   386, 
   +   488, -   390,    +   462, -   412,    +   488, -   390,    +   462, -  1282, 
   +   488, -   390,    +   456, -   416,    +   458, -  1292,    +   456, -  1288, 
   +   488, -  1258,    +   488, -   392,    +   456, -   422,    +   488, -   390, 
   +   484, -   392,    +   484, -  1262,    +   458, -   420,    +   484, -  1262, 
   +   482, -  1262,    +   488, -   392,    +   484, -   394,    +   484, -   416, 
   +   436, -   442,    +   458, -   416,    +   458, -   422,    +   430, -   448, 
   +   432, -   442,    +   458, -   416,    +   458, -  1296,    +   432, -  1314, 
   +   458, -  1288,    +   432, -  1312,    +   432, -  1322,    +   428, -   446, 
   +   428, -  1318,    +   432, -   442,    +   432, -  1318,    +   432, -  1318, 
   +   428, -   446,    +   428, -  1318,    +   428, -  1322,    +   430, -   448, 
   +   426, -   448,    +   428, -   452,    +   426, -   452,    +   426, -   448, 
   +   428, -   472,    +   400, -   478,    +   402, -   478,    +   402, -   472, 
   +   402, -   476,    +   402, -   472,    +   402, -   478,    +   402, -   472, 
   +   402, -  1348,    +   398, -  1348,    +   398, -  1352,    +   398, -   508, 
   +   370, -   478,    +   398, -   476,    +   398, -   512,    +   366, -   508, 
   +   370, -   502,    +   372, -   508,    +   340, -   538,    +   372, -   504, 
   +   344, -  1400,    +   344, -  1400,    +   346, -  1434,    +   314, -   560, 
   +   316, -   588,    +   290, -   560,    +   314, -   564,    +   396, -   400, 
   +   474, -   400,    +   480, -   394,    +   480, -   404,    +   474, -   400, 
   +   454, -   446,    +   454, -   426,    +   448, -   430,    +   424, -   450, 
   +   428, -   452,    +   448, -   426,    +   426, -   452,    +   424, -  1322, 
   +   454, -   426,    +   450, -   424,    +   426, -   452,    +   428, -   452, 
   +   450, -   424,    +   428, -   446,    +   426, -  1322,    +   454, -   426, 
   +   422, -   450,    +   454, -   426,    +   448, -   430,    +   454, -   426, 
   +   448, -   426,    +   428, -   446,    +   454, -   430,    +   454, -   422, 
   +   452, -   424,    +   424, -   452,    +   452, -   430,    +   424, -   452, 
   +   452, -   426,    +   448, -   426,    +   426, -   456,    +   448, -   426, 
   +   448, -  1296,    +   424, -  1322,    +   426, -  1326,    +   450, -  1270, 
   +   478, -   422,    +   454, -   424,    +   424, -   450,    +   454

uint16_t rawData[439] = {3512, 1714,  466, 408,  466, 1280,  470, 408,  466, 412,  466, 408,  466, 412,  462, 412,  466, 414,  466, 408,  466, 412,  462, 412,  466, 412,  466, 408,  466, 1280,  466, 412,  462, 416,  462, 412,  466, 408,  466, 412,  462, 416,  462, 412,  462, 1282,  462, 1284,  462, 1288,  466, 412,  462, 412,  462, 1284,  462, 416,  440, 438,  462, 412,  462, 412,  462, 416,  466, 412,  462, 412,  462, 412,  440, 442,  462, 412,  462, 412,  460, 418,  462, 416,  462, 412,  462, 418,  462, 412,  462, 416,  462, 412,  436, 442,  462, 412,  460, 418,  462, 416,  462, 412,  460, 412,  462, 420,  436, 438,  462, 412,  462, 416,  432, 448,  436, 438,  436, 1310,  436, 1310,  462, 420,  432, 442,  436, 438,  462, 416,  432, 444,  432, 10008,  3480, 1744,  492, 382,  492, 1254,  492, 386,  488, 390,  492, 382,  492, 386,  488, 386,  492, 386,  492, 386,  488, 386,  488, 386,  492, 386,  492, 382,  492, 1258,  488, 386,  488, 390,  492, 386,  488, 386,  488, 386,  492, 390,  488, 386,  488, 1256,  488, 1258,  488, 1262,  488, 390,  488, 386,  488, 1258,  488, 390,  488, 392,  488, 386,  488, 386,  488, 394,  488, 386,  488, 386,  488, 390,  488, 390,  488, 386,  488, 390,  462, 412,  488, 390,  462, 1282,  488, 390,  456, 416,  458, 1292,  456, 1288,  488, 1258,  488, 392,  456, 422,  488, 390,  484, 392,  484, 1262,  458, 420,  484, 1262,  482, 1262,  488, 392,  484, 394,  484, 416,  436, 442,  458, 416,  458, 422,  430, 448,  432, 442,  458, 416,  458, 1296,  432, 1314,  458, 1288,  432, 1312,  432, 1322,  428, 446,  428, 1318,  432, 442,  432, 1318,  432, 1318,  428, 446,  428, 1318,  428, 1322,  430, 448,  426, 448,  428, 452,  426, 452,  426, 448,  428, 472,  400, 478,  402, 478,  402, 472,  402, 476,  402, 472,  402, 478,  402, 472,  402, 1348,  398, 1348,  398, 1352,  398, 508,  370, 478,  398, 476,  398, 512,  366, 508,  370, 502,  372, 508,  340, 538,  372, 504,  344, 1400,  344, 1400,  346, 1434,  314, 560,  316, 588,  290, 560,  314, 564,  396, 400,  474, 400,  480, 394,  480, 404,  474, 400,  454, 446,  454, 426,  448, 430,  424, 450,  428, 452,  448, 426,  426, 452,  424, 1322,  454, 426,  450, 424,  426, 452,  428, 452,  450, 424,  428, 446,  426, 1322,  454, 426,  422, 450,  454, 426,  448, 430,  454, 426,  448, 426,  428, 446,  454, 430,  454, 422,  452, 424,  424, 452,  452, 430,  424, 452,  452, 426,  448, 426,  426, 456,  448, 426,  448, 1296,  424, 1322,  426, 1326,  450, 1270,  478, 422,  454, 424,  424, 450,  454};  // UNKNOWN 1CD756CF
```

#### Example code used
https://github.com/ToniA/arduino-panasonic-heatpump-controller/blob/master/PanasonicHeatpumpController.ino

Hello guys. I am here because I saw your amazing work on Panasonic A/C. I just wanted to share with you my testing. So basically, I am using the code aboce (github link) to control my AC unit. The controller is A75C3704, and it works well. But, as you can see IRRemoteESP can't decode it. So as the code is working above, and that I am willing to do all the testing necessary on my unit, I thought I could share with people more familiar with developping on IRRemote than me. 

The above code represent this state on the controller : Status : ON, Mode Cool, Fanspeed Auto, Swing auto (both V and H), temp  26 Celsius

I hope it could help you to extend compatibility with this unit. Again, I am willing to help you share code and test sendings on my unit at home. 

Thank you

crankyoldgit, 2018-10-02:
Hey @shyne99 Thanks for the detailed report!

I've done some tests and modification to get it to detect it okay (not perfect). Most of the changes thus far have been to increase the tolerances for the pulse/gap lengths. They were far more out of "spec" than normal. e.g. Normally we allow a +/- 25% error margin. For the message you posted, it required a +/-40% margin to pass.

Question: Is that message you captured from the actual remote, or generated by an esp/arduino running that heatpump library?

If it's from the real remote, I've got no qualms tweaking the values etc. If it is from a simulated/artificial source, then I may be allowing more slop than I should be for detecting it.

Anyway, I'll upload the code in a minute or two. It works, except the model detection isn't working.
My guess is it is an LKE (which allows horizontal and vertical swing).
Can you please include as much information about the A/C unit etc as possible so I can see if it is supposed to be an LKE series (I've got no idea how to confirm that yet ;-) or if it is something new.
e.g. Model Number of head unit/main unit, rough year of manufacture, link to manual if possible, etc etc etc.

crankyoldgit, 2018-10-02:
@shyne99 Can you please try out the code in this branch to see how it goes: https://github.com/markszabo/IRremoteESP8266/tree/issue_540

shyne99, 2018-10-03:
@crankyoldgit , The reading comes directly from the remote control, not from the ESP code. I use the DKE part of the code to make it works 
Here is the label's picture of the label of the AC unit : https://imgur.com/a/9QBPtWO
And a picture of the remote control : https://www.panaservicio.es/images/products/cwa75c3704-mando-original-panasonic.jpg

I'll give a try tonight to the code on this branch, and update the issue accordingly.

I'll also capture a lot of On/Off sequence and post them to pastebin, to see if the error margin is constant or if it is just an issue with this one reading.

crankyoldgit, 2018-10-03:
@shyne99 Cool. Thanks for confirming it came from a real remote. I've merged that PR, so you can just download/clone/pull the `master` branch.

Let us know how it goes with the other codes/messages.

shyne99, 2018-10-03:
Okay so, it seems that this is the MKE type (apparently it is simply written in the Pn of the indoor unit). Detection works great for simple command, but fail if I try those command 

- EcoNavi On/Off
- Auto/ Comfort
- Powerful / Quiet
- Patrol/ E-ion

Also, those command are just send to the unit and not stored inside the controller. To be clear, if I activate EcoNavi mode the Unit light the econavi light. Then I switch the unit off, and on again, the EcoNavi light is Off. So detection only fails If i press one of those special button.

As a user, those feature are not useful, and I never use them so it is not blocking at all for me, but it may not be the same for orhers, so I'll try to help in making them customizable. 

I'll try to pinpoint the byte modified by each one

But those command seems to be specific to MKE models.

On the other hand, I was not able to send data back to the unit at all (even in 2.5.1), I have looked inside the code the send method seems to be missing, and fails at compile time. I try to implement it but I have issue implementing the checksum part)...

crankyoldgit, 2018-10-03:
> I'll try to pinpoint the byte modified by each one

If you could, that would be great. Can you also provide some `rawData[]` & `state[]` captures/lines for those messages too. That will help me build test cases for it.

> But those command seems to be specific to MKE models.

I suspected as much (not one of the supported models) based on a quick glance at the `state[]` codes as there seemed to be bits changed I wasn't familiar with. Hopefully I can work out a signature to tell the difference.

>  I have looked inside the code the send method seems to be missing, and fails at compile time.

Eek. You are right. I missed it. I'll create a PR for it shortly.

crankyoldgit, 2018-10-03:
If you care to try branch`issue_540` it should contain the missing `send()` class method. https://github.com/markszabo/IRremoteESP8266/tree/issue_540

See PR #545 for the changes.

shyne99, 2018-10-04:
@crankyoldgit Okay so just to let You know that I have tested that 
```
/* Copyright 2017 crankyoldgit
* An IR LED circuit *MUST* be connected to the ESP8266 on a pin
* as specified by kIrLed below.
*
* TL;DR: The IR LED needs to be driven by a transistor for a good result.
*
* Suggested circuit:
*     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
*
* Common mistakes & tips:
*   * Don't just connect the IR LED directly to the pin, it won't
*     have enough current to drive the IR LED effectively.
*   * Make sure you have the IR LED polarity correct.
*     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
*   * Typical digital camera/phones can be used to see if the IR LED is flashed.
*     Replace the IR LED with a normal LED if you don't have a digital camera
*     when debugging.
*   * Avoid using the following pins unless you really know what you are doing:
*     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
*     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
*     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
*   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
*     for your first time. e.g. ESP-12 etc.
*/

#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Panasonic.h>

const uint16_t kIrLed = D1;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRPanasonicAc panair(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() {
  panair.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Sending...");

  // Set up what we want to send. See ir_Argo.cpp for all the options.
  panair.stateReset();
  panair.setModel(kPanasonicDke);
  panair.setPower(true);
  panair.setFan(kPanasonicAcFanAuto);
  panair.setMode(kPanasonicAcCool);
  panair.setTemp(25);

#if SEND_PANASONIC
  // Now send the IR signal.
  panair.send();
#else  // SEND_PANASONIC
  Serial.println("Can't send because SEND_PANASONIC has been disabled.");
#endif  // SEND_PANASONIC

  delay(5000);
}
```
and it works like a charm for the simple commands. I'll gather missing information and share them with you guys. Amazing.

shyne99, 2018-10-04:
Okay so I think I have gathered missing info.
It seems that those extra features send a smaller payload to the unit, and always the same one. (Patrol/e-ion for example just send the same payload if I press the key on my remote). It looks like this is a toggle command that is sent and I have nothing on the small screen that indicates that the feature is On or Off, which correlate with the fact that everything this is a constant payload that need to be sent to toggle the feature. 

I have gathered a lot of commands from my remote with AnalyzIR, I'll let you check that everything looks normal for the commited code, and to work with this. 

If you need more intel, just ask, and I'll see what I can do 👍 

https://imgur.com/gallery/UeVnYdh

please read them from bottom to top. 

Those 128 bits payload is what I was talking about, when I press the special keys that have no feedback in the remote control.

crankyoldgit, 2018-10-04:
Thanks for that. Top work.

Can you please capture some of those "short" 128bit messages with `IRrecvDumpV2` please?
I want to make sure of the underlying structure of those messages. i.e. The rawData[] lines.

In order to send those short(er) messages, I'll need to make some changes to `sendPanasonicAC()` and that data will help.

What I _think_ I'll need to do at least is:
Change the line in `ir_Panasonic.cpp` from:
```
  if (nbytes < kPanasonicAcStateLength)  return;
```
to:
```
  if (nbytes < kPanasonicAcSection1Length)  return;
```

and after that  _maybe_ you can send via:
```
uint8_t patrol[16] = (0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x80, 0x66, 0x33, 0x1F};
irsend.sendPanasonicAC(patrol, 16);
```

try it and let me know how it goes.

crankyoldgit, 2018-10-15:
I've gotten a "raw" copy of one of the short Panasonic a/c messages. The changes I've merged in to the `master` branch supports sending and basic decoding of these. It will be in the next release of the library.
The short messages are not deep decoded into specific functions etc, but it will give you the state and size to send with so you can use it.

Marking this closed for now.

closed 2018-10-15 by crankyoldgit

crankyoldgit, 2018-10-21:
This is now live in the new _v2.5.2_ release of the library.