crankyoldgit/IRremoteESP8266 #544: Panasonic AC On and Off timer

MikkelTb, 2018-10-03:
HI.

I find the Data in Panasonic protocol, which contain "Time", "On" and "Off"

Remote control:   (toggel on/off) -> 23C, Fan Auto, Swing Auto, Quiet Off, Powerful Off
AC -> CS-ME14CKPG
Remote -> A75C2311
(my AC does not have a Vertical Swing)
 
On Timer set to 16:00
Off Timer set to 6:30
Time for pusch the Button 18:16

_uint8_t state[27] = {0x02, 0x20, 0xE0,   0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4E, 0x2E, 0x80,   0xAF, 0x00, 0xCA, 0x6B, 0x98, 0x10, 0x00, 0x01, 0x48, 0x04, 0xDB};_

Byte nr 14 (0x4E) Contain information about State of On/Off timer
Byte nr. 19 to 21 (0xCA, 0x6B, 0x98) Contain the timer value
Byte nr 25 and 26 (0x48, 0x04)Contain the Time now

**Byte nr.: 14**
Both Cancel -> 0x48 -> 0100 1000
On -> 0x4A -> 0100 1010
Off -> 0x4C -> 0100 1100
On And Off -> 04E -> 0100 1110
Bit nr 6 and 7 show the state of On and Off timer

**Byte 19 to 21 (0xCA, 0x6B, 0x98)**
This represent the timer which, On timer 16:10, and Off timer 6:30
16:10 -> 16*60+10 -> 970 -> 0011 1100 1010 -> 3CA
CA is byte nr 19.
3 is comming from "0x6B" 0x6B -> 0110 1011 < But only the last 3 bit is taking 
The last 3 bit -> 011
0xCA -> 1100 1010
011 1100 1010 <- 16:10

6:30 -> 6*60+30 -> 390 -> 0001 1000 0110 -> 186
from 0x6B, the first 4 bit is use -> 0110 1000
from 0x98, the first bit is not in use -> 1001 1000
001 1000 0110 -> 0x186 <- 6:30
CA is byte nr 19.

**Byte nr 25 and 26 (0x48, 0x04) Contain the Time now**
The clock was 18:16
18:16 -> 18*60+16 -> 1096 -> 0100 0100 1000 -> 0x448
Byte nr 26 first, and only the last 3 bit, then Byte 25

I hope it make sens

crankyoldgit, 2018-10-04:
@MikkelTb  Nice work!

Yes, after reading it a few time, I think I've got it.

Just to confirm the edge cases. Can you please do the same analysis (for Bytes 19-21 and 25-26) for when the timers are `0:00` (which should be trivial), and for when the clock is `0:00` (again should be trivial) AND for when the clock is `23:59`, and the timers at `23:50`  i.e. their max values.
I suspect that will be 23\*60+59 == 1439 == 0x59F = 0b10110011111 &  23\*60+50 == 1430 == 0x596 == 0b10110010110 respectively. i.e. 11-bits long.
The `0:00` cases should be just 0's.

Once we've confirmed/checked that info. I'll code it up.
Also, could you please check what model number/series the head unit is please? Also the remote too.
See https://github.com/markszabo/IRremoteESP8266/issues/540#issuecomment-426732829 where @shyne99 found theirs.

crankyoldgit, 2018-10-04:
> (my AC does not have a Vertical Swing)

Are you sure about that? As the data/code seems to think it does, yet doesn't have **Horizontal** swing.

MikkelTb, 2018-10-04:
Hi,

I will provide the data for the timer, as soon I get home from work ;-)

My AC has Swing "Up and Down", and not a Swing for "Left and right"

Up and Down must be Horizontal

MikkelTb, 2018-10-04:
AC -> CS-ME14CKPG
Remote -> A75C2311

Picture of my Remote
[image]
[image]

This is my AC, but only with 3. indoor units, and with 3. Remote control
[image]

crankyoldgit, 2018-10-04:
@MikkelTb I got impatient and coded up what I think it probably is based on the available data.

Can you please try out this branch? https://github.com/markszabo/IRremoteESP8266/tree/panasonic-timers

I still want that data I asked for from earlier. ;-)

shyne99, 2018-10-04:
@crankyoldgit Okay so just to let You know that I have tested that 
```
/* Copyright 2017 crankyoldgit
* An IR LED circuit *MUST* be connected to the ESP8266 on a pin
* as specified by kIrLed below.
*
* TL;DR: The IR LED needs to be driven by a transistor for a good result.
*
* Suggested circuit:
*     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
*
* Common mistakes & tips:
*   * Don't just connect the IR LED directly to the pin, it won't
*     have enough current to drive the IR LED effectively.
*   * Make sure you have the IR LED polarity correct.
*     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
*   * Typical digital camera/phones can be used to see if the IR LED is flashed.
*     Replace the IR LED with a normal LED if you don't have a digital camera
*     when debugging.
*   * Avoid using the following pins unless you really know what you are doing:
*     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
*     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
*     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
*   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
*     for your first time. e.g. ESP-12 etc.
*/

#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Panasonic.h>

const uint16_t kIrLed = D1;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRPanasonicAc panair(kIrLed);  // Set the GPIO to be used to sending the message.

void setup() {
  panair.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Sending...");

  // Set up what we want to send. See ir_Argo.cpp for all the options.
  panair.stateReset();
  panair.setModel(kPanasonicDke);
  panair.setPower(true);
  panair.setFan(kPanasonicAcFanAuto);
  panair.setMode(kPanasonicAcCool);
  panair.setTemp(25);

#if SEND_PANASONIC
  // Now send the IR signal.
  panair.send();
#else  // SEND_PANASONIC
  Serial.println("Can't send because SEND_PANASONIC has been disabled.");
#endif  // SEND_PANASONIC

  delay(5000);
}
```
and it works like a charm for the simple commands. I'll gather missing information and share them with you guys. Amazing.

shyne99, 2018-10-04:
Okay so I think I have gathered missing info.
It seems that those extra features send a smaller payload to the unit, and always the same one. (Patrol/e-ion for example just send the same payload if I press the key on my remote). It looks like this is a toggle command that is sent and I have nothing on the small screen that indicates that the feature is On or Off, which correlate with the fact that everything this is a constant payload that need to be sent to toggle the feature. 

I have gathered a lot of commands from my remote with AnalyzIR, I'll let you check that everything looks normal for the commited code, and to work with this. 

If you need more intel, just ask, and I'll see what I can do 👍 

https://imgur.com/gallery/UeVnYdh

please read them from bottom to top. 

Those 128 bits payload is what I was talking about, when I press the special keys that have no feedback in the remote control.

MikkelTb, 2018-10-04:
Hi @crankyoldgit .

I have testede the latest release, and it seem to Work.

But it does not Work on my AC

I have done a small test.
1# Set up the setting in the program to reflect the remote control, and send the command to IR led, and received on IR receiver.

2# Send the settings from the remote to the IR receiver.

(And sorry i have add "const uint8_t kPanasonicAcSwingHOff = 0x00;" in "ir_Panasonic.h")

**_1#_**
_Panasonic A/C remote is in the following   state:
Power: 1,  Mode: 4, Temp: 23C,   Fan Speed: 7, Model: 3, Time: 1380, Ontime: 960, OffTime: 390
IR Code: 0x0220E0040000000602_
**Timestamp : 000053.200
Encoding    : PANASONIC_AC
Code        : 0220E004000000060220E00400472E80AF00C06B980000016405D7 (216 bits)
Mesg Desc.: Model: 3 (DKE), Power: On,   Mode: 4 (HEAT), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Swing   (Horizontal): 0 (Not in use), Quiet: Off, Powerful: Off, Clock: 23:00, On   Timer: 16:00, Off Timer: 6:30
Library     : v2.5.1**
 
_uint8_t state[27] = {0x02, 0x20, 0xE0,   0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x47, 0x2E, 0x80, 0xAF, 0x00, 0xC0,   0x6B, 0x98, 0x00, 0x00,   0x01, 0x64, 0x05, 0xD7};_
 
 **_2#_**
**Timestamp : 000077.414
Encoding    : PANASONIC_AC
Code        : 0220E004000000060220E004004E2E80AF00C06B981000016405EE (216 bits)
Mesg Desc.: Model: 3 (DKE), Power: Off,   Mode: 4 (HEAT), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Swing   (Horizontal): 0 (Not in use), Quiet: Off, Powerful: Off, Clock: 23:00, On   Timer: 16:00, Off Timer: 6:30
Library     : v2.5.1**
 
_uint8_t state[27] = {0x02, 0x20, 0xE0,   0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4E, 0x2E, 0x80, 0xAF, 0x00, 0xC0,   0x6B, 0x98, 0x10, 0x00,   0x01, 0x64, 0x05, 0xEE};_

Byte nr.: 14 and 22 is different from The Code, to the remote.

Byte 14
Code -> 0x47 -> b0100 **0**11**1** 
Remote -> 0x4E -> b0100 **1**11**0**
The 2 bit for timer on, is okay _b0100 1**11**0_ but what the 2 other bit is, i do not know

Byte 22
Code -> 0x00
Remote -> 0x10
I do not know what byte 22 is, all data from my remote, has that value _0x10_ in byte 22

crankyoldgit, 2018-10-04:
Re-stating this:
Could you please check what model series the head unit is please?
See https://github.com/markszabo/IRremoteESP8266/issues/540#issuecomment-426732829 where shyne99 found theirs.

crankyoldgit, 2018-10-04:
Can you also include the source code you are using to produce the states/messages?

crankyoldgit, 2018-10-04:
> I do not know what byte 22 is, all data from my remote, has that value 0x10 in byte 22

Byte 22/`state[21]` is used for Quiet & Powerful mode by all the other existing model series. I suspect that you have a different model that we don't have support for yet. Hence the questions about trying to confirm what model series the head unit is (i.e. the thing that is inside your house etc). See the link to the comment about where to find the model series code.

> Byte 14
Code -> 0x47 -> b0100 0111
Remote -> 0x4E -> b0100 1110
The 2 bit for timer on, is okay b0100 1110 but what the 2 other bit is, i do not know

The bits in Byte 14/`state[13]` are as follows.
The top four are the operating mode. e.g. b**0100**0000 is heat mode. i.e. `setMode()`
The next bit down (b0000**1**000) ... I have no idea yet. It may be specific to your model/series.
The next two (0b00000**11**0) you've decoded as the off & on timer switches respectively.
The last bit (0b0000000**1**) is the Power on/off switch. i.e. 1 = On, 0 = Off. At least for all the models seen thus far. i.e. `setPower()`, `on()`, `off()`. Maybe it's different on your device?

crankyoldgit, 2018-10-04:
Thanks for the image. Like you. I've go no idea what series that is (But by the production date, it's OLD ;-)
I'll do some research to see what I can find/make up for it. :-)

In the meantime. Can you do some captures/tests to check if Byte 14/`state[13]` and the text description behaves as expected for the Power On/Off control via the remote?

crankyoldgit, 2018-10-04:
I found a manual online, there also seems to be an "Odour wash" button on the remote, can you see what changes in the `state[]` with that too please.

MikkelTb, 2018-10-04:
OLD 😖 I got the system in 2008, and it was brand new.... but yes the date is 2003...

I will try to solved/test the Byte 14/state[13] on my AC.

And thanks for you work 👍

MikkelTb, 2018-10-04:
"Odour wash" yes I have that button. But I have never figure out what that means. But I will collect some information

crankyoldgit, 2018-10-05:
Well, we can see if it changes the `state[]` at all at least.

shyne99, 2018-10-05:
Looks like a CKP serie from the label

MikkelTb, 2018-10-05:
>           "Odour wash" yes I have that button. But I have never figure out what that means. But I will collect some information

This is the information, when i push the button:
Timestamp : 057505.742
Encoding  : UNKNOWN
Code      : 1FB51F79 (132 bits)
Library   : v2.5.1

uint16_t rawData[263] = {3496, 1734,  506, 366,  448, 1294,  504, 368,  498, 374,  452, 418,  448, 424,  444, 428,  450, 422,  446, 426,  450, 420,  448, 424,  452, 418,  448, 422,  444, 1300,  498, 374,  504, 368,  448, 424,  452, 418,  448, 424,  444, 428,  450, 422,  446, 1296,  500, 1242,  502, 1242,  504, 368,  498, 374,  452, 1292,  504, 366,  450, 422,  444, 426,  450, 420,  446, 424,  452, 418,  448, 424,  444, 428,  450, 422,  444, 426,  450, 420,  446, 424,  452, 418,  448, 422,  444, 428,  450, 422,  446, 426,  452, 420,  446, 426,  452, 418,  448, 424,  442, 428,  448, 422,  444, 426,  450, 420,  446, 426,  452, 418,  448, 424,  444, 428,  450, 422,  444, 1298,  500, 1244,  500, 372,  444, 428,  450, 422,  446, 426,  452, 418,  448, 10020,  3500, 1732,  498, 372,  452, 1290,  506, 366,  450, 422,  446, 426,  452, 420,  448, 424,  452, 418,  448, 422,  444, 426,  450, 420,  446, 426,  452, 420,  446, 1296,  500, 370,  444, 428,  450, 422,  446, 426,  452, 420,  446, 424,  442, 428,  448, 1294,  502, 1240,  504, 1238,  506, 366,  448, 422,  444, 1298,  498, 374,  452, 418,  448, 424,  444, 428,  450, 422,  446, 426,  450, 420,  446, 424,  452, 418,  448, 422,  444, 428,  450, 420,  446, 1298,  498, 1244,  500, 1242,  502, 368,  446, 1298,  500, 1244,  500, 372,  444, 428,  450, 1292,  504, 368,  446, 1296,  502, 370,  444, 426,  452, 1290,  504, 1238,  506, 366,  450, 422,  446, 1298,  498, 1246,  500, 372,  444, 428,  450, 1294,  452, 420,  446, 1296,  448, 422,  444};  // UNKNOWN 1FB51F79

Timestamp : 057507.664
Encoding  : UNKNOWN
Code      : 1FB51F79 (132 bits)
Library   : v2.5.1

uint16_t rawData[263] = {3502, 1728,  502, 370,  498, 1246,  498, 372,  504, 366,  450, 422,  444, 426,  450, 422,  446, 426,  452, 420,  446, 424,  442, 428,  448, 422,  444, 428,  450, 1320,  476, 368,  500, 372,  442, 428,  450, 422,  444, 426,  452, 420,  446, 424,  452, 1318,  478, 1238,  498, 1246,  498, 372,  504, 366,  448, 1294,  502, 368,  498, 374,  452, 418,  448, 424,  444, 428,  450, 422,  446, 426,  450, 420,  446, 424,  452, 418,  448, 424,  444, 428,  450, 422,  446, 426,  452, 420,  446, 424,  442, 428,  448, 424,  444, 426,  450, 420,  446, 426,  452, 420,  448, 424,  444, 428,  450, 420,  444, 426,  450, 422,  446, 424,  454, 418,  448, 424,  444, 1326,  470, 1246,  500, 372,  506, 366,  448, 422,  444, 428,  450, 422,  446, 10022,  3498, 1732,  498, 374,  452, 1290,  506, 366,  450, 422,  446, 426,  452, 420,  446, 424,  452, 418,  448, 422,  444, 428,  450, 422,  446, 426,  452, 420,  446, 1324,  474, 370,  444, 426,  450, 420,  446, 424,  452, 420,  448, 424,  444, 428,  450, 1294,  502, 1240,  504, 1238,  506, 366,  450, 422,  446, 1298,  498, 372,  454, 418,  448, 424,  444, 428,  450, 422,  446, 426,  452, 420,  446, 424,  442, 428,  448, 422,  444, 428,  450, 420,  446, 1296,  500, 1244,  502, 1240,  504, 368,  446, 1296,  500, 1242,  502, 370,  446, 424,  452, 1292,  506, 366,  450, 1294,  502, 368,  446, 424,  452, 1290,  496, 1246,  498, 374,  452, 420,  446, 1296,  500, 1242,  502, 368,  446, 426,  452, 1290,  444, 428,  450, 1294,  502, 370,  446};  // UNKNOWN 1FB51F79

Timestamp : 057513.378
Encoding  : UNKNOWN
Code      : 1FB51F79 (132 bits)
Library   : v2.5.1

uint16_t rawData[263] = {3498, 1734,  496, 374,  502, 1240,  504, 366,  500, 370,  496, 376,  450, 420,  446, 426,  452, 418,  448, 424,  444, 428,  448, 422,  444, 426,  450, 420,  446, 1296,  500, 372,  506, 366,  448, 422,  446, 426,  452, 420,  446, 424,  452, 418,  448, 1294,  502, 1240,  504, 1240,  504, 366,  500, 370,  444, 1298,  498, 374,  504, 368,  448, 422,  444, 428,  450, 422,  446, 426,  452, 420,  446, 424,  454, 418,  448, 422,  446, 426,  450, 420,  446, 426,  452, 418,  448, 422,  444, 428,  450, 422,  444, 426,  450, 420,  446, 424,  452, 418,  448, 424,  444, 428,  450, 420,  446, 426,  452, 420,  448, 424,  442, 428,  448, 422,  444, 426,  450, 1292,  504, 1266,  478, 366,  502, 370,  446, 426,  452, 420,  450, 420,  442, 10024,  3496, 1736,  442, 428,  450, 1294,  502, 368,  446, 424,  452, 418,  448, 424,  442, 428,  450, 422,  446, 426,  452, 420,  448, 424,  452, 418,  450, 422,  444, 1298,  498, 372,  452, 420,  448, 424,  444, 428,  450, 422,  446, 426,  452, 420,  446, 1296,  502, 1242,  502, 1240,  504, 368,  448, 424,  444, 1300,  496, 374,  452, 420,  448, 424,  442, 428,  450, 422,  444, 426,  452, 420,  446, 424,  452, 420,  448, 422,  444, 426,  450, 420,  446, 1298,  448, 1296,  500, 1242,  502, 370,  446, 1296,  500, 1244,  502, 368,  446, 426,  452, 1290,  506, 366,  448, 1294,  502, 368,  446, 424,  452, 1290,  444, 1298,  498, 374,  450, 420,  448, 1296,  450, 1294,  502, 368,  446, 426,  452, 1290,  444, 428,  450, 1294,  452, 420,  448};  // UNKNOWN 1FB51F79

Timestamp : 057514.616
Encoding  : UNKNOWN
Code      : 1FB51F79 (132 bits)
Library   : v2.5.1

uint16_t rawData[263] = {3504, 1728,  502, 368,  498, 1244,  502, 370,  496, 376,  502, 368,  446, 424,  452, 420,  448, 422,  444, 428,  450, 422,  444, 426,  450, 420,  446, 424,  452, 1290,  496, 376,  502, 370,  446, 426,  452, 420,  448, 424,  442, 428,  450, 422,  446, 1298,  498, 1244,  500, 1242,  502, 370,  498, 374,  452, 1318,  478, 366,  502, 37 [... cut ...]  1242,  504, 1240,  506, 366,  450, 422,  444, 1298,  498, 372,  452, 418,  450, 422,  444, 426,  450, 420,  446, 426,  452, 420,  446, 424,  442, 428,  448, 422,  444, 426,  450, 420,  446, 1296,  500, 1242,  502, 1242,  504, 368,  446, 1296,  500, 1242,  502, 370,  446, 426,  452, 1290,  506, 366,  450, 1294,  502, 368,  446, 424,  452, 1290,  496, 1248,  498, 374,  452, 420,  446, 1296,  502, 1242,  502, 368,  446, 424,  454, 1290,  444, 426,  450, 1292,  452, 418,  448};  // UNKNOWN 1FB51F79

MikkelTb, 2018-10-05:
>           OLD 😖 I got the system in 2008, and it was brand new.... but yes the date is 2003...
> I will try to solved/test the Byte 14/state[13] on my AC.
> And thanks for you work 👍

😖 It look lik the "Powerful" and "Quiet" button, is turn arround in the protocol for my unit.

The First 2 is actual the "Powerful" i turn On and then Off
The secund 2 is actual the "Quiet" i turn On and then Off

Timestamp : 057734.383
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E004004E2E80AF00000EE01100010006B7 (216 bits)
Mesg Desc.: Model: 3 (DKE), Power: Off, Mode: 4 (HEAT), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Swing (Horizontal): 0 (Not in use), Quiet: On, Powerful: Off, Clock: 0:00, On Timer: 0:00, Off Timer: 0:00
Library   : v2.5.1

uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4E, 0x2E, 0x80, 0xAF, 0x00, 0x00, 0x0E, 0xE0, 0x11, 0x00, 0x01, 0x00, 0x06, 0xB7};

Timestamp : 057736.707
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E004004E2E80AF00000EE01000010006B6 (216 bits)
Mesg Desc.: Model: 3 (DKE), Power: Off, Mode: 4 (HEAT), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Swing (Horizontal): 0 (Not in use), Quiet: Off, Powerful: Off, Clock: 0:00, On Timer: 0:00, Off Timer: 0:00
Library   : v2.5.1

uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4E, 0x2E, 0x80, 0xAF, 0x00, 0x00, 0x0E, 0xE0, 0x10, 0x00, 0x01, 0x00, 0x06, 0xB6};

Timestamp : 057750.486
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E004004E2E80AF00000EE03000010006D6 (216 bits)
Mesg Desc.: Model: 3 (DKE), Power: Off, Mode: 4 (HEAT), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Swing (Horizontal): 0 (Not in use), Quiet: Off, Powerful: On, Clock: 0:00, On Timer: 0:00, Off Timer: 0:00
Library   : v2.5.1

uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4E, 0x2E, 0x80, 0xAF, 0x00, 0x00, 0x0E, 0xE0, 0x30, 0x00, 0x01, 0x00, 0x06, 0xD6};

Timestamp : 057752.756
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E004004E2E80AF00000EE01000010006B6 (216 bits)
Mesg Desc.: Model: 3 (DKE), Power: Off, Mode: 4 (HEAT), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Swing (Horizontal): 0 (Not in use), Quiet: Off, Powerful: Off, Clock: 0:00, On Timer: 0:00, Off Timer: 0:00
Library   : v2.5.1

uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4E, 0x2E, 0x80, 0xAF, 0x00, 0x00, 0x0E, 0xE0, 0x10, 0x00, 0x01, 0x00, 0x06, 0xB6};

MikkelTb, 2018-10-07:
@crankyoldgit 

Thats okay, I still do some investigations on the protocol.

MikkelTb, 2018-10-11:
HI.

I have done two small modification to the "ir_Panasonic.h", and now it Work with my AC

**The first one is swop Quiet, and Powerfull**
//const uint8_t kPanasonicAcQuiet = 1;  // 0b1
//const uint8_t kPanasonicAcPowerful = 0x20;  // 0b100000
const uint8_t kPanasonicAcPowerful = 1;  // 0b1
const uint8_t kPanasonicAcQuiet = 0x20;  // 0b100000

The nest is change State[21] in the "kPanasonicKnownGoodState" from "0x00" to "0x10"
//const uint8_t kPanasonicKnownGoodState[kPanasonicAcStateLength] = {
//  0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06,
//  0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00,
//  0x00, 0x0E, 0xE0, 0x00, 0x00, 0x81, 0x00, 0x00, 0x00};

const uint8_t kPanasonicKnownGoodState[kPanasonicAcStateLength] = {
  0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06,
  0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x80, 0x00, 0x00,
  0x00, 0x0E, 0xE0, 0x10, 0x00, 0x81, 0x00, 0x00, 0x00};

I can turn it On, and Off now, and the State data look lige the remote control.

crankyoldgit, 2018-10-11:
Excellent. I'm still travelling, so my access to do development stuff is limited. I'll try to formulate a proper patch/system to handle this model. You data above is fantastic to help with that. Kudos for that.

crankyoldgit, 2018-10-11:
@MikkelTb Can you please try out the code in the  https://github.com/markszabo/IRremoteESP8266/tree/panasonic_ckp branch?

It should add support for you CKP series unit.
FYI, if you're constructing via the `IRPanasonicAc` class you'll need to choose your model type via `setModel(kPanasonicCkp);` and you'll need to do that before you make any `setPowerful()`/`setQuiet()` changes.

MikkelTb, 2018-10-13:
Hi Again

I have testede the Fan, and swing (Vertical), timer on/off, Temperatur, Powerfull and quiet.
Everything is working, on the Ckp unit.

MikkelTb, 2018-10-13:
Hi.

The unit i have testede it on is:
AC -> CS-ME14CKPG
AC -> CS-ME12CKPG
AC -> CS-ME10CKPG
Remote -> A75C2311 (same type of remote for alle 3)

The Number (10,12 and 14) is the model and the diffrent is how mush power (heet/cool) the unit can provide to the air

crankyoldgit, 2018-10-13:
@MikkelTb Thanks for that feedback. Appreciated.
As for the `remote_state[21] == 0x10` vs. `remote_state[21] & 0x10` bit. It needs to be the latter as `remote_state[21]` is changed by the swing states. i.e. If the swing is set to on, then it won't equal `0x10`;
Can you provide me with copies of the `state[]` lines for the ones that are decoding to a different (non-CKP) model for you? I'll then see what I can do to improve detection.

Regarding the "power on" control etc., I'm a little confused, so I'll try to explain back to you what I think I understand and you can correct/update me where needed.

(All of this is assuming data from the normal original remote and just for the CKP model. i.e. NOT sending via the ESP)
* For normal changes (e.g. temp, mode, swing, etc) the Power is read as "Off", even if the A/C unit is still running after receiving the command.
* All the `state[]`s for the above line's operation changes are 27 bytes long.
* Some special mode buttons `state[]`s are 16 bytes long. e.g. Odour Wash.
* Similar to previously discussed/fixed Powerful & Quiet modes, the horizontal and vertical swing settings (setting and reporting) are swapped too. **[Correct?]**
* There is no horizontal swing mode for the unit. **[which currently is reported as vertical, correct?]**

This leads me to some questions:
* Is there a special (short) "Power On" message when you toggle the power button on the remote? or does the ESP just report the normal `state[27]` but decoded as "Power: On"?
* Does it matter to the A/C unit when the ESP sends a command to the A/C unit to change a mode (temp, mode, fan speed, etc) if the `setPower(true);` is set? i.e. The ESP would detect the state as "Power: On", and the A/C unit would still a) Keep running, or start if it was previously off, and b) change to what ever else desired modes are requested?
e.g.
```
// Assuming the A/C unit is already running in Heat mode at 30C, and fan speed of minimum before this is sent.
IRPanasonicAc pana(IR_LED_PIN_YOU_ARE_USING);
pana.setPower(true);
pana.setMode(kPanasonicAcCool);
pana.setTemp(21);
pana.setFan(kPanasonicAcFanMax);
pana.send();
// A/C unit should now be still on/working, but in Cool mode, at 21C and at max fan speed.
```

crankyoldgit, 2018-10-13:
Ah. Okay. Cool. I now think I understand the power setting thing. Thanks for clarifying that. I'll add a note in the code for that and leave it alone, or do something fancy. I'll figure something out. :-)

MikkelTb, 2018-10-13:
That sounds great. And thanks for implement the other stuff 👍

crankyoldgit, 2018-10-13:
@MikkelTb 
From before:
Can you provide me with copies of the `state[]` lines for the ones that are decoding to a different (non-CKP) model for you? I'll then see what I can do to improve detection.

MikkelTb, 2018-10-13:
@crankyoldgit 

Unfortunately I do not have a another type of Panasonic AC.

So I will not be able to provide you with that kind of information.

crankyoldgit, 2018-10-14:
@MikkelTb I was referring to your comment earlier of: _"i have replace "&" with "==", now the program show the right Model."_

i.e. Can you give me a copy of a `state[]` line where the model was being reported incorrectly before?

MikkelTb, 2018-10-14:
**_data from the software_**
IRrecvDumpV2 is now running and waiting for IR input on Pin 14
Default state of the remote.
Panasonic A/C remote is in the following state:
Mesg Desc.: Model: 4, Power: 0,  Mode: 0, Temp: 0C, Fan Speed: 253, Swing (Vertical): 0, Quiet: 0, Powerful: 0,  Time: 0, Ontime: 0, OffTime: 0
IR Code: 0x0220E0040000000602
Setting desired state for A/C.
Panasonic A/C remote is in the following state:
Mesg Desc.: Model: 4, Power: 0,  Mode: 6, Temp: 23C, Fan Speed: 7, Swing (Vertical): 15, Quiet: 0, Powerful: 0,  Time: 1380, Ontime: 960, OffTime: 390
IR Code: 0x0220E0040000000602
I received: 49
stateon Send
Panasonic A/C remote is in the following state:
Mesg Desc.: Model: 4, Power: 1,  Mode: 6, Temp: 23C, Fan Speed: 7, Swing (Vertical): 15, Quiet: 0, Powerful: 0,  Time: 1380, Ontime: 960, OffTime: 390
IR Code: 0x0220E0040000000602
Timestamp : 000044.671
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E00400672E80AF00C06B98100081640587 (216 bits)
Mesg Desc.: Model: 4 (JKE), Power: On, Mode: 6 (FAN), Temp: 23C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Quiet: Off, Powerful: Off, Clock: 23:00, On Timer: 16:00, Off Timer: 6:30
Library   : v2.5.1
uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x67, 0x2E, 0x80, 0xAF, 0x00, 0xC0, 0x6B, 0x98, 0x10, 0x00, 0x81, 0x64, 0x05, 0x87};

**_(Data from my Remote control)_**
Timestamp : 000050.304
Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E004004F2C80AF00000EE01000010006B5 (216 bits)
Mesg Desc.: Model: 5 (CKP), Power: On, Mode: 4 (HEAT), Temp: 22C, Fan: 7 (AUTO), Swing (Vertical): 15 (AUTO), Quiet: Off, Powerful: Off, Clock: 0:00, On Timer: 0:00, Off Timer: 0:00
Library   : v2.5.1
uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x4F, 0x2C, 0x80, 0xAF, 0x00, 0x00, 0x0E, 0xE0, 0x10, 0x00, 0x01, 0x00, 0x06, 0xB5};

crankyoldgit, 2018-10-14:
@MikkelTb Thanks for that info. That's helped me try to narrow it down some more.
I've updated the branch: https://github.com/markszabo/IRremoteESP8266/tree/panasonic_ckp and created a PR (#554) for it.

Hopefully, it should now detect better for you. Let me know how it goes.

MikkelTb, 2018-10-14:
Hi @crankyoldgit

I will try it tomorrow. I have look in the code and it looks great 👍

MikkelTb, 2018-10-15:
I close this one, as the new program update, handle this function now.

And thanks to everybody, for helping with this, and special @crankyoldgit  👍

closed 2018-10-15 by MikkelTb