crankyoldgit/IRremoteESP8266 #556: Add support for SAGA by Tornado AC unit.

josh929800, 2018-10-16:
Support for SAGA by Tornado AC units (using Whirlpool protocol). 

### Version/revison of the library used
_v2.5.1

### Expected behavior
_What steps did you do and what should it have done?_

Send Short Whirlpool Command to AC Unit.

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

The ESP8266 crashes (no command sent) and I have to re-flash it. If I send the raw command it issuing the command successfully then crashes and I have to re-upload the program. If I have the raw command pre-loaded on the device it sends without a problem. 

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

Here are the Raw Dumps I also have worked out a spreadsheet with the decoding of what each bit is doing. I can even write a IR command and successfully send it to the device.  I have tried to reverse engineer some of your AC controls to make it work but so far I have not been able to make it work. 

[IR Remote Code modified4.xlsx](https://github.com/markszabo/IRremoteESP8266/files/2482949/IR.Remote.Code.modified4.xlsx)

###################
#   OFF Commands  #
###################

Fan mode low current time 0:00 timer off 0:01
message sentuint16_t rawData[343] = {9080, 4424,  682, 1568,  682, 1568,  684, 450,  680, 454,  682, 458,  682, 462,  682, 464,  682, 1578,  680, 446,  682, 1570,  682, 1572,  682, 454,  682, 458,  680, 462,  682, 464,  682, 454,  682, 1566,  682, 1570,  682, 452,  682, 454,  680, 460,  682, 460,  682, 464,  682, 454,  680, 444,  680, 448,  682, 1572,  680, 456,  680, 1582,  682, 1584,  682, 1586,  680, 454,  682, 444,  680, 448,  682, 450,  682, 454,  682, 456,  680, 462,  682, 464,  658, 478,  680, 444,  682, 446,  682, 450,  680, 454,  656, 482,  682, 462,  682, 464,  682, 436,  682, 7866,  656, 470,  682, 448,  682, 450,  682, 454,  656, 482,  680, 462,  656, 490,  656, 1602,  656, 470,  680, 448,  682, 450,  682, 454,  682, 458,  682, 460,  682, 464,  682, 1574,  684, 442,  682, 448,  682, 450,  680, 456,  682, 458,  654, 488,  682, 464,  682, 452,  684, 1566,  682, 448,  656, 478,  682, 452,  682, 458,  682, 460,  682, 466,  682, 454,  682, 444,  680, 448,  682, 452,  680, 456,  682, 458,  678, 464,  680, 466,  682, 452,  682, 444,  682, 448,  680, 450,  654, 482,  682, 456,  682, 460,  682, 464,  680, 454,  682, 444,  682, 446,  682, 450,  684, 452,  682, 458,  682, 462,  680, 464,  680, 456,  654, 470,  682, 1568,  682, 1572,  682, 454,  680, 1580,  682, 1584,  682, 1586,  682, 436,  682, 7870,  684, 442,  682, 446,  682, 450,  682, 454,  680, 458,  684, 460,  682, 466,  682, 454,  656, 1592,  682, 446,  682, 1574,  654, 1604,  680, 1582,  682, 462,  682, 464,  682, 454,  682, 444,  682, 448,  680, 452,  682, 454,  682, 458,  682, 460,  680, 466,  680, 456,  656, 470,  682, 446,  680, 452,  680, 454,  682, 456,  684, 460,  684, 464,  682, 454,  684, 442,  682, 448,  680, 452,  654, 1602,  682, 458,  682, 460,  680, 466,  682, 454,  682, 444,  654, 474,  654, 478,  682, 454,  682, 458,  682, 460,  682, 464,  656, 480,  680, 1566,  682, 448,  680, 1574,  682, 454,  682, 1580,  682, 460,  682, 464,  680, 438,  672};  // WHIRLPOOL_AC
09:38:50.392 -> uint8_t state[21] = {0x83, 0x06, 0x03, 0x74, 0x00, 0x00, 0x80, 0x80, 0x00, 0x01, 0x00, 0x00, 0x00, 0x76, 0x00, 0x1D, 0x00, 0x00, 0x08, 0x00, 0x15};

Fan mode high current time 0:00 timer off 0:01
message sentuint16_t rawData[343] = {9070, 4422,  680, 1566,  682, 1568,  682, 450,  682, 454,  680, 460,  680, 462,  654, 490,  680, 1576,  682, 444,  682, 1570,  682, 1574,  682, 454,  656, 484,  682, 462,  682, 466,  682, 454,  682, 1566,  654, 474,  682, 452,  682, 454,  682, 458,  682, 460,  684, 464,  680, 454,  656, 470,  656, 474,  682, 1574,  682, 454,  682, 1580,  682, 1582,  680, 1588,  682, 454,  682, 444,  682, 446,  682, 450,  680, 454,  682, 458,  680, 462,  654, 492,  680, 456,  680, 444,  682, 446,  682, 450,  682, 454,  682, 456,  682, 462,  682, 464,  682, 436,  682, 7868,  682, 444,  684, 446,  680, 452,  680, 458,  676, 458,  682, 458,  684, 464,  680, 1576,  684, 442,  680, 450,  680, 450,  680, 454,  682, 456,  682, 462,  680, 464,  684, 1574,  682, 444,  682, 446,  682, 450,  680, 456,  680, 458,  682, 460,  682, 464,  682, 454,  680, 1566,  680, 448,  680, 452,  684, 452,  654, 484,  682, 462,  682, 464,  682, 454,  682, 444,  680, 448,  682, 450,  682, 454,  682, 458,  682, 462,  682, 464,  682, 454,  680, 446,  682, 446,  682, 450,  682, 454,  674, 466,  680, 460,  682, 466,  682, 454,  682, 444,  654, 474,  680, 452,  682, 454,  680, 458,  682, 462,  682, 464,  680, 456,  682, 442,  682, 448,  680, 1572,  682, 454,  684, 1578,  682, 1584,  682, 1588,  682, 436,  684, 7866,  682, 444,  682, 446,  682, 450,  682, 454,  656, 484,  682, 460,  682, 464,  656, 480,  682, 1566,  682, 446,  682, 1572,  682, 1578,  682, 1580,  682, 460,  682, 464,  680, 456,  682, 442,  682, 446,  682, 450,  682, 454,  680, 458,  682, 460,  684, 464,  680, 456,  680, 444,  682, 446,  682, 452,  684, 452,  682, 456,  682, 462,  680, 466,  682, 454,  682, 442,  680, 450,  654, 478,  682, 1576,  682, 458,  680, 462,  684, 464,  656, 480,  684, 442,  682, 448,  680, 452,  682, 454,  684, 456,  682, 462,  682, 464,  682, 454,  682, 1568,  682, 448,  682, 1572,  682, 456,  682, 1580,  682, 462,  682, 466,  682, 438,  674};  // WHIRLPOOL_AC
09:45:00.501 -> uint8_t state[21] = {0x83, 0x06, 0x01, 0x74, 0x00, 0x00, 0x80, 0x80, 0x00, 0x01, 0x00, 0x00, 0x00, 0x74, 0x00, 0x1D, 0x00, 0x00, 0x08, 0x00, 0x15};

Dehumidify Current time 0:00 Timer Off 0:01 
message sentuint16_t rawData[343] = {9068, 4420,  682, 1564,  682, 1568,  656, 476,  680, 454,  680, 458,  682, 460,  654, 490,  680, 1576,  680, 444,  680, 1568,  682, 1572,  654, 480,  682, 456,  680, 460,  682, 462,  682, 454,  682, 442,  656, 472,  654, 478,  682, 454,  678, 460,  682, 462,  678, 466,  682, 454,  682, 1564,  654, 1596,  680, 452,  680, 456,  680, 1580,  680, 1584,  652, 1614,  654, 482,  678, 448,  652, 476,  650, 482,  652, 482,  650, 490,  652, 490,  650, 496,  602, 534,  600, 524,  602, 528,  602, 530,  600, 536,  600, 538,  602, 542,  602, 544,  602, 518,  600, 7948,  602, 524,  602, 528,  600, 530,  602, 534,  602, 538,  600, 542,  602, 546,  600, 1656,  602, 524,  602, 528,  600, 532,  600, 534,  602, 538,  602, 542,  602, 544,  602, 1656,  600, 524,  600, 528,  602, 530,  600, 534,  600, 538,  600, 542,  600, 544,  602, 534,  600, 1646,  600, 528,  600, 530,  602, 534,  600, 538,  602, 540,  602, 544,  602, 534,  600, 524,  602, 528,  600, 532,  600, 534,  600, 538,  602, 542,  602, 544,  602, 536,  600, 524,  600, 528,  600, 530,  602, 534,  602, 538,  602, 542,  602, 546,  600, 534,  600, 524,  600, 528,  600, 532,  600, 534,  600, 538,  600, 542,  600, 546,  600, 534,  600, 524,  602, 1650,  600, 532,  600, 534,  602, 1660,  602, 1664,  630, 1638,  602, 516,  602, 7948,  658, 468,  658, 470,  658, 474,  658, 478,  656, 482,  658, 484,  660, 486,  660, 476,  658, 1588,  660, 468,  660, 1594,  660, 1598,  660, 1602,  660, 482,  660, 486,  660, 474,  660, 464,  660, 468,  662, 470,  660, 476,  660, 478,  660, 482,  660, 486,  660, 476,  660, 464,  660, 470,  660, 472,  660, 476,  660, 478,  662, 482,  660, 488,  658, 478,  658, 466,  660, 470,  658, 474,  658, 1598,  658, 480,  658, 486,  656, 488,  658, 478,  658, 468,  656, 472,  658, 474,  656, 480,  654, 484,  656, 486,  656, 490,  654, 480,  654, 1594,  626, 502,  652, 1602,  626, 508,  626, 1636,  624, 518,  624, 522,  626, 492,  646};  // WHIRLPOOL_AC
11:48:44.158 -> uint8_t state[21] = {0x83, 0x06, 0x00, 0x73, 0x00, 0x00, 0x80, 0x80, 0x00, 0x01, 0x00, 0x00, 0x00, 0x72, 0x00, 0x1D, 0x00, 0x00, 0x08, 0x00, 0x15};

####################
#  AC On Commands  #
####################

Dehumidify Current time 0:00 Timer On 0:01 
message sentuint16_t rawData[343] = {9048, 4450,  682, 1566,  682, 1568,  682, 450,  682, 454,  682, 456,  682, 460,  654, 490,  682, 1576,  682, 442,  682, 1568,  684, 1570,  684, 452,  682, 456,  656, 486,  682, 464,  682, 452,  682, 444,  682, 446,  682, 452,  680, 454,  682, 456,  682, 462,  682, 464,  682, 454,  682, 1564,  684, 1568,  654, 478,  684, 452,  656, 1606,  680, 1586,  682, 1586,  682, 454,  682, 442,  682, 446,  682, 450,  684, 452,  684, 456,  684, 460,  682, 464,  682, 454,  656, 470,  682, 446,  682, 452,  682, 454,  682, 458,  680, 462,  656, 490,  684, 434,  684, 7868,  654, 472,  682, 448,  684, 448,  682, 454,  658, 482,  682, 460,  682,  [... cut ...] emp between 16 and 30. Note: I am using the timer function to turn on and off the AC as my home automation scripts don't know the current state of the AC unit and there is only a power toggle. 
 
### Has this library/code previously worked as expected for you?
_Yes it is working Not as I expect. 

### Other useful information
_I am willing to do the work on the coding front but I need a little help understanding how to input settings and get the desired bits out. 

Thank You for Everything!
Josh

crankyoldgit, 2018-10-16:
What is the MQTT message you are sending to the IRMQTTServer code that causes it to crash?
Have you tried sending it via the web interface instead?
Can you include the 'info' section of the main IRMQTTServer web page so I can see what version/libraries are being used?

Also any serial output from IRMQTTServer and/or the MQTT responses sent back on the 'ack' channel.

josh929800, 2018-10-16:
I tried 45,0x83, 0x06, 0x00, 0xC2, 0x00, 0x00, 0xB5, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x00, 0x78, 0x00, 0x00, 0x00, 0x00, 0x08, 0x00, 0x08
and 45,83,06,00,C2,00,00,B5,0F,00,00,00,00,00,78,00,00,00,00,08,00,08
Then I switched to raw...
30,38000,9078,4424,678,1575,678,1575,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,1575,678,1575,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,1575,678,1575,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,7868,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,1575,678,1575,678,1575,678,457,678,7868,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,1575,678,1575,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,457,678,1575,678,457,678,1575,678,457,678,1575,678,457,678,457,678,457,678

The webpage worked for the raw but not the AC code.

crankyoldgit, 2018-10-16:
Per the documentation in the code: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L72

```
 *   ac_protocol_num,really_long_hexcode  e.g. 18,190B8050000000E0190B8070000010F0
 *                           Kelvinator (18) Air Con on, Low Fan, 25 deg etc.
 *                           NOTE: Ensure you zero-pad to the correct number of
 *                                 digits for the bit/byte size you want to send
 *                                 as some A/C units have units have different
 *                                 sized messages. e.g. Fujitsu A/C units.
```
No extra commas in the A/C code.

So the code you'd use would be:
**via mqtt:**
`45,830600C20000B50F00000000007800000000080008`
**via web:**
Select `Whirlpool` and use `830600C20000B50F00000000007800000000080008` in the appropriate field, then click send.

josh929800, 2018-10-17:
Thanks the short commands are still not working probably because my AC is not Whirlpool. But I was able to make the Raw work this time.

crankyoldgit, 2018-10-17:
The `45,830600C20000B50F00000000007800000000080008` wasn't working because the sendWhirlpoolAC() call in the example was missing. i.e. _D'oh!_ See PR #558 for the suggested fix.

closed 2018-10-18 by crankyoldgit

crankyoldgit, 2018-10-21:
This is now live in the new _v2.5.2_ release of the library.

PR #558 (2018-10-18): Add missing sendWhirlpoolAC() to IRMQTTServer.ino
Fixes #556
commit 47fab510 (2018-10-18): Add missing sendWhirlpoolAC() to IRMQTTServer.ino (#558)

Fixes #556