crankyoldgit/IRremoteESP8266 #559: [Housekeeping] Add bwarden as a contributer

crankyoldgit, 2018-10-19:
Hey @markszabo , I'd like to recommend @bwarden as a formal contributer to the library, only if they want it of course. i.e. Contributer access.

@bwarden has shown fairly deep understanding of the library by adding a send & decode routine for a very non-trivial protocol (MWM), and (swoon!) wrote unit tests for it.

Brett, if you don't mind, even if you don't contribute any more code (which I hope isn't the case) I'd love you to accept so I can hit you up for code reviews. Poor Mark cops the brunt of them.

David

markszabo, 2018-10-19:
Great idea @crankyoldgit, I've added them. @bwarden thank you for your contribution, it is greatly appreciated.

crankyoldgit, 2018-11-05:
Excellent. Didn't want to add you without your consent. ;-)

closed 2018-11-05 by crankyoldgit