crankyoldgit/IRremoteESP8266 #565: Possible issue with RC5 Protocol

arihantdaga, 2018-11-02:
I am using Version 2.5.2 of the library.

I think there is a possible issue with the RC5 Protocol. 
In my case i am trying to fire a code irsend.sendRC5(0x145A, 14); And in analysIr Code detector i get 0x122D. 
I really think there is some issue with RC5.

arihantdaga, 2018-11-02:
My bad once again.. I feel really embarrassed... But now I think the issue is with the decoder(AnalysIr) which i was trusting blindly till now. But somehow AnalysIr is not decoding codes properly. 

There is still some gotcha - for example if i send a code like sendRC5(0x18B5, 14), in arduino IrRemote library its decoded as 0x185B - 13 bit. (a bit less then what we sent). 
Anyway, right now i patched it for my condition. I am closing the issue anyway. I am sorry once again. 
But i think a check on it once again could be done.

closed 2018-11-02 by arihantdaga

crankyoldgit, 2018-11-02:
See https://github.com/markszabo/IRremoteESP8266/issues/380#issuecomment-435404885

`sendRC5(0x18B5, 14)` is likely wrong. Use either `sendRC5(0x18B5, 13)` or `sendRC5(0x18B5)` for RC5-X or `sendRC5(0x18B5, 12)` for RC5.

The lengths used by this library (and IRremote) are different to that of AnalysIR.
It's really best to use `IRrecvDumpV2` to decode the raw signal and it will tell you the parameters you need to call this library with etc.
In most cases, just passing the data and no length to the appropriate send routine is all that is needed in IRremoteESP8266. It knows what to do in most cases. i.e.   `sendRC5(0x18B5);`