crankyoldgit/IRremoteESP8266 #566: NEC decoding sometimes misses key change and continues to output FFFFFFFFFF

Sascha8, 2018-11-02:
### Using version V2.51

### Expected behavior
Loaded the included example "IRrecvDemo" and have trouble with NEC decoding. Using an ESP8266 with an TSOP31238 IR-receiver on Pin D3

### Actual behavior

Sometimes decode results have the value of the last key was pressed. If you hold i.e VolumeUp for a couple of seconds you get one code and in the following the repeating code (FFFFFFFFF). If you release the key, wait two seconds and hold a **different** key pressed, you get FFFFFFFF without the new code! Example output:

20DF40BF                <-- VolumeUp hold down
FFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFF   <-- Released key. Wait 2 seconds
FFFFFFFFFFFFFFFF   <-- VolumeDOWN hold down
FFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFF   <-- VolumeDown released. Wait some time. Here shouldn't be FFFFFFFF
20DFC03F               <-- VolumeDown hold down again. Got the correct code
FFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFF

I use  a LG remote control. But tried it with different NEC coded controls. The problem is easy to reproduce for me. Sometimes with waiting time of 2 seconds, somtimes only 300ms after releasing the key.

Any tips to solve this issue?

Thanks,
Sascha

Sascha8, 2018-11-02:
Thanks for response!
Yes, this example is indeed much better and much more control is possible.
But the problem is still there. Not as often as previous but reproducing is still quite easy.

If you use the RC in a straight line of the receiver, everything works fine. But if you hold the RC in a bigger angle/distance to the IR-Rec then the problem occurs again. 
Feels like the repeating code will be sent, even if the receiver couldn't get a clear signal (due to distance, angle or noise from surrounding light) to decode safely.

Try following:

Hold down a key until FFFFFFF are send. While holding down the key, move the RC to a complete different direction, that nothing will received. Then hold down a different key and move the RC back to the receiver --> the receiver continues to output FFFFFFFFF. It should output the real (new) code or at least should not output FFFFFFFFF.

crankyoldgit, 2018-11-02:
Ah, you might have a slight mis-understanding of how the NEC repeat works.

For the NEC (and some similar protocols) there is a special "short" code that is sent to signify a repeat/button held down message.

e.g. Hold down the "Volume" button, and it will send a 32-bit hex code, then it will send a special 1-bit code while the button is held down. e.g. "Volume code" + "repeat code" + "repeat code" + "repeat code" + ...
It is NOT: "Volume code" + "Volume code" + "Volume code" + ...

(Note: the repeat code is NOT "FFFFFFFFFFF" etc, that just the library making up a value for the special code.)

So your example of:
(Point remote at receiver &  Hold down "Volume Up") .. the receiver sees the single "Volume up" and the many "repeat" messages.
(While still holding down the button, turn the remote away) The receiver no longer sees the "repeat" messages.
(Stop holding the "Volume Up" button) Nothing is sent. The receiver sees nothing.
(Still pointed away from the receiver, press and hold the "Volume Down" button) .. The receiver doesn't see the single "Volume Down" message, and the many "repeat" messages.
(Now point the remote at the receiver, still with the button held down) The receiver now can see the many "repeat" messages the remote is sending. So, the receiver doesn't see the "new" code for the different key that was pressed.

Or putting it another way. Each button press sends a single message code for what button was pressed, and then many special messages that effectively say "A button is being held down". It stops sending the repeat messages when the button is no longer held.
i.e. The "code" for what key was sent is only sent once, at the very beginning of the sequence. The repeat code DOES NOT include what key was pressed.

crankyoldgit, 2018-11-02:
I understand and appreciate you would want the code (32-bits of data) being repeated in subsequent messages. However, the library doesn't store or know what the "previous" message was. It reports on what it saw/captured.

e.g. Imagine a NEC 32-bit code is seen (0xDEADBEEF), and then a Pioneer message (0xF00BAAC0FFEE), and then a NEC repeat message. What should the library do? Both protocols use the same "NEC style" repeat code. Give the NEC 32-bit number, or the Pioneer? How long should it remember it for. Note also, that the NEC repeat "code" is used in several other protocols too. So, just reporting that the library sees is the safest thing the library can do.

The FFFFF-value-thing is legacy from the original IRremote Arduino library. It's kept that way because that's the way it's been and lets people not have to change their code and makes porting code easier. Each captured/decoded message _must_ have a data value due to the data structure used. Someone years and years and years ago decided that FFFFFFFF etc was the safest value to use and be obvious it means nothing. i.e. The repeat code doesn't contain any data, but the library has to report some data. Zero could have been an alternative. Anyway. It's legacy. It won't be changed as it would break backward/cross compatibility.

As to why the NEC protocol does what it does. I have no definitive answer. It's just the way the NEC engineers designed it. It is what it is. It has plenty of advantages and disadvantages. I'm not going to go into them here.

You are more than welcome to code up something if you want that relies on what IR messages you've previously seen. The library lets you do that. It's just the library doesn't store any previous state. If you want to store previous state, that's up to you. Not storing the previous state gives the library the greatest flexibility for potential different use cases.

closed 2018-11-02 by crankyoldgit

Sascha8, 2018-11-02:
okay, understood why FFFF value is necessary and you want to keep backwards compatibility. Its just a few lines code to replace FFFFF... No big matter.

What about the main topic "repeat timeout" as I described above? In current state of the lib its to easy to get a misinterpreted code value.

crankyoldgit, 2018-11-02:
I'm not 100% sure what you mean by:
> What about the main topic "repeat timeout" as I described above? In current state of the lib its to easy to get a misinterpreted code value.

Can you please rephrase exactly what you mean?

The library reports on what it sees. If it doesn't see a 32 bit NEC "data" message, it can't report it.

If you think it is seeing a 32bit message, and **not** reporting it, can you please include the `uint16_t rawData = {...];` line and output from `IRrecvDumpV2` and I can try to debug it.

crankyoldgit, 2018-11-03:
Yes, that is the kind of procedural logic a receiving device would typically implement when using the NEC protocol.

The IRrecvdemo & IRrecvDump etc programs are not trying to emulate a receiving device. The former is just a very simple example of how to use the library when receiving.  The later is trying to display everything it sees in as much detail as the library can provide. Neither are trying to emulate the logic a "controlled" device has. The library does not try to make any assumptions as protocols differ, devices are implemented differently etc. It's virtually impossible for it to be have "correctly" in all possible use cases as "correctly" is subjective. Hence it is up to the programmer/developer to work out what to do with the results/input signals.
e.g. Some devices/protocols/interactions may consider a repeated signal differently, such as "the button is held down, so keep doing the same thing as before" or "the button is being held down, so the action hasn't finished yet." Think of how your computer & mouse, and phone & touch interactions work. Depending on the application/instance/situation. A mouse "down" click" could be a short click, a long click, a hold & drag, a desire to start an action, or the action could start when the "down" click is finished.

This library is situated at a level similar to the mouse/touch screen driver in a computer. It provides raw/raw-ish information to a the next layer, which has to determine the "intention" of such inputs. When implementing the "intentions" you need to have access to low level implementations. Cooking the results some to do what you suggest removes some of that low level information that can allow people to implement what they want to do. This library wants to allow the most flexibility when dealing with IR, thus keeping it low-level and raw-ish helps achieve that.