crankyoldgit/IRremoteESP8266 #567: How to turn on/off Panasonic TV

Lahim26, 2018-11-02:
Guys, how do you switch off/on Panasonic TV?
I am trying from web page without luck.
I checked the JVSPANASONICSendDemo and it works (led blinks) but I am not able to get it working with MQTT server.

_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used
Version: v0.7.0
Period Offset: -5us
IR Lib Version: 2.5.2
ESP8266 Core Version: 2_4_2

### Expected behavior
Checked JVCPANSONICSendDemo and it works.

### Actual behavior
Nothing happens

#### Example code used
Sending from ESP8266 IR MQTT Server page code 
PANASONIC 0x100BCBD -> 48 bits -> repeat 3 times
Does nothing - the diode does not even blink

crankyoldgit, 2018-11-02:
_JVCSPANASONICSendDemo_ uses `sendPanasonic()` whilst IRMQTTServer uses `sendPanasonic64()`

In short, you need to use `0x40040100BCBD` instead of `0x100BCBD`.
i.e. `0x100BCBD + 0x400400000000 = 0x40040100BCBD`

crankyoldgit, 2018-11-04:
Glad it worked for you.

closed 2018-11-04 by crankyoldgit