crankyoldgit/IRremoteESP8266 #568: How to calculate NEC code?

Bluebrain2000, 2018-11-04:
What I have:
NEC1
device 128
subdevice 126
OBC 32

I found several converters, e.g. to Pronto Hex, but how to calculate the NEC code needed for sendNEC()?

crankyoldgit, 2018-11-04:
If you compile & use the program in the tools directory called `gc_decode` it will help you convert the pronto code into a NEC hex string suitable for `sendNEC();`
e.g.
(Pronto code used was grabbed from some random place on the internet. i.e. not related to your question.)
```
IRremoteESP8266/tools$ make gc_decode
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/IRutils.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/IRtimer.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/IRsend.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/IRrecv.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_NEC.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Sony.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Samsung.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_JVC.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_RCMM.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_RC5_RC6.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_LG.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Mitsubishi.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Fujitsu.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Sharp.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Sanyo.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Denon.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Dish.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Panasonic.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Whynter.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Coolix.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Aiwa.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Sherwood.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Kelvinator.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Daikin.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Gree.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Pronto.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_GlobalCache.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Nikai.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Toshiba.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Midea.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Magiquest.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Lasertag.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Carrier.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Haier.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Hitachi.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_GICable.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Whirlpool.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Lutron.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Electra.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_Pioneer.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -c ../src/ir_MWM.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -I../src -I../test -c gc_decode.cpp
g++ -DUNIT_TEST -g -Wall -Wextra -pthread -lpthread IRutils.o IRtimer.o IRsend.o IRrecv.o ir_NEC.o ir_Sony.o ir_Samsung.o ir_JVC.o ir_RCMM.o ir_RC5_RC6.o ir_LG.o ir_Mitsubishi.o ir_Fujitsu.o ir_Sharp.o ir_Sanyo.o ir_Denon.o ir_Dish.o ir_Panasonic.o ir_Whynter.o ir_Coolix.o ir_Aiwa.o ir_Sherwood.o ir_Kelvinator.o ir_Daikin.o ir_Gree.o ir_Pronto.o ir_GlobalCache.o ir_Nikai.o ir_Toshiba.o ir_Midea.o ir_Magiquest.o ir_Lasertag.o ir_Carrier.o ir_Haier.o ir_Hitachi.o ir_GICable.o ir_Whirlpool.o ir_Lutron.o ir_Electra.o ir_Pioneer.o ir_MWM.o gc_decode.o -o gc_decode
IRremoteESP8266/tools$ ./gc_decode -prontohex "0000 006C 0022 0002 015B 00AD 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0016 0016 0016 0016 0041 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0016 0041 0016 0016 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0041 0016 0622 015B 0057 0016 0E6C"
Code length 76
Code type      3 (NEC)
Code bits      32
Code value     0x17e40bf
Code address   0x7e80
Code command   0x2
```

Then you would use:
`sendNEC(0x17e40bf);`

See also [encodeNEC()](https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_NEC.cpp#L47), it may help you.

It **_may_** be as simple as `sendNEC(encodeNEC(0x807E, 32));` for your initial example. That might be completely be wrong. I'm not sure about the device/subdevice/OBC thing. Sorry. Completely untested.

Bluebrain2000, 2018-11-04:
Thanks A LOT! This really helped me!
In fact, both works. The code value calculation with gc_decode but also (the much nicer way) with encodeNEC.

gc_decode gave me the hint that the device and subdevice address needs to be swapped.
So instead of 0x807E, it's 0x7E80. (even though the manufactor writes the address as 0x807e)

sendNEC(encodeNEC(0x[subdevice][device], [function]));
e.g. sendNEC(encodeNEC(0x7E80, 32));

As I got all codes in the form of address+function, this makes it now much easier to implement instead of having to convert every single code before.

closed 2018-11-04 by Bluebrain2000