crankyoldgit/IRremoteESP8266 #569: GREE Power mode reported as off, when the remote sends ON

jimmys01, 2018-11-10:
### Version/revison of the library used
2.5.2

### Expected behavior
IR DUMP v2 reports from a GREE remote report 

### Actual behavior
Recorded messages from a CREE remote and the message description regarding the power is reporting falsly. It always reports OFF. Examples from the capture below

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
**TRUE ON**
Encoding  : GREE
Code      : 0C0E205014200070 (64 bits)
Mesg Desc.: **Power: Off**, Mode: 4 (HEAT), Temp: 30C, Fan: 0 (AUTO), Turbo: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Manual, Swing Vertical Pos: 4
uint16_t rawData[139] = {9324, 4454,  682, 522,  682, 520,  702, 1638,  682, 1634,  736, 498,  680, 522,  684, 524,  752, 494,  736, 500,  736, 1640,  704, 1660,  680, 1662,  680, 522,  680, 522,  702, 500,  680, 524,  718, 498,  684, 520,  702, 500,  678, 524,  678, 520,  680, 1662,  702, 500,  704, 498,  680, 522,  682, 520,  694, 508,  678, 522,  680, 1662,  704, 500,  680, 1662,  678, 496,  704, 524,  678, 1640,  702, 522,  682, 20658,  762, 500,  742, 498,  718, 1640,  742, 524,  684, 1660,  682, 522,  704, 498,  734, 470,  766, 500,  714, 522,  742, 502,  736, 474,  762, 500,  740, 1638,  704, 498,  704, 500,  704, 498,  704, 500,  706, 498,  682, 522,  704, 498,  706, 500,  682, 550,  722, 494,  738, 502,  740, 498,  738, 500,  742, 498,  740, 1614,  714, 1656,  714, 1640,  740, 500,  740};  // GREE
uint8_t state[8] = {0x0C, 0x0E, 0x20, 0x50, 0x14, 0x20, 0x00, 0x70};

**TRUE ON**
Encoding  : GREE
Code      : 0C0D205014200060 (64 bits)
Mesg Desc.: **Power: Off**, Mode: 4 (HEAT), Temp: 29C, Fan: 0 (AUTO), Turbo: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Manual, Swing Vertical Pos: 4
uint16_t rawData[139] = {9240, 4426,  702, 522,  676, 522,  676, 1660,  674, 1636,  704, 520,  674, 522,  674, 522,  698, 498,  688, 1650,  698, 498,  676, 1660,  674, 1638,  700, 522,  676, 522,  674, 522,  676, 524,  698, 498,  676, 522,  698, 498,  676, 522,  676, 522,  674, 1662,  676, 522,  698, 500,  674, 522,  700, 498,  698, 500,  676, 522,  676, 1636,  702, 522,  676, 1634,  702, 496,  702, 522,  676, 1636,  700, 520,  676, 20658,  698, 522,  676, 522,  674, 1664,  674, 522,  698, 1614,  700, 524,  674, 522,  698, 500,  698, 500,  676, 520,  698, 498,  676, 522,  674, 522,  688, 1652,  698, 498,  676, 522,  674, 524,  700, 498,  674, 496,  702, 522,  676, 522,  674, 522,  676, 522,  676, 522,  676, 522,  676, 522,  674, 522,  698, 500,  676, 522,  674, 1662,  676, 1660,  676, 522,  674};  // GREE
uint8_t state[8] = {0x0C, 0x0D, 0x20, 0x50, 0x14, 0x20, 0x00, 0x60};

**TRUE OFF**
Encoding  : GREE
Code      : 040E2050142000F0 (64 bits)
Mesg Desc.: **Power: Off**, Mode: 4 (HEAT), Temp: 30C, Fan: 0 (AUTO), Turbo: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Manual, Swing Vertical Pos: 4
uint16_t rawData[139] = {9306, 4430,  706, 522,  680, 524,  680, 1638,  726, 502,  706, 498,  704, 498,  684, 496,  734, 500,  708, 496,  686, 1636,  732, 1644,  680, 1666,  702, 498,  680, 522,  680, 522,  680, 522,  706, 498,  680, 522,  682, 522,  678, 522,  704, 474,  704, 1660,  702, 500,  680, 522,  682, 522,  680, 522,  680, 520,  702, 500,  680, 1660,  706, 498,  684, 1638,  712, 524,  680, 522,  710, 1638,  704, 498,  680, 20684,  678, 524,  680, 522,  704, 1614,  710, 522,  680, 1662,  704, 498,  680, 522,  680, 522,  680, 522,  704, 500,  680, 522,  678, 522,  678, 498,  704, 1660,  684, 520,  684, 518,  706, 500,  710, 498,  678, 524,  682, 522,  682, 522,  704, 500,  682, 522,  682, 522,  682, 522,  680, 522,  704, 498,  686, 522,  704, 1636,  706, 1612,  708, 1636,  730, 1638,  708};  // GREE
uint8_t state[8] = {0x04, 0x0E, 0x20, 0x50, 0x14, 0x20, 0x00, 0xF0};

**TRUE OFF**
Encoding  : GREE
Code      : 040C2050142000D0 (64 bits)
Mesg Desc.: **Power: Off**, Mode: 4 (HEAT), Temp: 28C, Fan: 0 (AUTO), Turbo: Off, XFan: Off, Light: On, Sleep: Off, Swing Vertical Mode: Manual, Swing Vertical Pos: 4
uint16_t rawData[139] = {9354, 4450,  714, 522,  686, 524,  686, 1662,  712, 498,  716, 496,  686, 522,  690, 520,  686, 522,  688, 522,  710, 500,  684, 1636,  738, 1636,  708, 500,  686, 522,  688, 522,  688, 522,  686, 522,  708, 500,  710, 500,  686, 522,  686, 498,  734, 1612,  736, 500,  686, 522,  708, 498,  708, 500,  686, 522,  686, 522,  686, 1660,  686, 522,  710, 1614,  710, 522,  708, 500,  708, 1640,  708, 500,  686, 20656,  734, 500,  686, 522,  686, 1662,  708, 500,  710, 1640,  684, 496,  736, 500,  708, 474,  710, 522,  686, 522,  684, 522,  690, 518,  686, 522,  708, 1638,  686, 524,  686, 522,  708, 498,  684, 524,  712, 500,  686, 522,  686, 524,  710, 472,  738, 496,  710, 500,  688, 522,  686, 522,  712, 498,  686, 524,  710, 1638,  712, 498,  686, 1662,  712, 1636,  688};  // GREE
uint8_t state[8] = {0x04, 0x0C, 0x20, 0x50, 0x14, 0x20, 0x00, 0xD0};

crankyoldgit, 2018-11-10:
Hi. I've had a look at the data and the source code. It appears your A/C unit is a different model to the one we've got code for. Looking at other implementations available on the Internet, there seems to be several subtle differences between codes used by Gree models.

So, a couple of questions for you:

1) Can you tell us what make & model numbers are on the head unit *and* the remote?
2) Is everything else reported in `Mesg Desc.:` correct for your model? If not, what else is wrong?

I may need you to collect some more samples for us later, to help us identify something in the message that enables us to identify your model vs. the current one.

crankyoldgit, 2018-11-14:
@jimmys01 That last message does look like a Kelvinator A/C message. No idea why your remote produced that, but given it has checksums etc, it's highly likely it is what it says it is.

Let us know when you've tracked down the other remote.

jimmys01, 2018-11-16:
I have tried more than 10 remotes so far I cant find this one outlier

crankyoldgit, 2018-11-16:
Bummer. Perhaps we should close this issue, and re-open if you can get it to reproduce the same problem.
If 10+ remotes and the library are behaving like we expect, then I think we can be fairly sure library is working okay.