crankyoldgit/IRremoteESP8266 #573: Fujitsu A/C Remote AR-REB1E

sidwin9, 2018-11-20:
Is there a hope to have an evolution to integrate the code of the remote control AR-REB1E for Futjisu air conditioner?

sidwin9, 2018-11-22:
**Hello,
Here is a code for Arduino that works for an AR-RBE1 remote controlDespite an adaptation attempt for ESP8266 it does not work. The code contains all the information on the frames emitted by the remote control. Thanks for your help.**

#include <OneWire.h>
#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"
#include "IRremote.h"

#define WORD_TEMP 9
#define WORD_MODE 10
#define WORD_FAN  11
#define WORD_CHECKSUM 16

#define MODE_AUTO 0
#define MODE_COOL 1
#define MODE_DRY  2
#define MODE_FAN  3
#define MODE_HEAT 4

#define FAN_AUTO   0
#define FAN_HIGH   1
#define FAN_MEDIUM 2
#define FAN_LOW    3
#define FAN_QUIET  4

#define FIRST_SPACE 3200
#define FIRST_MARK 1600
#define ONE_IN_MS 1183
#define ZERO_IN_MS 350
#define SPACE_IN_MS 445

//---------------

IRsend irsend;
unsigned char currentTemp;
unsigned char currentMode;
unsigned char currentFan;

unsigned char mode;
unsigned char temp;
unsigned char fan; 

unsigned char trame[16] = {0x28, 0xc6, 0x0, 0x8, 0x8, 0x7f, 0x90, 0xc, 0x7, 0x20, 0x40, 0x0, 0x0, 0x0, 0x4, 0x53};
unsigned char trameOFF[7] = {0x28, 0xc6, 0x00, 0x08, 0x08, 0x40, 0xbf};

unsigned int times[259] = {FIRST_SPACE, FIRST_MARK, SPACE_IN_MS};   //259 = 16 words * (8 bytes + 8 spaces) +  first space + first mark + 1 space in ms
unsigned int timesOFF[115] = {FIRST_SPACE, FIRST_MARK, SPACE_IN_MS};  // 115 =  7 words * (8 bytes + 8 spaces) +  first space + first mark + 1 space in ms

uint32_t timer;

// Radio Hardware configuration
// Set up nRF24L01 radio on SPI bus plus pins 9 & 10

RF24 radio(9,10);

// Radio pipe addresses for the 2 nodes to communicate.
const uint64_t pipes[2] = { 0xF0F0F0F0E1LL, 0xF0F0F0F0D2LL };

int DS18S20_Pin = 2; //DS18S20 Signal pin on digital 2
//Temperature chip i/o
OneWire ds(DS18S20_Pin);  // on digital pin 2

unsigned long timerTempCheck = 0;
unsigned int roomTemp;

void setup() {               
 
 
[... 268 lines cut ...]
      }
      times[index++] = SPACE_IN_MS;
      mask = mask >> 1;
    }
  }
   
}

// Inverser l'ordre des bits dans un octet.
// I.e. MSB est échangé avec LSB, etc.
unsigned char Bit_Reverse( unsigned char x )
{
    x = ((x >> 1) & 0x55) | ((x << 1) & 0xaa);
    x = ((x >> 2) & 0x33) | ((x << 2) & 0xcc);
    x = ((x >> 4) & 0x0f) | ((x << 4) & 0xf0);
    return x;   
}

float getTemp(){
  //returns the temperature from one DS18S20 in DEG Celsius
	
	//j'ai enlevé le code pour que ca ne soit pas trop long

 
}

sidwin9, 2018-11-22:
Hello,

Cancel my last post, your library works for an AR-RBE1E remote control for Fujitsu A / C, I used in my circuit an NPN 2222 transistor for my IR LED.

Thanks for your help.

crankyoldgit, 2018-11-22:
Okay. Closing this issue as it seems resolved now.

closed 2018-11-22 by crankyoldgit