crankyoldgit/IRremoteESP8266 #574: Possible attack vector in IRMQTTServer.ino example

gipnokote, 2018-11-21:
What does prevent an attacker from doing this from the street?:

1. Jam the router WiFi signal
2. ESP8266 goes into WiFiManager mode and becomes an AP, thus being available to the attacker
3. Connect the ESP to his own access point
4. Via browser upload a modified firmware which would report the WiFi access details to the attacker's remote server, or even implements any kind of backdoor
5. A network owner wakes up and realizes that ESP became disconnected, reconnects it to his router
6. A modified firmware reports original WiFi details and/or opens a backdoor
7. Pwned!

An OTA password in the code and better yet, disabling OTA by default, is a way to mitigate this.

closed 2018-11-23 by crankyoldgit

gipnokote, 2018-11-23:
Hi David! Thank you for the reply, it's great to see that this code is being improved!
Although it's an example code and I was realizing it from the beginning, it still works great for me as something that I can use on my ESP8266 connected to a Home Assistant instance with almost no changes. That's why I was concerned about improving it's security :)

crankyoldgit, 2018-11-23:
Glad you like it and that you find it useful.

FYI, v2.5.3 of the library has just been released, and it includes the changes discussed here.

PR #575 (2018-11-23): Improve security options for IRMQTTServer example.
Security improvements:
- Disabling of OTA firmware updates. (default off)
- Basic HTTP Auth for OTA firmware updates. (default on) (tested)
- Basic HTTP Auth for general HTML interface. (default off) (tested)

Improve documentation:
- Add security text making sure users are informed.
- Make clear where users need to change things for their use.

Fixes #574
commit 5c1c85f5 (2018-11-23): Improve security options for IRMQTTServer example. (#575)

* Improve security options for IRMQTTServer example.
Security improvements:
- Disabling of OTA firmware updates. (default off)
- Basic HTTP Auth for OTA firmware updates. (default on) (tested)
- Basic HTTP Auth for general HTML interface. (default off) (tested)

Improve documentation:
- Add security text making sure users are informed.
- Make clear where users need to change things for their use.

Fixes #574