crankyoldgit/IRremoteESP8266 #578: Need help for IRremoteESP8266 optimization

snr-lab, 2018-11-26:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revison of the library used v2.5.3

Thank you very much for IRremoteESP8266 library.

Because there are lots of other components I need to optimize the code as much as possible. For my project I only need to decode NEC. From the release note I came to know that this library is capable of "optionally chose which protocols are included". But I could not figure out how to do that. Can you please help me regarding that? A very basic example would be much appreciated.

crankyoldgit, 2018-11-26:
See https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L53
Basically disable what ever decoders that you don't need.

closed 2018-11-26 by crankyoldgit

snr-lab, 2019-01-08:
Thank you very much for your response.