crankyoldgit/IRremoteESP8266 #579: Coolix variant (?) - sendRaw works, but not sending a hex code

abstrakct, 2018-11-28:
### Version/revison of the library used
2.5.3

### Expected behavior
1. Read power off code from remote with IR receiver connected to nodeMCU board
2. Send same code with IR diode (connected via transistor, as per schematic in wiki)
3. Device should power off when I point the diode at the device and the code is transmitted.

### Actual behavior
Background:
I'm new to coding IR stuff, but I've been a hobby programmer for many years. I've been toying with esp8266 and arduinos for about a year.
I rent an apartment with an older AC/heat pump unit, the brand is NEX, but I can't find any info about the brand online. I'm trying to make a little wifi-connected remote to control this unit.

I uploaded the IRrecvDumpV2 program to a nodeMCU board with an IR receiver diode connected. The program recognized the remote control codes as the Coolix type/protocol, and the codes read from the remote seem to closely match the Coolix codes.
However, when I try to use the library to send codes to my unit, nothing happens. I've tried many variations, with no luck.
First I tried variations of this sequence:
```
ac.begin();
ac.setPower(true);
ac.setTemp(20);
ac.setMode(kCoolixHeat);
ac.setFan(kCoolixFanAuto);
ac.send();
```
No luck.
I also tried things like `irsend.sendCOOLIX(0xB2BF2C);`  (B2BF2C is a code read from the remote control with the IRrecvDumpV2 program.)
No luck.
(Btw, looking at the diode with cell phone camera I can see the diode lighting up, so something is being sent.)

BUT! When I send the rawData captured from the remote it works perfectly! The unit turns on/off just as expected!
So there has to be a way to make this work, hopefully without capturing/sending rawData for all combinations of settings. With 5 modes, 14 different temperatures and 4 different fan settings it'll be rather inconvenient to only deal with rawData... Probably doable, but I'd rather use this nice library!

I suspect my unit uses a slightly different version of the Coolix protocol? Or there is some data sent by my remote that isn't captured by IRrecvDumpV2? I've tried playing around with the kTimeout and kMinUnknownSize values, but no change in what IRrecvDumpV2 outputs.
My unit only lets me select Auto fan speed when the mode is Auto or Dry. I discovered that in those cases the library doesn't recognize the Auto fan speed. See output further down.
Also note that the power off command in the library (0xB27BE0) matches what is read from my remote, but sending the code still doesn't work.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Power off command:
```
Timestamp : 000039.489
Encoding  : COOLIX
Code      : B27BE0 (24 bits)
Mesg Desc.: Power: Off
Library   : v2.5.3

Raw Timing[199]:
   +  4526, -  4412,    +   594, -  1600,    +   624, -   498,    +   624, -  1568, 
   +   594, -  1600,    +   610, -   514,    +   594, -   524,    +   610, -  1556, 
   +   622, -   500,    +   624, -   496,    +   622, -  1572,    +   624, -   498, 
   +   622, -   500,    +   622, -  1570,    +   592, -  1580,    +   612, -   508, 
   +   698, -  1490,    +   622, -   526,    +   588, -  1578,    +   622, -  1570, 
   +   624, -  1568,    +   592, -  1576,    +   614, -   532,    +   676, -  1488, 
   +   626, -  1568,    +   624, -  1568,    +   588, -   534,    +   592, -   530, 
   +   566, -   554,    +   592, -   528,    +   588, -  1582,    +   702, -   442, 
   +   644, -   528,    +   610, -  1584,    +   608, -  1560,    +   618, -  1572, 
   +   624, -   496,    +   624, -   498,    +   624, -   496,    +   624, -   496, 
   +   626, -   496,    +   624, -   496,    +   626, -   496,    +   622, -   500, 
   +   626, -  1566,    +   626, -  1566,    +   610, -  1558,    +   702, -  1488, 
   +   626, -  1566,    +   626, -  4678,    +  4540, -  4410,    +   594, -  1600, 
   +   626, -   496,    +   626, -  1566,    +   594, -  1602,    +   590, -   506, 
   +   702, -   446,    +   590, -  1574,    +   626, -   496,    +   626, -   496, 
   +   626, -  1566,    +   624, -   496,    +   624, -   498,    +   626, -  1568, 
   +   624, -  1568,    +   610, -   512,    +   610, -  1584,    +   590, -   532, 
   +   592, -  1600,    +   590, -  1576,    +   626, -  1566,    +   624, -  1568, 
   +   592, -   528,    +   594, -  1576,    +   702, -  1488,    +   626, -  1568, 
   +   624, -   498,    +   624, -   498,    +   624, -   496,    +   626, -   496, 
   +   624, -  1568,    +   610, -   510,    +   594, -   528,    +   592, -  1600, 
   +   610, -  1582,    +   590, -  1576,    +   618, -   528,    +   594, -   526, 
   +   610, -   512,    +   594, -   528,    +   594, -   528,    +   610, -   510, 
   +   594, -   528,    +   594, -   528,    +   608, -  1586,    +   590, -  1576, 
[... 134 lines cut ...]
kBitMark   = 660
kOneSpace  = 1520
kZeroSpace = 458
kSpaceGap = 4654

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+101100100100110110111111010000000110110010010011GAP(4654)Traceback (most recent call last):
  File "./auto_analyse_raw_data.py", line 432, in <module>
    main()
  File "./auto_analyse_raw_data.py", line 428, in main
    parse_and_report(data, arg_options.margin, arg_options.gen_code)
  File "./auto_analyse_raw_data.py", line 247, in parse_and_report
    total_bits = decode_data(message, defines, function_code, output)
  File "./auto_analyse_raw_data.py", line 315, in decode_data
    message.display_binary(binary_value)
  File "./auto_analyse_raw_data.py", line 88, in display_binary
    (bits, "0x{0:0{1}X}".format(num, bits / 4),
ValueError: Precision not allowed in integer format specifier
```
I don't have a lot of experience with python, and didn't figure out how to fix this error.

If someone can help me decode the raw data (or get the auto analyse script to work), I wouldn't mind writing the code to support my unit. If the unit is too obscure to be included in the official library I can of course make my own fork.

Thanks in advance for any help!

crankyoldgit, 2018-11-28:
FWIW, `auto_analyse_raw_data.py` appears to be working for me.
e.g.
```
IRremoteESP8266/tools$ cat /tmp/j
uint16_t powerOffRawData[199] = {4444, 4434,  590, 1578,  698, 446,  590, 1578,  622, 1596,  622, 500,  644, 476,  644, 1548,  588, 532,  594, 530,  612, 1578,  590, 532,  588, 534,  672, 1518,  594, 1598,  590, 510,  612, 1580,  644, 480,  612, 1578,  644, 1548,  644, 1548,  594, 1598,  642, 506,  644, 1550,  644, 1548,  594, 1600,  644, 478,  644, 478,  642, 480,  644, 478,  642, 1548,  594, 530,  590, 532,  614, 1578,  644, 1548,  594, 1600,  588, 534,  566, 556,  588, 530,  590, 532,  586, 514,  612, 532,  588, 532,  590, 534,  588, 1578,  642, 1576,  642, 1550,  588, 1602,  588, 1580,  642, 4712,  4546, 4406,  588, 1606,  642, 478,  644, 1550,  590, 1604,  588, 534,  586, 532,  586, 1582,  642, 480,  642, 480,  668, 1550,  642, 480,  642, 478,  642, 1552,  612, 1578,  586, 538,  588, 1580,  674, 472,  590, 1602,  586, 1580,  618, 1576,  642, 1548,  594, 530,  590, 1584,  608, 1578,  644, 1550,  642, 480,  642, 478,  642, 480,  642, 480,  642, 1550,  590, 530,  592, 528,  592, 1602,  642, 1548,  592, 1604,  586, 584,  642, 480,  640, 480,  640, 480,  642, 480,  642, 480,  642, 480,  642, 480,  642, 1552,  590, 1604,  588, 1578,  642, 1552,  640, 1550,  592};  // COOLIX B27BE0
IRremoteESP8266/tools$ ./auto_analyse_raw_data.py -f /tmp/j
Found 199 timing entries.
Potential Mark Candidates:
[4546, 698]
Potential Space Candidates:
[4712, 4434, 1606, 584]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4495
kHdrSpace  = 4420
kBitMark   = 618
kOneSpace  = 1572
kZeroSpace = 504
kSpaceGap = 4712

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+101100100100110101111011100001001110000000011111GAP(4712)
  Bits: 48
  Hex:  0xB24D7B84E01F (MSB first)
        0xF80721DEB24D (LSB first)
  Dec:  196045854531615 (MSB first)
        272709516702285 (LSB first)
  Bin:  0b101100100100110101111011100001001110000000011111 (MSB first)
        0b111110000000011100100001110111101011001001001101 (LSB first)
kHdrMark+kHdrSpace+101100100100110101111011100001001110000000011111
  Bits: 48
  Hex:  0xB24D7B84E01F (MSB first)
        0xF80721DEB24D (LSB first)
  Dec:  196045854531615 (MSB first)
        272709516702285 (LSB first)
  Bin:  0b101100100100110101111011100001001110000000011111 (MSB first)
        0b111110000000011100100001110111101011001001001101 (LSB first)

Total Nr. of suspected bits: 96
```

That's the raw data line from your code example. If it is breaking on a different input, can you please supply it me and I can see what might be causing it.

Now, looking at that output, yes, it is two messages back to back, with the same payload. i.e. message + repeat.

abstrakct, 2018-11-28:
Thanks for the help and feedback!
I figured out the problem with the auto analyse script - I was running it on python 3.7! Running it on python 2.7 works. Sorry about that. 

Unfortunately `irsend.sendCOOLIX(0xB2BF2C, 1);` didn't work.
I even tried 2, 3, 4 and 5 repeats, but still nothing.
Where do we go from here? I can of course provide more raw data from the remote if that's helpful.
I'll keep looking at it myself and see if I figure something out.

abstrakct, 2018-11-28:
I figured it out! I needed to do `irsend.sendCOOLIX(0xB2BF2C, kCoolixBits, 1)` because repeat is the third parameter! It works!!!

abstrakct, 2018-11-28:
Another thing: `void IRCoolixAC::send()` currently doesn't support setting a repeat. I guess this could be solved by overloading the method with a `void IRCoolixAC::send(uint16_t repeat)` method - or is there another preferred way to do it? Just changing the current method to `void IRCoolixAC::send(uint16_t repeat = 0)` ?

crankyoldgit, 2018-11-29:
@abstrakct Sorry for giving you the incorrect code example. i.e. I forgot to add the bit length.
You correctly figured it out thankfully.

I wanted to wait till you confirmed the repeat solution worked for you before I coded a solution for the class method.

I've just pushed a change to that branch that should make the class approach work for you.
e.g.
```
ac.begin();
ac.setPower(true);
ac.setTemp(20);
ac.setMode(kCoolixHeat);
ac.setFan(kCoolixFanAuto);
ac.send();  // Send with a single repeat. i.e. two messages back-to-back
```

Let me know if that works please.

I'll look into why Python3.7 has an issue with the decoder.

crankyoldgit, 2018-11-29:
@Hamper & @Hollako, Can you two please see if the Raw messages (via IRrecvDumpV2) from your Coolix remotes look like the ones here? i.e. approx 199 elements long vs. approx 98 elements long.
Also, can you test/confirm if you use `sendCOOLIX()` with a repeat, that it works as expect with your A/C units?
e.g.
```
irsend.sendCOOLIX(code_to_send, kCoolixBits, 1);
```

I'm asking as I'd like to investigate if we should change the default to be a repeat of 1. i.e. would that change break things for other Coolix A/C users.

Xref #484

crankyoldgit, 2018-11-29:
@abstrakct Thanks for the confirmation that the patch works.

PR #580 (2018-11-29): Handle send() repeats in A/C classes. Improve Coolix support.
Issue #579 reported a variant of the Coolix remote that uses a second value for fan auto and has a repeat.
* Add support for alt auto fan speed for coolix.
  * Add unit tests to cover real example and 2nd auto value.
* Add support for optionally specifying the number of times a message should be repeated when sent via the A/C's specific `::send()` method.

Fixes #579
commit 13635eed (2018-11-29): Handle send() repeats in A/C classes. Improve Coolix support. (#580)

Issue #579 reported a variant of the Coolix remote that uses a second value for fan auto and has a repeat.
* Add support for alt auto fan speed for coolix.
  * Add unit tests to cover real example and 2nd auto value.
* Add support for optionally specifying the number of times a message should be repeated when sent via the A/C's specific `::send()` method.

Fixes #579
closed 2018-11-29 by crankyoldgit