crankyoldgit/IRremoteESP8266 #584: GPIO16 problem

lucasromeiro, 2018-12-06:
Hello, this library works great and I have been using it for a long time. It always worked.
I use version 2.4.2 on esp8266.
But I was trying to use the GPIO16 pin to send an IR command and it does not work!
It works on the other pins I tested, but that does not work!

Is there any limitation on this pin or on others?

I'm not using the most current version of the library for fear of some incompatibility, because the version I'm using works fine.

crankyoldgit, 2018-12-06:
A quick Google search have me the answer.
GPIO16 is often connected to RESET on most boards. It's done that way to wake it from deep sleep. Sounds like your board is one of them.
There are lots of other limitations on some pins.

crankyoldgit, 2018-12-06:
See https://bbs.espressif.com/viewtopic.php?t=632

closed 2018-12-06 by crankyoldgit