crankyoldgit/IRremoteESP8266 #585: Reverse lookup of AC codes

mihalski, 2018-12-12:
Hi,

Thank you for the awesome library!
Is it possible to do reverse lookup of received AC codes for display in human readable format?

Regards,
Michal

crankyoldgit, 2018-12-12:
See the [IRrecvDumpV2](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L117) example code for how to do that.

***IF*** the library has support deep support for the particular model of A/C unit remote you have, then yes it can. You need to instantiate the relevant AC class, populate the class with the state/code you got via the `.setRaw()` method. Then use the `.toString()` method to convert the state/code into human readable string.
e.g.
```
// Assumes `state` is a Kelvinator AC state you've already captured.
IRKelvinatorAC ac(0);
ac.setRaw(state);
printf("The captured message was: %s\n", ac.toString());
```

mihalski, 2018-12-12:
Working great for my bedroom Mitsubishi (most importantly) but sadly not for some other Mitsubishi's and a Toshiba units in the house. Will look into all that but for now I'm just very happy that this works.

crankyoldgit, 2018-12-12:
Glad it's working for one of your units. Alas, not all A/Cs are supported. They typically take a lot of effort to reverse engineer. Try looking through old closed issues for how to do it if you are interested in volunteering to help decode new makes and models.

closed 2018-12-12 by crankyoldgit

mihalski, 2018-12-12:
I had plans on doing just this a couple of years ago and in a few months I may actually have time. I already own AnalysIR and have LearnIR, DetectIR and SendIR modules.

crankyoldgit, 2018-12-12:
I look forward to a future issue/PR with your efforts! :) Good luck.