crankyoldgit/IRremoteESP8266 #586: Question/ Request

rasp111, 2018-12-17:
Hi
First of all Thanks for the wonderful work done by you guys.
The code works amazingly well for me & is very consistent.
I have configured Openhab to publish an MQTT command to an ESP that is acting as MQTT IR BROKER..
Whenever I publish an MQTT message to an switch (say TV ON), the ESP publishes the relevant IR_code on the "ir_server/send" Topic. 
The ESP contains all the IR codes i need in my case stored & i call the relevant codes and publish on the "ir_server/send" Topic.
example:
IR Code to turn ON/ OFF Samsung TV is stored as -> const char* C1 = "7,E0E040BF,32";      
I have the ESP publish channel configured to "ir_server/sent" topic using:
if (String(MQTT_V1_COMMAND_TOPIC).equals(p_topic)) {Serial.println(payload);
    if (payload.equals(String(ACTION_ON))) {V1();}  
char IRT[15]= "ir_server/send";
void V1() {client.publish(IRT,C1);}     // SAMSUNG TV ON/OFF

This succesfully turns on the Samsung TV.

Now that i want to move my code also into the same ESP that is doing IRMQTTSERVER, i want help to avoid client.publish(IRT,C1); and some 
how format into appropriate format and call the relevant function that already exists like callback(char* topic, byte* payload, unsigned int length).
I am pretty unskilled at programing & hence looking forward for some help to correctly format the data to be sent as payload while trying to use the callback() function:

void V1() {callback(..., ....);}     // SAMSUNG TV ON/OFF
 
Regards
Ravi

crankyoldgit, 2018-12-18:
Hey Ravi,
 I think I understand what you want to do. It seems a little confusing to me, so apologies if I am not understanding it correctly.

Before we get into the appropriate calls to make, I'd like to suggest you think about having Openhab make the appropriate MQTT requests directly to the IRMQTTServer example code directly. It may cut out a middle-man for you. I'm not familiar with openhab, but it'd be very likely it can produce an appropriate MQTT topic & message for what you want to do.

Anyway, on to your request.

If I understand it correctly: You want to call _something_ internal to the IRMQTTServer code with a string like `"7,E0E040BF,32"` so that it transmits the IR message. Like it would have done if received via MQTT etc, but you don't want to send/resend this message via MQTT in the first place. Correct?

The _least programming effort_ would be to use the following procedure:
[void receivingMQTT(String const topic_name, String const callback_str)](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L1636)

e.g.
```
receivingMQTT("topic_name only used in debug message", String(C1));  // Using your C1 defined earlier.
```
Note: That will produce behaviour like it would **if** it had received it via MQTT directly. i.e. debug messages and a response/reply on the 'sent' topic.

A much more cleaner way (i.e. no inappropriate MQTT messages etc being sent/reported etc) would be to use the following function:
[bool sendIRCode(int const ir_type, uint64_t const code, char const * code_str, uint16_t bits, uint16_t repeat)](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L1327)

e.g.
```
sendIRCode(7,              // First part of "7,E0E040BF,32"
           0xE0E040BF,     // Second part of  "7,E0E040BF,32"
           "E0E040BF,32",  // Everything but the first part of  "7,E0E040BF,32", but not needed really for this (7) protocol. Only for aircons etc.
          32,              // Third part of  "7,E0E040BF,32"
          0                // Fourth part of  "7,E0E040BF,32", which there isn't so 0.
          );
```
But that requires you to split up and convert the string &/or not store values as a simple string etc.

rasp111, 2018-12-19:
Thanks for quick support...
It worked for me.

Regards
Ravi