crankyoldgit/IRremoteESP8266 #588: Pushing the data out to other devices

everythingdaniel, 2018-12-26:
Hi, I am running version 2.5.3 on a Node-MCU. I am newish to this hardware as well as this type of programming. 

My complete project would be the Node-MCU will send a UDP sting out to another device then it gets something, and then when it receives a UDP string it'll fire off a IR signal.

Right now, I am trying to get the IR signal out to the other device, my code will be below. My UDP routines are confirmed working, I can send any string I type manually. I can also see the IR incoming data...I am not having success combining the two.

```
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

const char *ssid = "Network";
const char *password = "Password";
const char *nrip = "192.168.0.110";

WiFiUDP Udp;

// an IR detector/demodulator is connected to GPIO pin 2
uint16_t RECV_PIN = 2;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup() {
    Serial.begin(115200);
    Serial.println();

    Serial.printf("Connecting to %s ", ssid);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
    delay(500);
    Serial.print(".");
    }
    Serial.println(" connected");
   irrecv.enableIRIn();  // Start the receiver
}

void dump(decode_results *results) {
  // Dumps out the decode_results structure.
  // Call this after IRrecv::decode()
  uint16_t count = results->rawlen;
  if (results->decode_type == UNKNOWN) {
    Serial.print("Unknown encoding: ");
  } else if (results->decode_type == NEC) {
    Serial.print("Decoded NEC: ");
  } else if (results->decode_type == SONY) {
    Serial.print("Decoded SONY: ");
  } else if (results->decode_type == RC5) {
    Serial.print("Decoded RC5: ");
  } else if (results->decode_type == RC5X) {
    Serial.print("Decoded RC5X: ");
  } else if (results->decode_type == RC6) {
    Serial.print("Decoded RC6: ");
  } else if (results->decode_type == RCMM) {
    Serial.print("Decoded RCMM: ");
  } else if (results->decode_type == PANASONIC) {
    Serial.print("Decoded PANASONIC - Address: ");
    Serial.print(results->address, HEX);
    Serial.print(" Value: ");
  } else if (results->decode_type == LG) {
    Serial.print("Decoded LG: ");
  } else if (results->decode_type == JVC) {
    Serial.print("Decoded JVC: ");
  } else if (results->decode_type == AIWA_RC_T501) {
    Serial.print("Decoded AIWA RC T501: ");
  } else if (results->decode_type == WHYNTER) {
    Serial.print("Decoded Whynter: ");
  }
  serialPrintUint64(results->value, 16);
  Serial.print(" (");
  Serial.print(results->bits, DEC);
  Serial.println(" bits)");
  Serial.print("Raw (");
  Serial.print(count, DEC);
  Serial.print("): ");

  for (uint16_t i = 1; i < count; i++) {
    if (i % 100 == 0)
      yield();  // Preemptive yield every 100th entry to feed the WDT.
    if (i & 1) {
      Serial.print(results->rawbuf[i] * RAWTICK, DEC);
    } else {
      Serial.write('-');
      Serial.print((uint32_t) results->rawbuf[i] * RAWTICK, DEC);
    }
    Serial.print(" ");
  }
  Serial.println();
}

void loop() {
  if (irrecv.decode(&results)) {
    serialPrintUint64(results.value, 16);
    
    Udp.beginPacket(nrip, 1112);
    Udp.write(results.value);
    Udp.endPacket();
    yield();
    Serial.println("Send UDP...");
    dump(&results);
    irrecv.resume();  // Receive the next value
  }
}
```

I have been pretty much working off the example sketch which got me off to a good start, but the `Udp.write(results.value);` line is not sending anything useful. Wireshark says 1 character is being sent, and the receiving device picks up garbage. I figure `serialPrintUint64(results.value, 16);` is converting the data to HEX before it prints to the serial port? So by calling just the `results.value` isn't going to work? I found a couple examples on how to get the info to print out as serial data, and that works, but I can't wrap my head around how to get that data as a string to use elsewhere...such as sending via UDP.

Thank you for any guidance.

crankyoldgit, 2018-12-26:
Note: Your problem has nothing to do with the IRremoteESP8266 library really. You've got a problem/issue with WifiUDP, not us.

For the record, `results.value` will be a `uint64_t`, which has a `sizeof()` of 8 bytes.

According to https://www.arduino.cc/en/Reference/WiFiUDPWrite:
You are using the invocation of `WifiUDP.write(byte)`, hence it is only going to send a single byte of 8 byte long `uint64_t`/`results.value`

You either need to convert `results.value` into a buffer (alternative .write() option per ref) or call it 8 times of each **byte** of the value.

e.g.
```

WifiUDP.write((uint_8t)(results.value >> 56) & 0xFF));   // Top most 8 bits (64 - 8)
WifiUDP.write((uint_8t)(results.value >> 48) & 0xFF));   // Next 8 bits (64 - 8 - 8)
WifiUDP.write((uint_8t)(results.value >> 40) & 0xFF));   // Next 8 bits (64 - 8 - 8 - 8)
...
...
WifiUDP.write((uint_8t)(results.value >> 8) & 0xFF));   // 2nd last 8 bits
WifiUDP.write((uint_8t)(results.value & 0xFF));   // last 8 bits
```

crankyoldgit, 2018-12-26:
If you want to covert it to a string, use [uint64ToString(uint64_t input, uint8_t base)](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRutils.cpp#L47)

crankyoldgit, 2019-01-02:
@everythingdaniel Just chasing up. How did you go?