crankyoldgit/IRremoteESP8266 #593: Problems with scan.

GiampaMan, 2019-01-04:
Hi!

I'm a beginner.
I was trying to scan a Panasonic air conditioner code. 
First, buffer on library was so small, so I upgraded it to 700.
Problem is that when I press a key on my remote, always see on serial port same value, I though this could be a timing issue, so I upgraded timeout to 90ms. Still same problem.
I will appreciate any kind of help.
Have a nice day!

crankyoldgit, 2019-01-04:
What code are you using? What version of the library? What is the output from the program?
Without data and more information we can't help you much.

I would **strongly** recommend you use the latest version (2.5.4 at time of writing) of the library and the `IRrecvDumpV2` example code (i.e. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) that shipped with that version of the library.
It's had a buffer size of over 1000 for approx 18 months now, hence I'm guessing you may be using an old version of the software.

GiampaMan, 2019-01-04:
> What code are you using? What version of the library? What is the output from the program?
> Without data and more information we can't help you much.
>
> I would **strongly** recommend you use the [...]

Hi!
Thanks! 
I'm testing IRrecvDumpV2. Seems to be working.
What is the example code I need to send the code received on serial port? I want to send the code field, not in raw.

Thanks for all!

crankyoldgit, 2019-01-04:
https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino is the simplest example code I'd recommend for send a code.

The routine you need depends a lot on what type of message/code IRrecvDumpV2 reported to you. Without that I can't really say for sure.

GiampaMan, 2019-01-04:
> https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino is the simplest example code I'd recommend for send a code.
> 
> The routine you need depends a lot on what type of message/code IRrecvDumpV2 reported to you. Without that I can't really say for sure.

Ok, I saw it, but one question, should I specify protocol (For example: Sony, Panasonic, etc...) at time to send? Is there a way to don't specify it to make it more easyer? I told you this because I want to learn and send codes.

crankyoldgit, 2019-01-04:
Yes. You have to specify the protocol. They are almost all different.

GiampaMan, 2019-01-08:
> Yes. You have to specify the protocol. They are almost all different.

Hello! 
Sorry about late response. I'm going to try.
Thanks for your help!

crankyoldgit, 2019-01-09:
That question is answered in the FAQ: https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code

closed 2019-01-11 by crankyoldgit