crankyoldgit/IRremoteESP8266 #595: when I analyze with file auto_analyse_raw_data.py I got this with error in pycharm

BABITA0005, 2019-01-10:
when I analyze with file auto_analyse_raw_data.py I got this with error in pycharm

usage: scratch.py [-h] [-g] [-f FILE] [-r MARGIN] [--stdin] [rawdata]
scratch.py: error: one of the arguments rawdata -f/--file --stdin is required

_Originally posted by @BABITA0005 in https://github.com/markszabo/IRremoteESP8266/issues/482#issuecomment-452982221_

crankyoldgit, 2019-01-10:
> `one of the arguments rawdata -f/--file --stdin is required`

The error message tries to tell you what to do. You need to supply the `"uint16_t rawData[231] = {3066, 9020,...};"` **or** `--stdin` **or** `-f` **or** `--file` on the command-line as an argument to the program.

I suggest you use the `-h` (for help) argument and it will give you some more detailed hints.

e.g.
```
$ ./auto_analyse_raw_data.py -h
usage: auto_analyse_raw_data.py [-h] [-g] [-f FILE] [-r MARGIN] [--stdin]
                                [rawdata]

Read an IRremoteESP8266 rawData declaration and tries to analyse it.

positional arguments:
  rawdata               A rawData line from IRrecvDumpV2. e.g. 'uint16_t
                        rawbuf[37] = {7930, 3952, 494, 1482, 520, 1482, 494,
                        1508, 494, 520, 494, 1482, 494, 520, 494, 1482, 494,
                        1482, 494, 3978, 494, 520, 494, 520, 494, 520, 494,
                        520, 520, 520, 494, 520, 494, 520, 494, 520, 494};'
                        (default: None)

optional arguments:
  -h, --help            show this help message and exit
  -g, --code            Generate a C++ code outline to aid making an IRsend
                        function. (default: False)
  -f FILE, --file FILE  Read in a rawData line from the file. (default: None)
  -r MARGIN, --range MARGIN
                        Max number of micro-seconds difference between values
                        to consider it the same value. (default: 200)
  --stdin               Read in a rawData line from STDIN. (default: False)
```

BABITA0005, 2019-01-11:
Thank you sir...

closed 2019-01-11 by crankyoldgit

Nakul93, 2021-12-22:
@crankyoldgit When I run this script with rawdata argument I get the following error:
```
Traceback (most recent call last):
  File "/Users/nakulnair/Documents/Arduino/libraries/IRremoteESP8266-2.7.19/tools/auto_analyse_raw_data.py", line 305, in convert_rawdata
    results.append(int(timing))
ValueError: invalid literal for int() with base 10: 'rawdata'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/nakulnair/Documents/Arduino/libraries/IRremoteESP8266-2.7.19/tools/auto_analyse_raw_data.py", line 773, in <module>
    main()
  File "/Users/nakulnair/Documents/Arduino/libraries/IRremoteESP8266-2.7.19/tools/auto_analyse_raw_data.py", line 768, in main
    parse_and_report(raw_data, arg_options.margin,
  File "/Users/nakulnair/Documents/Arduino/libraries/IRremoteESP8266-2.7.19/tools/auto_analyse_raw_data.py", line 374, in parse_and_report
    rawdata = convert_rawdata(rawdata_str)
  File "/Users/nakulnair/Documents/Arduino/libraries/IRremoteESP8266-2.7.19/tools/auto_analyse_raw_data.py", line 307, in convert_rawdata
    raise ValueError(
ValueError: Raw Data contains a non-numeric value of 'rawdata'.
```

Can you pls guide how to run this script with an example

crankyoldgit, 2021-12-22:
@Nakul93 Most people can work it out.
But here are some examples:
```
david@ubuntu:~/git/IRremoteESP8266$ cat /tmp/j
uint16_t rawData[68] = {4480, 4480,  560, 1680,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 560,  560, 560,  560, 560,  560, 1680,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 560,  560, 560,  560, 560,  560, 1680,  560, 560,  560, 560,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 1680,  560, 560,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 1680,  560, 1680,  560, 560,  560};
david@ubuntu:~/git/IRremoteESP8266$ tools/auto_analyse_raw_data.py -f /tmp/j
Found 67 timing entries.
Potential Mark Candidates:
[4480, 560]
Potential Space Candidates:
[4480, 1680, 560]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4480
kHdrSpace  = 4480
kBitMark   = 560
kOneSpace  = 1680
kZeroSpace = 560

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11100000111000001001100101100110
  Bits: 32
  Hex:  0xE0E09966 (MSB first)
        0x66990707 (LSB first)
  Dec:  3772815718 (MSB first)
        1721304839 (LSB first)
  Bin:  0b11100000111000001001100101100110 (MSB first)
        0b01100110100110010000011100000111 (LSB first)

Total Nr. of suspected bits: 32
```
```
david@ubuntu:~/git/IRremoteESP8266$ cat /tmp/j | tools/auto_analyse_raw_data.py --stdin
Found 67 timing entries.
Potential Mark Candidates:
[4480, 560]
Potential Space Candidates:
[4480, 1680, 560]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4480
kHdrSpace  = 4480
kBitMark   = 560
kOneSpace  = 1680
kZeroSpace = 560

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11100000111000001001100101100110
  Bits: 32
  Hex:  0xE0E09966 (MSB first)
        0x66990707 (LSB first)
  Dec:  3772815718 (MSB first)
        1721304839 (LSB first)
  Bin:  0b11100000111000001001100101100110 (MSB first)
        0b01100110100110010000011100000111 (LSB first)

Total Nr. of suspected bits: 32
```
```
david@ubuntu:~/git/IRremoteESP8266$ tools/auto_analyse_raw_data.py "4480, 4480, 560, 1680, 560, 1680, 560, 1680, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1680, 560, 1680, 560, 1680, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1680, 560, 560, 560, 560, 560, 1680, 560, 1680, 560, 560, 560, 560, 560, 1680, 560, 560, 560, 1680, 560, 1680, 560, 560, 560, 560, 560, 1680, 560, 1680, 560, 560, 560"
Found 67 timing entries.
Potential Mark Candidates:
[4480, 560]
Potential Space Candidates:
[4480, 1680, 560]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4480
kHdrSpace  = 4480
kBitMark   = 560
kOneSpace  = 1680
kZeroSpace = 560

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11100000111000001001100101100110
  Bits: 32
  Hex:  0xE0E09966 (MSB first)
        0x66990707 (LSB first)
  Dec:  3772815718 (MSB first)
        1721304839 (LSB first)
  Bin:  0b11100000111000001001100101100110 (MSB first)
        0b01100110100110010000011100000111 (LSB first)

Total Nr. of suspected bits: 32
david@ubuntu:~/git/IRremoteESP8266$ tools/auto_analyse_raw_data.py "uint16_t rawData[68] = {4480, 4480,  560, 1680,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 560,  560, 560,  560, 560,  560, 1680,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 560,  560, 560,  560, 560,  560, 1680,  560, 560,  560, 560,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 1680,  560, 560,  560, 1680,  560, 1680,  560, 560,  560, 560,  560, 1680,  560, 1680,  560, 560,  560};"
Found 67 timing entries.
Potential Mark Candidates:
[4480, 560]
Potential Space Candidates:
[4480, 1680, 560]

Guessing encoding type:
Looks like it uses space encoding. Yay!

Guessing key value:
kHdrMark   = 4480
kHdrSpace  = 4480
kBitMark   = 560
kOneSpace  = 1680
kZeroSpace = 560

Decoding protocol based on analysis so far:

kHdrMark+kHdrSpace+11100000111000001001100101100110
  Bits: 32
  Hex:  0xE0E09966 (MSB first)
        0x66990707 (LSB first)
  Dec:  3772815718 (MSB first)
        1721304839 (LSB first)
  Bin:  0b11100000111000001001100101100110 (MSB first)
        0b01100110100110010000011100000111 (LSB first)

Total Nr. of suspected bits: 32
```