crankyoldgit/IRremoteESP8266 #602: Gree protocol compatible with Cooper&Hunter?

lexesv, 2019-01-14:
Hello!
I found that Mi Remote working with a Cooper&Hunter AC in "Gree mode".
https://play.google.com/store/apps/details?id=com.duokan.phone.remotecontroller

Please tell me, "Gree protocol" of the IRremoteESP8266  compatible with Cooper & Hunter?

crankyoldgit, 2019-01-14:
No idea. Try it out and see, and let us know how you go.

crankyoldgit, 2019-01-26:
Friendly chase up. How did you go?

lexesv, 2019-01-26:
I am waiting for delivery of the receiver module.

crankyoldgit, 2019-02-26:
Friendly chase up (again). How did you go?

lexesv, 2019-02-26:
Sorry. No time. 
If you let me reopen this issue after I can resolve it.

closed 2019-02-26 by lexesv

mcsa2003, 2019-08-19:
My cooper&hunter AC work with me remote as Midea
But I cant use RremoteESP8266 for control this AC - not decode, not working when I send command in RAW

crankyoldgit, 2019-08-19:
@mcsa2003 Please create a new issue for this and include the full output of `IRrecvDumpV2` in that issue.