crankyoldgit/IRremoteESP8266 #604: IRremoteESP8266 cannot switch off SAMSUNG AC

kalpakchiev, 2019-01-24:
Dear Sir,

So far I can set the temperature, fan speed, mode and turn it on, but I have never managed to switch Samsung AC off.

### Version/revison of the library used
IRremoteESP8266.h v2.5.4

### Actual behavior
In both cases it sends the code for on
For example: 
if the AC is off and I send  **ac.off();**  **ac.send();** it turns on.
if it is on and I send  **ac.off();**  **ac.send();** it just beeps.

Similar code works properly with a Toshiba AC

```
bool              ButtonState          =   false;
const int         buttonPin1           =   5;
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Samsung.h>
const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRSamsungAc ac(kIrLed);  // Set the GPIO to be used for sending messages.

void setup() {
  Serial.begin(115200);
  Serial.println(" ");
  pinMode(buttonPin1,     INPUT_PULLUP);
  Serial.print("READY : "); Serial.println(millis());
  ac.begin();
}

void loop() {
  ButtonState = digitalRead(buttonPin1);

    if (ButtonState == 0)
    {
      Serial.println("Switch it On : ");
      ac.on();
      ac.setFan(kSamsungAcFanAuto);
      ac.setMode(kSamsungAcHeat);
      ac.setTemp(28);
      Serial.println("AC ON, Fan AUTO, Heat mode, 28°C");
      ac.send();
    }
    else
    {
      Serial.println("Switch it Off : ");
      ac.off();
      Serial.println("AC OFF");
      ac.send();
    }
  }
}
```

crankyoldgit, 2019-01-24:
@kalpakchiev I believe you need to use `ac.sendExtended();` when doing _power_ changes. See https://github.com/markszabo/IRremoteESP8266/blob/master/examples/ControlSamsungAC/ControlSamsungAC.ino#L92 for an example.

If that doesn't work, please let me know and supply a IRrecvDumpV2 capture of a valid "off" signal from the actual remote that best matches the settings in your example. That way I can debug what might be the cause of the issue.

crankyoldgit, 2019-01-24:
And also see: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Samsung.cpp#L266

crankyoldgit, 2019-01-24:
FYI'ing @Hollako who also has a samsung a/c and used the code, so may also be able to help.

kalpakchiev, 2019-01-25:
Thanks for the prompt response.

Last night I tried  ac.sendExtended(); , but unfortunately with no success.
I ended using raw. With raw it works.

uint16_t SendOFF[349] = {642, 17730,  3056, 8916,  542, 448, 552, 1440, 552, 444,
                         552, 444,  552, 444,  552, 440,  556, 440,  556, 440,  556,
                         1436,  552, 444,  552, 444,  552, 1440,  548, 470,  526, 1464,
                         470, 526,  516, 1470,  552, 1440,  552, 1440,  550, 1436,
                         556, 1434,  552, 444,  552, 444,  552, 444,  552, 442, 552,
                         444,  546, 470,  526, 470,  526, 470,  470, 524,  518, 474,
                         548, 448,  552, 444,  552, 442,  552, 444,  550, 444,  552,
                         440,  556, 440,  556, 438,  556, 440,  552, 442,  552, 444,
                         552, 442,  552, 444,  550, 470,  526, 466,  524, 470,  470,
                         524,  470, 524,  518, 476,  548, 444,  552, 444,  556, 440,
                         552, 442,  552, 444,  550, 1436,  556, 1436,  552, 2946,
                         3026, 8918,  550, 1440,  552, 444,  548, 468,  526, 468,
                         470, 526,  470, 526,  542, 452,  548, 444,  552, 1440,  550,
                         444,  552, 444,  552, 1436,  556, 438,  552, 442,  552, 1440,
                         552, 1440,  552, 1460,  526, 1464,  470, 1516,  548, 1444, 552,
                         444,  552, 442,  552, 444,  552, 438,  556, 440,  556, 440, 552,
                         444,  552, 444,  552, 444,  552, 444,  552, 444,  548, 448,  546,
                         470,  526, 468,  526, 470,  470, 524,  520, 470,  548, 448,  552,
                         444,  552, 444,  552, 444,  552, 444,  552, 438,  556, 440,  556,
                         438,  552, 444,  552, 442,  552, 444,  552, 444,  552, 444,  552,
                         470,  526, 466,  526, 470,  470, 524,  518, 478,  546, 448,  552,
                         2920,  3052, 8916,  552, 1434,  556, 440,  556, 438,  552, 444,
                         552, 442,  552, 442,  552, 442,  552, 444,  548, 1444,  548, 470,
                         526, 470,  522, 1466,  470, 1520,  548, 1438,  556, 1436,  552,
                         1440,  552, 442,  552, 1436,  552, 1440,  552, 1440,  552, 442,
                         552, 470,  522, 1466,  526, 1466,  470, 1516,  552, 444,  552, 442,
                         552, 444,  552, 1436,  556, 1436,  552, 1440,  550, 444,  552, 444,
                         552, 444,  548, 448,  546, 448,  548, 470,  526, 1462,  474, 1518,
                         548, 1440,  552, 1438,  556, 440,  550, 444,  552, 444,  552, 444,
                         552, 440,  556, 1436,  552, 444,  552, 444,  552, 444,  550, 470,
                         522, 470,  524, 470,  470, 524,  518, 1474,  548, 1440,  556
                        };

                        
uint16_t SendON[349] = {638, 17808,  3056, 8912,  552, 444,  552, 1440,  552, 444,
                        552, 440,  556, 440,  556, 440,  556, 440,  552, 444,  552, 1440,
                        552, 444,  552, 442,  552, 1462,  526, 470,  526, 470,  472, 524,
                        544, 1444,  552, 1440,  552, 1434,  556, 1434,  552, 1440,  552,
                        442,  552, 444,  550, 444,  552, 442,  552, 444,  548, 448,  546,
                        468,  526, 468,  526, 468,  470, 524,  544, 452,  548, 444,  552,
                        444,  550, 444,  552, 442,  552, 442,  552, 444,  552, 440,  556,
                        440,  556, 438,  556, 438,  552, 444,  552, 448,  546, 448,  546,
                        470,  526, 470,  470, 524,  470, 520,  522, 474,  548, 448,  552,
                        444,  552, 444,  552, 1440,  552, 1434,  556, 1436,  556, 1436,
                        550, 2944,  3028, 8922,  546, 1440,  552, 468,  526, 470,  470,
                        520,  476, 520,  546, 448,  548, 448,  552, 444,  552, 1436,  556,
                        440,  556, 440,  552, 1438,  552, 444,  552, 444,  552, 1440,  552,
                        1438,  548, 1466,  526, 1466,  516, 1474,  548, 1440,  552, 442,
                        556, 438,  552, 444,  552, 444,  552, 444,  552, 444,  552, 444,
                        552, 444,  552, 444,  550, 444,  548, 448,  548, 470,  526, 470,
                        526, 470,  470, 524,  518, 478,  542, 448,  552, 444,  552, 444,
                        552, 444,  552, 444,  552, 442,  552, 444,  552, 440,  556, 440,
                        550, 444,  552, 448,  548, 448,  548, 448,  548, 468,  526, 468,
                        526, 468,  470, 522,  470, 524,  548, 446,  548, 448,  552, 2918,
                        3052, 8916,  552, 1440,  552, 442,  552, 438,  556, 440,  552,
                        444,  552, 448,  546, 448,  548, 446,  548, 1466,  526, 470,
                        470, 520,  474, 1516,  546, 1444,  552, 442,  552, 1440,  552,
                        1434,  556, 438,  552, 1440,  552, 1440,  552, 1438,  552, 444,
                        550, 470,  522, 1466,  470, 1520,  518, 1474,  546, 448,  548,
                        444,  556, 438,  556, 1434,  552, 1438,  552, 1436,  556, 444,
                        552, 444,  546, 448,  546, 448,  546, 470,  526, 468,  526, 1466,
                        470, 1516,  548, 1444,  552, 1438,  552, 444,  550, 440,  556, 440,
[... cut ...]
      checkState();
      irsend.sendRaw(SendON, 349, 38); delay(200);
    }
    else
    { Serial.println("Switch it Off : ");
    ac.setBeep  (true);ac.send     ();//ac.sendExtended     ();
      irsend.sendRaw(SendOFF, 349, 38); delay(200);
    }
  }
}

void checkState() {
  Serial.printf("  %s\n", ac.toString().c_str());
  unsigned char* ir_code = ac.getRaw();
  Serial.print("IR Code: 0x");
  for (uint8_t i = 0; i < kSamsungACSectionLength; i++)
    Serial.printf("%02X", ir_code[i]);
  Serial.println();
}

kalpakchiev, 2019-01-25:
With:

"  for (uint8_t i = 0; i < kSamsungACStateLength; i++) "

the sketch does not compile.

“ 'kSamsungACStateLength' was not declared in this scope "

With kSamsungACSectionLength it compiles and print correctly

> On 25 Jan 2019, at 8:29, David Conran <notifications@github.com> wrote:
> 
> kSamsungACStateLength

crankyoldgit, 2019-01-25:
Sorry, it should be: `kSamsungAcStateLength` from https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L371

crankyoldgit, 2019-01-26:
I've found a number of issues with the "extended" message creation/decoding.
I'll upload some improvements later, but the main issue is the checksum calculation doesn't work for these extended messages. Until we that out, extended messages won't work via the Class methods.

kalpakchiev, 2019-01-26:
Thanks for the update.

Currently there are sample sketches only for some of the supported AC models. Maybe it worths to consider adding sample sketches for each of the supported AC models.

crankyoldgit, 2019-02-07:
Some of the improvements have been included in the new v2.5.5 release of the library.

dresende, 2019-03-20:
I'm using v2.5.6 and the issue still stands, or at least I still cannot turn off a Samsung A/C.

Hollako, 2019-03-20:
Hello Dresende,

I have a Samsung AC using the same IRremote library, and i am able to turn off the Samsung AC but with a small workaround for the Power commands. here is below the commands that should be used.

> IRsend ACirsend (IR_LED);
> uint8_t stateoff[21] = {0x02, 0xB2, 0x0F, 0x00, 0x00, 0x00, 0xC0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0xFF, 0x71, 0x80, 0x11, 0xC0};
> uint8_t stateon[21] = {0x02, 0x92, 0x0F, 0x00, 0x00, 0x00, 0xF0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0xE2, 0xFE, 0x71, 0x80, 0x11, 0xF0};

> To turn it ON you should send:  ACirsend.sendSamsungAC(stateon, 21);
> To turn it OFF you should send: ACirsend.sendSamsungAC(stateoff, 21); 

Regards.

dresende, 2019-03-20:
Thank you @Hollako, will try this tomorrow and report back 👍

crankyoldgit, 2019-03-21:
@dresende Yes, I know it isn't fixed, though thank for confirming it. I haven't yet been able to workout the checksum calculations required for an extended message. Hence the issue/bug is still open. The only workaround at present is to manually play a known-good state for doing the on/off as @Hollako has suggested.

dresende, 2019-03-21:
Works perfectly 😃

closed 2019-05-01 by crankyoldgit

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

PR #695 (2019-05-01): Make sure special power mode messages are sent when power is set.
Create a workaround deep in the IRSamsungAc class that handles the special
power message sending so it's automatic for the user.

Fixes #604
commit c8036ce2 (2019-05-01): Make sure special power mode messages are sent when power is set. (#695)

* Make sure special power mode messages are sent when power is set.
Create a workaround deep in the `IRSamsungAc` class that handles the special
power message sending so it's automatic for the user.

Fixes #604