crankyoldgit/IRremoteESP8266 #609: Any global define to limit minimum pulse time?

KonstantinMastak, 2019-01-30:
### Version/revison of the library used
2.5.4

I try to decode remotes I have (3 for ACs and one for LG TV) with help of IRrecvDumpV2 and for all of them I get long sequence of extremely short (<25 us) pulses with occasional long pulses.

Below you can see an example of IRrecvDumpV2 output for LG Smart TV Remote (power on/off button). The distance between remote and my device is about 1 meter, but I tried to make it much longer (i.e. even capture IR mirrored from the wall) with the same result. Shorter - like 10 cm doesn't help either.

The device is some hacked version of chinese universal IR remote based on ESP8266 and other proprietary chip that I excluded. I attached modified schematics below.

I understand that the possible reason can be frequency mismatch, but I tried 4 different remotes from different manufacturers which give similar artefacts and it is highly unlikely that at least one of them is not 38 kHz. Maybe the input circuit of the device has sensivity level of more than needed and it makes some noise. It is important to notice that if I leave the device alone with not sending any codes for a long time, I have NO any parasite messages. I.e. IRrecvDumpV2 captures data only when I really press some buttons.

So, I have two questions now:
1) Any possible errors from my side?
2) Is there any global define or constant I can change somewhere in the library that will help me to cut out all these short pulses, say shorter than 25 us?

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

This is from LG Smart TV remote:

Timestamp : 000495.728
WARNING: IR code is too big for buffer (>= 1024). This result shouldn't be trusted until this is resolved. Edit & increase kCaptureBufferSize.
Encoding  : UNKNOWN
Code      : 7977A6F7 (512 bits)
Library   : v2.5.4

Raw Timing[1023]:
   +    12, -    16,    +    12, -    14,    +    12, -    14,    +    10, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    10, -    14,    +    12, -    14,    +    12, -    16, 
   +    10, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    10, -    14, 
   +    12, -  1698,    +    12, -    14,    +    12, -    16,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    10, -    14, 
   +    12, -    16,    +    10, -    14,    +    12, -    14,    +    12, -    14, 
   +    10, -    16,    +    12, -    14,    +    10, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -  1698,    +    12, -    16,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    10, -    14,    +    12, -    14, 
   +    12, -    14,    +    10, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    16,    +    12, -    12,    +    12, -    16,    +    10, -    40, 
   +    12, -    14,    +    10, -    16,    +    10, -    14,    +    10, -    14, 
   +    12, -    14,    +    12, -  1700,    +    10, -    16,    +    12, -    14, 
   +    12, -    14,    +    12, -    16,    +    12, -    12,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    40,    +    10, -    16, 
   +    10, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    10, -    14,    +    12, -    14,    +    12, -    14,    +    10, -    16, 
   +    12, -    14,    +    10, -  1700,    +    10, -    16,    +    12, -    14, 
   +    10, -    14,    +    12, -    14,    +    12, -    16,    +    10, -    12, 
   +    12, -    16,    +    12, -    14,    +    10, -    14,    +    12, -    14, 
   +    12, -    12,    +    12, -    14,    +    12, -    16,    +    12, -    12, 
   +    12, -    16,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -   578,    +    12, -    14, 
   +    12, -    16,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    10, -    16,    +    12, -   578, 
   +    12, -    16,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -   578,    +    12, -    16,    +    12, -    16,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    40,    +    12, -    14, 
   +    12, -    16,    +    12, -    14,    +    12, -    14,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    10, -    14,    +    12, -    14, 
   +    12, -  1698,    +    12, -    16,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12, -    14, 
[... 78 lines cut ...]
   +    12, -    16,    +    12, -    12,    +    12, -    14,    +    12, -    16, 
   +    12, -    14,    +    10, -    14,    +    12, -    16,    +    12, -    12, 
   +    12, -    14,    +    12, -    16,    +    12, -    12,    +    12, -    14, 
   +    12, -    16,    +    12, -    12,    +    12, -    14,    +    12, -    14, 
   +    12, -    40,    +    12, -    12,    +    12, -    16,    +    12, -    12, 
   +    12, -    14,    +    12, -    14,    +    12, -    14,    +    12

uint16_t rawData[1023] = {12, 16,  12, 14,  12, 14,  10, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  10, 14,  12, 14,  12, 16,  10, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  10, 14,  12, 1698,  12, 14,  12, 16,  12, 14,  12, 14,  12, 14,  12, 14,  10, 14,  12, 16,  10, 14,  12, 14,  12, 14,  10, 16,  12, 14,  10, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 1698,  12, 16,  12, 14,  12, 14,  12, 14,  10, 14,  12, 14,  12, 14,  10, 14,  12, 14,  12, 14,  12, 16,  12, 12,  12, 16,  10, 40,  12, 14,  10, 16,  10, 14,  10, 14,  12, 14,  12, 1700,  10, 16,  12, 14,  12, 14,  12, 16,  12, 12,  12, 14,  12, 14,  12, 14,  12, 40,  10, 16,  10, 14,  12, 14,  12, 14,  12, 14,  10, 14,  12, 14,  12, 14,  10, 16,  12, 14,  10, 1700,  10, 16,  12, 14,  10, 14,  12, 14,  12, 16,  10, 12,  12, 16,  12, 14,  10, 14,  12, 14,  12, 12,  12, 14,  12, 16,  12, 12,  12, 16,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 578,  12, 14,  12, 16,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  10, 16,  12, 578,  12, 16,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 578,  12, 16,  12, 16,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 40,  12, 14,  12, 16,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  10, 14,  12, 14,  12, 1698,  12, 16,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 16,  12, 12,  12, 578,  12, 16,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 16,  12, 12,  12, 14,  12, 12,  12, 16,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 578,  12, 16,  12, 14,  12, 16,  12, 12,  12, 14,  12, 14,  12, 40,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 16,  12, 14,  10, 14,  12, 578,  12, 14,  12, 16,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 16,  10, 578,  12, 14,  12, 16,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  12, 12,  12, 14,  12, 14,  12, 14,  12, 14,  12, 14,  [... cut ...] 2, 12,  12, 14,  12, 14,  12, 40,  12, 12,  12, 16,  12, 12,  12, 14,  12, 14,  12, 14,  12};  // UNKNOWN 7977A6F7

#### Example code used

IRrecvDumpV2 without any modifications.

#### Circuit diagram and hardware used (if applicable)

Please see it here:
https://drive.google.com/file/d/1gy7qdb0lmFRy8jiQInLgmfJBB_kb3aXO/view?usp=sharing

### I have followed the steps in the [Troubleshooting Guide]

Yes, to my best knowledge.

### Has this library/code previously worked as expected for you?

No.

crankyoldgit, 2019-01-30:
TL;DR: I think you have a hardware issue. Simplify the circuit and/or use a different IR demodulation component.

> Is there any global define or constant I can change somewhere in the library that will help me to cut out all these short pulses, say shorter than 25 us?

In short, no. There is nothing that will do that. Even if there was, the poor ol' ESP will be being hammered with an interrupt every 12-14 useconds. I'm surprised it could function at that massive rate of interrupts.

> Any possible errors from my side?

Most likely. If I'm reading your issue correctly, it's not a circuit or hardware of your design. Correct?

Looking at that rawData, I'm pretty sure what the ESP8266 is seeing is the unmodulated (think very raw) IR signal, not the output of the IR demodulator component.
A _Mark_ pulse is normally, say for example 500 usecs long. During that time, the IR LED will pulse at 38kHz, rather than being on solid for that time. That's about the 12-14 usecs on and then off and then on and then off, for the entire ~500 usecs of the Mark signal. That's kind of what the raw data is showing.
The IR demodulator component (IRM 3638 in your circuit diagram) should convert that _raw_ 38kHz mark signal into a single pulse of 500 usecs, rather than the 20 odd individual pulses.

For what ever reason, the circuit you are using doesn't seem to be doing that. I'm no electrical/hardware engineer, sorry. I can't debug that circuit diagram etc,

I **strongly** recommend you try out this suggested receiving circuit here to eliminate hardware issues: https://github.com/markszabo/IRremoteESP8266/wiki#ir-receiving

The "Frequency mismatch" issue described when using the wrong demodulator frequency typically occurs as the very occasional <~50usec pulses. Not a constant barrage as in this case. i.e. maybe one in 20+ timing values. This is not it.

KonstantinMastak, 2019-01-31:
David,

Thank you very much! Your advice really makes sense. I didn't notice that
what I get is mostly modulated signal at 38 kHz instead of demodulated one.

I will try changing the circut as you suggested.

ᐧ

KonstantinMastak, 2019-01-31:
David,

After I changed TSOP everything started working as expected. Thank you very
much for your help!

The issue can be closed.
ᐧ

чт, 31 янв. 2019 г. в 03:17, Konstantin Artemev <konstantinartemev@gmail.com