crankyoldgit/IRremoteESP8266 #612: How to send unmodulated signal?

andreimos, 2019-02-02:
Hello, could you please help me with my problem?

I’m trying to send signal to my SONY projector by using a simple code:

 irsend.sendSony(0X542a, 15,2);

But I need to send the signal without 40kHz modulation because I want to do it without using IR emitter/receiver, but via direct wiring. 

What should I change in the code of library to be able to send unmodulated signal?

Thank you in advance. 
Andrey

crankyoldgit, 2019-02-02:
See https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L32

e.g.
```
IRsend irsend(
    PIN_YOU_WANT_TO_USE,  // GPIO to send the messages on.
    false,  // Don't use an inverted signal.
    false  // Don't use modulation.
);
```

That, in theory should do what you're asking.

crankyoldgit, 2019-02-02:
You may also need to change `33` to `100` at https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L32 too if that doesn't work.