crankyoldgit/IRremoteESP8266 #614: Consider adding a badge from ArduBadge

gilmaimon, 2019-02-04:
Hi, 
Please consider adding this badge from [ArduBadge](https://www.ardu-badge.com):
[image]
Markdown:
`[[image]](https://www.ardu-badge.com/IRremoteESP8266)`

It shows the latest version that is available on the Arduino IDE (and that the library is indeed in the IDE's Library Manager). Also, clicking on the badge redirects to a custom page with instructions for installing IRremoteESP8266 from the IDE and from a ZIP file.
Hope you liked it :)
(btw, ArduBadge is free, open source and non profit)

crankyoldgit, 2019-02-05:
Nice work!
Done in https://github.com/markszabo/IRremoteESP8266/commit/5ee8b8a6daf95c1ee41efe67f5d417727ba85580

crankyoldgit, 2019-04-25:
@gilmaimon I notice the url/service no longer works. Should we remove the badge or is it a transient issue?

gilmaimon, 2019-04-25:
I apologise for the issue, it is up again now. I will do my best to prevent this from happening again.

crankyoldgit, 2019-04-26:
Thanks. Seems to be working again. Nothing for us to do. You might want to consider putting some monitoring on your service so you know if it goes down again.

closed 2019-04-26 by crankyoldgit