crankyoldgit/IRremoteESP8266 #620: LG IR send does not work properly

jimmys01, 2019-02-20:
I capture this from the original remote (MED FAN 22c COLD)

```
ORIGINAL REMOTE

Timestamp : 001935.312
Encoding  : LG
Code      : 8808721 (28 bits)
Library   : v2.5.5

Raw Timing[59]:
   +  8886, -  4152,    +   560, -  1538,    +   532, -   502,    +   532, -   504, 
   +   530, -   484,    +   558, -  1536,    +   508, -   516,    +   558, -   502, 
   +   532, -   484,    +   558, -   502,    +   532, -   500,    +   534, -   508, 
   +   532, -   502,    +   532, -  1518,    +   558, -   510,    +   532, -   484, 
   +   556, -   486,    +   556, -   510,    +   532, -  1518,    +   558, -  1560, 
   +   532, -  1528,    +   556, -   504,    +   530, -   506,    +   530, -  1520, 
   +   558, -   508,    +   534, -   500,    +   532, -   512,    +   530, -   484, 
   +   556, -  1536,    +   532

uint16_t rawData[59] = {8886, 4152,  560, 1538,  532, 502,  532, 504,  530, 484,  558, 1536,  508, 516,  558, 502,  532, 484,  558, 502,  532, 500,  534, 508,  532, 502,  532, 1518,  558, 510,  532, 484,  556, 486,  556, 510,  532, 1518,  558, 1560,  532, 1528,  556, 504,  530, 506,  530, 1520,  558, 508,  534, 500,  532, 512,  530, 484,  556, 1536,  532};  // LG 8808721
uint32_t address = 0x88;
uint32_t command = 0x872;
uint64_t data = 0x8808721;
```

I play the command LG,8808721  and re capture the playback

```

REPLAY COMMAND LG,8808721 REMOTE

Timestamp : 001985.828
Encoding  : LG
Code      : 8808754 (28 bits)
Library   : v2.5.5

Raw Timing[59]:
   +  8544, -  4218,    +   590, -  1598,    +   590, -   516,    +   562, -   544, 
   +   590, -   518,    +   590, -  1564,    +   566, -   540,    +   592, -   512, 
   +   566, -   540,    +   564, -   542,    +   564, -   542,    +   564, -   540, 
   +   592, -   516,    +   566, -  1592,    +   566, -   538,    +   598, -   512, 
   +   570, -   540,    +   590, -   516,    +   566, -  1566,    +   616, -  1566, 
   +   566, -  1590,    +   564, -   542,    +   586, -  1542,    +   592, -   542, 
   +   562, -  1590,    +   566, -   542,    +   566, -  1588,    +   596, -   514, 
   +   568, -   550,    +   570

uint16_t rawData[59] = {8544, 4218,  590, 1598,  590, 516,  562, 544,  590, 518,  590, 1564,  566, 540,  592, 512,  566, 540,  564, 542,  564, 542,  564, 540,  592, 516,  566, 1592,  566, 538,  598, 512,  570, 540,  590, 516,  566, 1566,  616, 1566,  566, 1590,  564, 542,  586, 1542,  592, 542,  562, 1590,  566, 542,  566, 1588,  596, 514,  568, 550,  570};  // LG 8808754
uint32_t address = 0x88;
uint32_t command = 0x875;
uint64_t data = 0x8808754;

```
The playbacked command is not the same as the original, the AC does not recognize the command. The only command that i have seen to be faitfull in replay and in original remote is the OFF command.(LG,88C0051)

crankyoldgit, 2019-02-20:
Thanks for the report. I'll look into what might be causing the problem.

jimmys01, 2019-02-20:
AIR dehumidify command 
Original

```
Timestamp : 003011.388
Encoding  : LG
Code      : 8809902 (28 bits)
Library   : v2.5.5

Raw Timing[59]:
   +  8936, -  4088,    +   532, -  1550,    +   558, -   510,    +   510, -   524, 
   +   508, -   526,    +   508, -  1560,    +   508, -   526,    +   510, -   532, 
   +   532, -   484,    +   558, -   502,    +   530, -   484,    +   556, -   504, 
   +   530, -   504,    +   532, -  1544,    +   532, -   502,    +   532, -   486, 
   +   556, -  1560,    +   532, -  1526,    +   548, -   528,    +   530, -   486, 
   +   536, -  1562,    +   558, -   502,    +   530, -   486,    +   556, -   510, 
   +   532, -   512,    +   530, -   486,    +   556, -   502,    +   532, -  1530, 
   +   532, -   514,    +   558

uint16_t rawData[59] = {8936, 4088,  532, 1550,  558, 510,  510, 524,  508, 526,  508, 1560,  508, 526,  510, 532,  532, 484,  558, 502,  530, 484,  556, 504,  530, 504,  532, 1544,  532, 502,  532, 486,  556, 1560,  532, 1526,  548, 528,  530, 486,  536, 1562,  558, 502,  530, 486,  556, 510,  532, 512,  530, 486,  556, 502,  532, 1530,  532, 514,  558};  // LG 8809902
uint32_t address = 0x88;
uint32_t command = 0x990;
uint64_t data = 0x8809902;
```

Playback
```
Timestamp : 003022.329
Encoding  : LG
Code      : 8809946 (28 bits)
Library   : v2.5.5

Raw Timing[59]:
   +  8488, -  4236,    +   594, -  1590,    +   594, -   514,    +   594, -   514, 
   +   594, -   514,    +   590, -  1564,    +   598, -   516,    +   576, -   528, 
   +   596, -   514,    +   564, -   542,    +   588, -   518,    +   562, -   544, 
   +   570, -   540,    +   590, -  1538,    +   620, -   516,    +   566, -   544, 
   +   588, -  1566,    +   566, -  1592,    +   566, -   564,    +   540, -   542, 
   +   588, -  1568,    +   568, -   512,    +   592, **-  1592**,    +   594, -   512, 
   +   566, -   544,    +   566, -   540,    +   592, **-  1562,**    +   598, -  1562, 
   +   564, -   548,    +   568

uint16_t rawData[59] = {8488, 4236,  594, 1590,  594, 514,  594, 514,  594, 514,  590, 1564,  598, 516,  576, 528,  596, 514,  564, 542,  588, 518,  562, 544,  570, 540,  590, 1538,  620, 516,  566, 544,  588, 1566,  566, 1592,  566, 564,  540, 542,  588, 1568,  568, 512,  592, 1592,  594, 512,  566, 544,  566, 540,  592, 1562,  598, 1562,  564, 548,  568};  // LG 8809946
uint32_t address = 0x88;
uint32_t command = 0x994;
uint64_t data = 0x8809946;

```

crankyoldgit, 2019-02-20:
> I play the command LG,8808721 and re capture the playback

@jimmys01 Can you please tell me how you are sending that command? e.g. Can you please include the code you are using to send with?

I've done some preliminary tests and the library appears to be sending and decoding `irsend.sendLG(0x8808721);` correctly.

If you have a look at this (https://github.com/markszabo/IRremoteESP8266/commit/cb7d5d032b2a74950e6508113ca54cfcc83c3388) you'll see it seems to decode the sent code as the same code. Thus I suspect it might be something with the program you are using to send the code with.

jimmys01, 2019-02-20:
Hmmm.. OK.

uint64_t IrCode=0;
IrCode = strtoull(ircodestr.c_str(), NULL, 16);  //Converst string to unsigned long long integer
Plugin_035_irSender->sendLG(IrCode);

The only room for error I can see is the strtoull command.

crankyoldgit, 2019-02-20:
Yeah, I figured you might have been using ESPEasy.
Can you put in some debug statements etc to see what the `ircodestr` value is, and what the `IrCode` value is after the `strtoull()`?

My tests really do point to something wrong with the value you're calling `sendLG()` with.

Can you try using the [IRsendDemo.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) example code to send the LG command on the same hardware? You'll have to modify it for you GPIO and LG code etc? If that doesn't work, then something is wrong with our library or the hardware, if it does, then it clearly something in ESPEasy.

jimmys01, 2019-02-20:
Ok so this does not seem to be related to the library.

the code 

```
void loop() {
uint64_t IrCode=0;
String ircodestr = "8808721";
IrCode = strtoull(ircodestr.c_str(), NULL, 16);
  
  Serial.println("LG");
  irsend.sendLG(0x8808721);
  delay(2000);

  Serial.println("LG - TEST");
  irsend.sendLG(IrCode);
  delay(2000);
}
```
Both work and produce correct ir messages. 

Thank you for your help, i will try to figure whats wrong with the ESPEasy code.

crankyoldgit, 2019-02-21:
Phew. Thanks for confirming we are fine. Good luck hunting in ESPEasy.
Marking this closed.

closed 2019-02-21 by crankyoldgit