crankyoldgit/IRremoteESP8266 #626: Is there any mean to make a simple "learn" and "send" commands ?

lgm42, 2019-02-26:
### Version/revison of the library used
2.5.5

### Expected behavior
I would like to learn command from an unknown protocol. 

To manage this, I used the example IRrecvDumpV2. It's works and generate the code for the example IRSendDemo that's works too. Your examples are very clear so it is easy to test them.

The remaining problem is that your are using resultToSourceCode() method. Have you another method that takes the decode_results too and returns directly the uint16_t array to use it without code modification ?
Like this  
int resultToRawData(decode_results *results, uint16_t * resultArray)
?

I could make it but I don't understand what you're doing between the decode_results and the generated code. 

Is this method exists and I missed it ?
Have you at least an explanation of the behavior of the function to help me to code it?

crankyoldgit, 2019-02-26:
TL;DR: No, it doesn't exist. It's complicated. There isn't a compelling reason to offer it, and people have enough trouble with the API as it is.

Long answer:
The method you are after doesn't exist. It's not really hard to code, but the memory handling is the _tricky_ part.

Essentially, the Raw Data, as printed to the serial port, is stored in the [decode_results](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L87) structure.

To convert `decode_results` to the array that is by `sendRaw()`, you need to multiply each value of [decode_results.rawbuf[]](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L102) by `kRawTick` (typically 2). The elements of `rawbuf[]` array are stored as _ticks_, where the array used by `sendRaw()` uses entries of useconds. e,g, _1 tick == 2 usecs_; or _nr. of ticks * `kRawTick` == nr. of usecs_ for each _pulse_.

The elements of`rawbuf` and the array used by `sendRaw()` are `uint16_t` (16 bit integer), which has a maximum value of 65,535 (2^16 - 1). In practice, it is common/possible for protocol `spaces` to exceed 70,000 usecs. e.g. 100,000+. We need to shrink such times down to `ticks` by dividing by some factor (`kRawTick`) to fit inside of 16 bits.
Note: The first meaningful element of [decode_results.rawbuf[]](https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRrecv.h#L102) is `decode_results.rawbuf[1]`, the element 0 is not used. The last element of captured data is effectively `decode_results.rawbuf[decode_results.rawlen]`.

`resultToSourceCode()` does all the conversion needed handling the special cases when a value larger than 65,535 is needed. As it is outputting to a string/serial it doesn't need to do memory management.

Now to the reason why your suggested solution doesn't exist as as a standard library method: **Memory Management**. As the array needed to store the `sendRaw()` Raw Data array can't easily be pre-determined without inspecting all the data in `decode_results` you have two options for passing the desired _result_ back to the calling routine.
- Pass a pre-allocated array & size to the new method:
  Have the calling code allocate a large enough array(memory) to store the result. The size of which _may_ need to be large than the `rawbuf` array size due to the _ticks to usecs_ conversion. That's potentially a large amount of memory to waste. e.g. A capture buffer could be 2k+ of memory depending on user request etc.
A lot of messages only require 200-800 bytes in `sendRaw()` form. e.g. 10x memory wastage. Memory is tight on these platforms. We also have to ensure we don't exceed the user supplied array, otherwise we trample the memory of the ESP8266. This solution leaves the _allocation_ of memory as the caller's problem.

- Have the new method allocate the correct size required for the `sendRaw()` array and return a pointer:
  After parsing `rawbuf`, we can determine the size needed, allocate the memory with a `malloc()` call, and return a pointer etc. However, the user then has to handle `free()`ing the memory after it is done. If not, the ESP will quickly run out of memory. It has the advantage of the using the least memory (only the max size required etc) but it requires the user to do the memory clean up. Most arduino codes/users of this library are not familiar with dynamic memory management, so that path can lead to problems fast.

Both cases have issues where there isn't enough space for the result. Either not enough memory passed to the method to fit the result, or not enough memory available to allocate to return a result. Not major problems, but something to consider.

As the library itself and 99% of the users don't need to do this level of manipulation, those methods are not provided as standard as described can quickly escalate into memory problems. C++ doesn't have [Garbage Collection](https://en.wikipedia.org/wiki/Garbage_collection_(computer_science)) hence we have to worry about this, especially with the limited memory of embedded systems like the ESP8266.

If you have a safe way to implement it, I'm all ears. We could offer both above solutions, but a) It's not a common use/need & b) it may lead to more support issues than needed.

i.e. If someone really needs this function/method, they should be able to implement it themselves & will most likely have the skills required to know what method they will want and how to handle the memory issues.

lgm42, 2019-02-26:
Thank you for your detailed answer.
I understand the memory problem. I will implement it by myself assuming I use only "classic remotes codes" which fit into  less than 50 bits.
Assuming that, I can provide a null pointer that the function will allocate and fill ( b) method).
I will just add a check for unexperiment users that's checks if pointer to allocate is not null, it will free it before. So, if they use always the same pointer, it prevent from running out of memory even without any free calls.
When I finish the method, I could propose you onto this issue

crankyoldgit, 2019-02-26:
> When I finish the method, I could propose you onto this issue

Sure.

Yep, passing a result pointer in, and a size (for bounds checking), and returning an `uint16_t` success (size used if it fitted) or `0` if it didn't.

FYI `uint16t buf[200]` will fit almost all Non-A/C codes.

lgm42, 2019-02-26:
Here the function:

```c++
uint16_t IRManager::resultToRawData(const decode_results *results, 
                           uint16_t ** rawData, const int maxSize) {
  
  if (rawData == NULL)
	return -1;

  if ((*rawData) != NULL) {
	free(*rawData);
	*rawData = NULL;
  }
  //check for memory 
  uint16_t expectedSize = getCorrectedRawLength(results);
  if (expectedSize > maxSize)
	return -2;

  *rawData = (uint16_t *)malloc(sizeof(uint16_t) * expectedSize);

  if (*rawData == NULL)
	return -3;

  int index = 0;

  // Dump data
  for (uint16_t i = 1; i < results->rawlen; i++) {
    uint32_t usecs;
    for (usecs = results->rawbuf[i] * kRawTick; usecs > UINT16_MAX;
         usecs -= UINT16_MAX) {
      (*rawData)[index++] = UINT16_MAX;
	  (*rawData)[index++] = 0;
    }
	(*rawData)[index++] = usecs;
  }

  if (index != expectedSize)
	return -4;

  //everything ok
  return index;
}
```
The prototype
```c++
uint16_t resultToRawData(const decode_results *results, 
                                uint16_t ** rawData, const int maxSize = 200);
```

And the usage:
```c++
decode_results results;
if (_irReceiver.decode(&results)) {
	uint16_t *rawData = NULL;
	int size = resultToRawData(&results, &rawData);
	if (size > 0)
	{
		irsend.sendRaw(rawData, size, 38); 
		free(rawData);
		rawData = NULL;
	}
	_irReceiver.resume();  // Receive the next value
}
```

Hope it can help you or someone that have same "special needs" like me.

closed 2019-02-27 by lgm42