crankyoldgit/IRremoteESP8266 #628: Improve IRsend::send()

leobel96, 2019-03-03:
The IRsend class has the very useful method `IRsend::send()` which lets you send commands to many different devices with [just one function](https://github.com/markszabo/IRremoteESP8266/blob/583366c476e2685172e5505543f6f21462f2ac96/src/IRsend.cpp#L492). Anyway it supports few types of devices (no Panasonic,...). In my opinion it would be great to add all the not-AC devices to this method and add a similar method for all the AC devices with arguments like temperature, mode, fan, type,...

crankyoldgit, 2019-03-03:
I'll update `send()` for all the simple protocols. Creating a generic AC one is tricky and a lot more effort, so it won't happen any time soon.

closed 2019-03-05 by crankyoldgit

glmnet, 2019-03-12:
> I'll update send() for all the simple protocols. Creating a generic AC one is tricky and a lot more effort, so it won't happen any time soon.

Can we dive a little on the generic AC one? I was thinking about polymorphism but that seems a no-go as @devyte metioned [here](https://github.com/markszabo/IRremoteESP8266/issues/634)

The main use case I have in mind would be the mqttserver example or the EspEasy tx plugin

crankyoldgit, 2019-03-13:
You're not the only one who want's it. I've had a couple of attempts to add it, and each time the code became a horrible mess. I want to do it in such a way that it works with Home Assistant's "climate" feature and/or present a JSON interface etc.

@glmnet There are a number of issues. Mainly that not all A/Cs operate the same way. some have special On/Off messages, etc etc. I need to come up with a way that is generic and supports most of the A/C protocols with as much of the features as I can, and maybe a way of supporting other features that some models don't etc.

So, It hasn't been forgotten. But hey, as usual, feel free to contribute a PR/solution. ;-)

crankyoldgit, 2019-03-16:
V2.5.6 has just been release which includes the changes/improvements mentioned.

crankyoldgit, 2019-04-30:
FYI, this (generic a/c) has been included in the newly released v2.6.0 of the library.

leobel96, 2019-04-30:
Fantastic!

PR #629 (2019-03-05): Update IRsend::send() to include all simple protocols.
- Add missing simple (single integer) IR protocols.
- Sort the switch statement alphabetically.

Fixes #628
commit c1c21530 (2019-03-05): Update IRsend::send() to include all simple protocols. (#629)

* Update IRsend::send() to include all simple protocols.
- Add missing simple (single integer) IR protocols.
- Sort the switch statement alphabetically.
- Add unit tests for irsend::send()

Fixes #628
PR #664 (2019-04-12): Initial prototype of a common interface for all A/Cs.
* Provide a single class/function with standardised common arguments for controlling
  all deeply supported A/C units.
* Extend unit testing for the `ac.send()` class method.
* Supports:
  - Argo
  - Coolix
  - Daikin
  - Gree
  - Fujitsu
  - Haier
  - Hitachi
  - Kelvinator
  - Midea
  - Mitsubishi
  - Panasonic
  - Samsung (inc. On/Off fixes)
  - Tcl112
  - Teco
  - Toshiba
  - Trotec (inc. some bug fixes)
  - Vestel (inc. some improvements)
  - Whirlpool
* Unit tests for the above.
* Allow unit tests to use the `c++11` standard
* Require [... cut]
commit 2c698dc5 (2019-04-12): Initial prototype of a common interface for all A/Cs. (#664)

* Provide a single class/function with standardised common arguments for controlling
  all deeply supported A/C units.
* Extend unit testing for the `ac.send()` class method.
* Supports:
  - Argo
  - Coolix
  - Daikin
  - Gree
  - Fujitsu
  - Haier
  - Hitachi
  - Kelvinator
  - Midea
  - Mitsubishi
  - Panasonic
  - Samsung (inc. On/Off fixes)
  - Tcl112
  - Teco
  - Toshiba
  - Trotec (inc. some bug fixes)
  - Vestel (inc. some improvements)
  - Whirlpool
* Unit tests for the above.
* Allow unit tests to use the `c++11` standard
* [... cut]