crankyoldgit/IRremoteESP8266 #636: Trying to calculate Carrier AC raw data by self

simonfongnt, 2019-03-14:
### Version/revison of the library used
IRremoteESP8266 2.5.5

### Expected behavior
AC is Carrier 42KCEG12A with remote RH56BGEF-CA
I want to control it via ESP8266 ESP-12E

Sorry if this is a dumb question since I lacked of basic knowledge of IR codec. 
The captured code sometimes works so I wish to re-enter all the values to improve my array. 
I've tried to use getraw() in COOLIX.cpp without success so I wish to have some hints of COOLIX encoding so as to find the most accurate value of each bit.
I appreciate if there is any example of bit calculation in 38kHz IR remote.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
```
Encoding  : COOLIX
Code      : B21FB8 (24 bits)
Mesg Desc.: Power: On, Fan: 0 (AUTO0), Mode: 2 (AUTO), Temp: 30C, Zone Follow: Off, Sensor Temp: Ignored
Library   : v2.5.5

Raw Timing[199]:
   +  4494, -  4296,    +   634, -  1548,    +   632, -   484,    +   610, -  1552, 
   +   632, -  1548,    +   632, -   484,    +   610, -   484,    +   610, -  1548, 
   +   634, -   484,    +   614, -   482,    +   692, -  1468,    +   706, -   412, 
   +   610, -   484,    +   610, -  1570,    +   612, -  1546,    +   634, -   482, 
   +   610, -  1554,    +   634, -   486,    +   610, -   484,    +   610, -   484, 
   +   610, -  1570,    +   610, -  1550,    +   632, -  1548,    +   634, -  1548, 
   +   634, -  1548,    +   634, -  1554,    +   634, -  1546,    +   634, -  1570, 
   +   610, -   484,    +   610, -   482,    +   612, -   482,    +   610, -   484, 
   +   610, -   484,    +   612, -  1548,    +   634, -   482,    +   610, -  1552, 
   +   632, -  1546,    +   634, -  1546,    +   634, -   482,    +   610, -   482, 
   +   610, -   486,    +   610, -   484,    +   610, -  1550,    +   634, -   482, 
   +   610, -   482,    +   610, -   484,    +   610, -  1550,    +   634, -  1546, 
   +   634, -  1548,    +   628, -  5160,    +  4490, -  4296,    +   636, -  1546, 
   +   634, -   484,    +   610, -  1550,    +   634, -  1546,    +   634, -   482, 
   +   612, -   482,    +   610, -  1548,    +   634, -   484,    +   610, -   484, 
   +   612, -  1550,    +   634, -   482,    +   612, -   482,    +   610, -  1546, 
   +   634, -  1550,    +   632, -   482,    +   610, -  1552,    +   654, -   466, 
   +   610, -   484,    +   610, -   482,    +   612, -  1546,    +   634, -  1546, 
   +   634, -  1546,    +   634, -  1546,    +   634, -  1548,    +   634, -  1552, 
   +   634, -  1546,    +   634, -  1568,    +   612, -   484,    +   610, -   482, 
   +   610, -   482,    +   612, -   482,    +   612, -   484,    +   610, -  1548, 
   +   634, -   482,    +   612, -  1550,    +   634, -  1546,    +   634, -  1546, 
   +   634, -   482,    +   612, -   482,    +   612, -   484,    +   610, -   484, 
   +   612, -  1550,    +   634, -   482,    +   612, -   482,    +   610, -   482, 
   +   612, -  1550,    +   634, -  1548,    +   632, -  1552,    +   626

uint16_t rawData[199] = {4494, 4296,  634, 1548,  632, 484,  610, 1552,  632, 1548,  632, 484,  610, 484,  610, 1548,  634, 484,  614, 482,  692, 1468,  706, 412,  610, 484,  610, 1570,  612, 1546,  634, 482,  610, 1554,  634, 486,  610, 484,  610, 484,  610, 1570,  610, 1550,  632, 1548,  634, 1548,  634, 1548,  634, 1554,  634, 1546,  634, 1570,  610, 484,  610, 482,  612, 482,  610, 484,  610, 484,  612, 1548,  634, 482,  610, 1552,  632, 1546,  634, 1546,  634, 482,  610, 482,  610, 486,  610, 484,  610, 1550,  634, 482,  610, 482,  610, 484,  610, 1550,  634, 1546,  634, 1548,  628, 5160,  4490, 4296,  636, 1546,  634, 484,  610, 1550,  634, 1546,  634, 482,  612, 482,  610, 1548,  634, 484,  610, 484,  612, 1550,  634, 482,  612, 482,  610, 1546,  634, 1550,  632, 482,  610, 1552,  654, 466,  610, 484,  610, 482,  612, 1546,  634, 1546,  634, 1546,  634, 1546,  634, 1548,  634, 1552,  634, 1546,  634, 1568,  612, 484,  610, 482,  610, 482,  612, 482,  612, 484,  610, 1548,  634, 482,  612, 1550,  634, 1546,  634, 1546,  634, 482,  612, 482,  612, 484,  610, 484,  612, 1550,  634, 482,  612, 482,  610, 482,  612, 1550,  634, 1548,  632, 1552,  626};  // COOLIX B21FB8
uint64_t data = 0xB21FB8;
```
### Actual behavior
```
uint16_t rawData[12][199] = {
// Power Off
{ 4494, 4294,  636, 1546,  634, 482,  612, 1548,  636, 1544,  636, 482,  612, 482,  612, 1546,  634, 484,  612, 484,  612, 1550,  634, 482,  610, 460,  640, 1538,  636, 1546,  634, 482,  612, 1552,  634, 484,  612, 1546,  634, 1546,  634, 1544,  636, 1544,  634, 482,  612, 1550,  634, 1548,  636, 1552,  634, 482,  612, 482,  612, 482,  612, 482,  612, 1546,  634, 482,  612, 484,  612, 1552,  634, 1546,  636, 1546,  636, 482,  612, 482,  612, 482,  612, 482,  612, 484,  612, 484,  612, 482,  612, 482,  612, 1572,  612, 1546,  634, 1546,  636, 1546,  634, 1546,  626,   5162,  
  4492, 4294,  634, 1546,  634, 482,  612, 1550,  634, 1544,  634, 482,  612, 482,  612, 1548,  632, 484,  612, 484,  612, 1550,  634, 482,  612, 482,  612, 1546,  634, 1546,  634, 482,  612, 1552,  634, 460,  640, 1540,  636, 1544,  636, 1544,  636, 1544,  634, 482,  612, 1550,  634, 1546,  636, 1550,  634, 482,  612, 482,  612, 482,  612, 482,  612, 1544,  636, 482,  612, 484,  612, 1552,  634, 1546,  636, 1544,  636, 482,  612, 482,  612, 482,  612, 482,  612, 484,  612, 484,  612, 482,  612, 482,  612, 1550,  636, 1544,  636, 1544,  634, 1546,  634, 1570,  590},
// 20C
{ 4444, 4356,  630, 1574,  606, 488,  582, 1580,  628, 1554,  624, 490,  578, 516,  588, 1572,  624, 492,  568, 528,  586, 1578,  622, 492,  568, 526,  580, 1580,  620, 1558,  632, 486,  560, 1604,  620, 498,  566, 528,  568, 526,  568, 1590,  612, 1570,  608, 1594,  586, 1574,  606, 1576,  604, 1582,  604, 1576,  600, 1580,  596, 518,  564, 530,  564, 530,  564, 530,  564, 532,  562, 534,  560, 534,  558, 1600,  588, 526,  558, 1602,  590, 526,  558, 534,  560, 536,  558, 1606,  582, 1592,  588, 528,  560, 1602,  588, 528,  560, 1602,  582, 1598,  580, 1600,  650,   5138,  
  4438, 4348,  582, 1598,  582, 534,  560, 1602,  582, 1598,  582, 534,  558, 534,  560, 1598,  582, 536,  558, 536,  560, 1604,  580, 534,  558, 534,  560, 1598,  582, 1598,  582, 534,  560, 1604,  582, 536,  558, 534,  560, 534,  560, 1598,  586, 1594,  582, 1598,  582, 1598,  582, 1600,  582, 1604,  582, 1598,  582, 1598,  582, 534,  560, 534,  558, 534,  560, 534,  558, 536,  560, 536,  560, 534,  558, 1598,  582, 534,  560, 1602,  582, 534,  560, 534,  560, 536,  560, 1600,  582, 1598,  582, 534,  560, 1602,  582, 534,  558, 1602,  582, 1598,  582, 1624,  608},
// 21C
{ 4518, 4288,  582, 1598,  582, 534,  560, 1602,  582, 1598,  582, 534,  560, 534,  558, 1600,  582, 536,  560, 536,  560, 1602,  582, 534,  560, 534,  560, 1620,  558, 1598,  582, 534,  558, 1604,  582, 536,  558, 534,  558, 534,  558, 1598,  582, 1598,  582, 1598,  580, 1598,  582, 1600,  582, 1626,  560, 1600,  580, 1598,  582, 534,  558, 534,  560, 534,  560, 534,  558, 536,  560, 536,  560, 1602,  588, 1592,  582, 534,  560, 1602,  582, 534,  560, 534,  558, 538,  558, 1600,  582, 534,  558, 534,  560, 1598,  582, 534,  558, 1604,  582, 1598,  582, 1600,  628,   5158,  
  4438, 4348,  582, 1598,  582, 534,  560, 1602,  582, 1598,  582, 534,  558, 534,  560, 1598,  582, 536,  558, 536,  560, 1602,  582, 534,  558, 536,  558, 1600,  582, 1600,  582, 534,  560, 1604,  582, 536,  560, 534,  560, 534,  560, 1600,  580, 1598,  582, 1598,  582, 1600,  580, 1600,  582, 1604,  582, 1600,  580, 1598,  582, 534,  560, 534,  558, 536,  558, 534,  560, 536,  560, 536,  558, 1602,  582, 1620,  560, 534,  558, 1602,  582, 534,  560, 534,  558, 538,  558, 1600,  588, 506,  584, 534,  558, 1598,  582, 534,  558, 1604,  582, 1598,  582, 1602,  624},
//22C
{ 4508, 4296,  584, 1598,  582, 534,  558, 1604,  582, 1598,  582, 534,  560, 534,  560, 1598,  584, 514,  586, 532,  560, 1604,  580, 534,  558, 536,  558, 1600,  582, 1598,  582, 534,  560, 1604,  582, 536,  560, 534,  560, 534,  560, 1600,  580, 1598,  582, 1598,  582, 1602,  578, 1600,  582, 1606,  582, 1598,  582, 1598,  582, 534,  560, 534,  560, 534,  560, 534,  560, 536,  560, 536,  558, 1602,  582, 1598,  582, 1598,  582, 1598,  582, 534,  560, 534,  558, 536,  560, 1600,  582, 534,  560, 534,  558, 534,  560, 534,  560, 1598,  582, 1600,  580, 1600,  628,   5138,  
[... cut ...]
  4438, 4348,  582, 1600,  592, 500,  582, 1602,  582, 1598,  582, 534,  560, 534,  558, 1598,  582, 536,  558, 538,  558, 1604,  580, 534,  558, 534,  560, 1598,  582, 1620,  558, 536,  558, 1626,  560, 536,  560, 534,  558, 534,  560, 1598,  582, 1600,  580, 1600,  582, 1600,  580, 1600,  582, 1606,  580, 1598,  582, 1598,  582, 534,  560, 534,  558, 534,  558, 534,  560, 540,  556, 1600,  582, 534,  558, 1602,  582, 1600,  580, 1600,  580, 534,  560, 534,  558, 536,  558, 536,  560, 1602,  582, 534,  560, 534,  558, 536,  558, 1604,  582, 1598,  582, 1602,  622}
  };
```

crankyoldgit, 2019-03-14:
Hi,
  I think you may be a little confused with how to best use the library. That's okay, it can be quite complex at first.

If you modified the [IRSendDemo](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino) to do the following:
```
irsend.sendCOOLIX(0xB21FB8);  // Value taken from the IRrecvDumpV2 capture you provided.
```

It should send your _Power: On, Fan: 0 (AUTO0), Mode: 2 (AUTO), Temp: 30C, Zone Follow: Off, Sensor Temp: Ignored_ message you listed.

i.e. You only need the short code, not the raw data, if you use the appropriate `sendBlah(0xfoobaa);` method. You can save a lot of space by not storing the raw data arrays.

You can also construct custom codes by using the `IRCoolixAC` class.
e.g. Suitably modify an [AC example sketch](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/TurnOnTrotecAC/TurnOnTrotecAC.ino) for Coolix etc. and use the class methods you need to get the desired functions etc.

Note: I strongly recommend you use the `master` branch of this library, which has some recent fixes to Coolix that are not in the 2.5.5 release.

You can also extract the numerical short code from the class object if you need by using `code = ac.getRaw();` after you have constructed it to what you want too.

Hope that helps.

simonfongnt, 2019-03-14:
I did encounter a problem with using the irsend.sendCOOLIX(). The size of the packet (=99) is half of the captured one (= 199) as follows:
Timestamp : 000043.739
Encoding  : COOLIX
```
Code      : B21FB8 (24 bits)
Mesg Desc.: Power: On, Fan: 0 (AUTO0), Mode: 2 (AUTO), Temp: 30C, Zone Follow: Off, Sensor Temp: Ignored
Library   : v2.5.5

Raw Timing[99]:
   +  4548, -  4444,    +   642, -  1608,    +   642, -   482,    +   642, -  1602, 
   +   642, -  1602,    +   642, -   482,    +   638, -   484,    +   642, -  1600, 
   +   642, -   486,    +   642, -   482,    +   642, -  1602,    +   642, -   482, 
   +   642, -   482,    +   642, -  1602,    +   642, -  1602,    +   642, -   484, 
   +   642, -  1602,    +   642, -   482,    +   642, -   482,    +   642, -   482, 
   +   642, -  1602,    +   642, -  1600,    +   642, -  1602,    +   642, -  1602, 
   +   642, -  1602,    +   642, -  1602,    +   642, -  1602,    +   642, -  1600, 
   +   642, -   482,    +   642, -   482,    +   640, -   482,    +   642, -   482, 
   +   642, -   482,    +   642, -  1602,    +   642, -   484,    +   642, -  1602, 
   +   638, -  1604,    +   642, -  1602,    +   642, -   482,    +   642, -   484, 
   +   642, -   482,    +   640, -   482,    +   642, -  1602,    +   642, -   482, 
   +   642, -   482,    +   642, -   484,    +   640, -  1602,    +   642, -  1602, 
   +   640, -  1604,    +   642

uint16_t rawData[99] = {4548, 4444,  642, 1608,  642, 482,  642, 1602,  642, 1602,  642, 482,  638, 484,  642, 1600,  642, 486,  642, 482,  642, 1602,  642, 482,  642, 482,  642, 1602,  642, 1602,  642, 484,  642, 1602,  642, 482,  642, 482,  642, 482,  642, 1602,  642, 1600,  642, 1602,  642, 1602,  642, 1602,  642, 1602,  642, 1602,  642, 1600,  642, 482,  642, 482,  640, 482,  642, 482,  642, 482,  642, 1602,  642, 484,  642, 1602,  638, 1604,  642, 1602,  642, 482,  642, 484,  642, 482,  640, 482,  642, 1602,  642, 482,  642, 482,  642, 484,  640, 1602,  642, 1602,  640, 1604,  642};  // COOLIX B21FB8
uint64_t data = 0xB21FB8;
```
I suspect the Carrier Protocol is not exactly Coolix Protocol. Btw, I also had a look on the ir_Carrier.cpp which said the message contains only 32 bits, but it is sent three(3) times. i.e. normal + inverted + normal. That is not the same to what I found so I reckon I have to do in this way to build my own array.

crankyoldgit, 2019-03-14:
Try it with a repeat. e.g.

```
// 0xB21FB8 with one repeat. i.e. 2 messages back to back.
irsend.sendCOOLIX(0xB21FB8, kCoolixBits, 1);
```

or
```
ac.send(1);  // Send via the class, but with a single repeat.
```

simonfongnt, 2019-03-14:
That's a nice trick. But there are some differences in the packet and unable to turn on the AC...
```
{ 4536, 4460,  642, 1610,  638, 486,  642, 1602,  642, 1602,  642, 482,  642, 482,  642, 1602,  642, 486,  642, 482,  642, 1602,  642, 482,  640, 482,  642, 1602,  642, 1602,  640, 484,  642, 1602,  642, 482,  640, 484,  640, 482,  642, 1602,  642, 1602,  640, 1604,  642, 1602,  642, 1602,  640, 1602,  642, 1602,  640, 1604,  642, 482,  642, 484,  642, 482,  640, 482,  642, 484,  640, 1602,  638, 486,  642, 1602,  636, 1608,  642, 1602,  642, 482,  640, 484,  640, 484,  640, 482,  642, 1602,  640, 484,  638, 484,  640, 484,  640, 1604,  638, 1604,  640, 1602,  642, 4966,  
  4558, 4408,  638, 1606,  638, 486,  640, 1604,  638, 1606,  646, 478,  640, 482,  640, 1604,  642, 482,  640, 484,  640, 1604,  630, 494,  638, 484,  638, 1604,  638, 1604,  640, 484,  636, 1610,  640, 484,  636, 486,  640, 484,  638, 1604,  634, 1608,  638, 1606,  634, 1610,  634, 1612,  638, 1606,  638, 1606,  632, 1610,  632, 492,  634, 490,  634, 492,  636, 486,  632, 492,  632, 1612,  634, 490,  638, 1606,  636, 1608,  636, 1608,  640, 486,  632, 492,  628, 496,  636, 488,  632, 1610,  634, 488,  628, 496,  634, 490,  626, 1618,  636, 1608,  632, 1612,  628};  // COOLIX B21FB8
```
do you have any clue?
Is it possible to vary the 4966 between the repeats? 
or where can I found these kinds of commands?

crankyoldgit, 2019-03-15:
Yes, it is possible to change that gap my modifying `kCoolixMinGap` here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_Coolix.cpp#L42

I'm surprised that the current value (5040) is not compatible. Your original values average around 5150-ish and I would have thought they would be close enough.

Let us know what value works for you and we might be able to update it in the library.

P.S. I'm not sure what the raw data you've provided immediately above is for. Is it a signal that doesn't work, or one that does? How was it generated? What is that sequence supposed to do? etc etc.

simonfongnt, 2019-03-15:
Yes, it is the value with 
`irsend.sendCOOLIX(0xB21FB8, kCoolixBits, 1);`

I retried your code and can control my AC now. The problem was my IR emitter was not close enough. Sorry to waste your time and thanks for the great library.
```
uint64_t data[12] = {
  0xB21F28, // 20C
  0xB21F68, // 21C
  0xB21F78, // 22C
  0xB21F58, // 23C
  0xB21F48, // 24C
  0xB21FC8, // 25C
  0xB21FD8, // 26C
  0xB21F98, // 27C
  0xB21F88, // 28C
  0xB21FA8, // 29C
  0xB21FB8, // 30C
  0xB27BE0 // OFF
};
```
Your suggested command made MQTT control available.
Instead of irsend.sendRaw(), is there any generic command for type input? 
Something like `irsend.send('coolix', 0xB21FB8)`

crankyoldgit, 2019-03-16:
FYI, v2.5.6 has just been release which includes the changes/improvements mentioned.

crankyoldgit, 2019-03-16:
_(Apologies, it seems I never clicked send on this earlier)_
> Instead of irsend.sendRaw(), is there any generic command for type input?
> Something like `irsend.send('coolix', 0xB21FB8)`

Yes, there is. `IRsend::send(decode_type_t type, uint64_t data, uint16_t nbits)`
See: https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L499

closed 2019-03-16 by crankyoldgit