crankyoldgit/IRremoteESP8266 #641: LEGO PF

mike2nl, 2019-03-16:
@markszabo 
Hi,
No issue at all, a question only.
- Why is the Lego Power Function not added in your repository?
- In z3t0's (march 2018) and crankyoldgit's (april 2017) repo i find them back.

Ok yours is the most accurate and actual version of the IR control for the ESP.
Is there any issue with the Lego PF Function, or has no user asked about it?
- An issue: you have any information? Possible i can fix it.
- No user asked: i will do the job.

Thanks for any information and/or advise.

crankyoldgit, 2019-03-16:
I can't remember why I didn't move it over. I'll take another look at it and see what can be done.

crankyoldgit, 2019-03-16:
Looking at the code again, I can see why I didn't move it over into v2.0+. It doesn't have a decoder and the code was difficult to read, so it got put in the "I'll get around to it eventually" basket.

You're the first person to notice.

Do you have any raw data or specifications for the protocol at all? If I have those, I can write a clean(er) implementation of it.

mike2nl, 2019-03-16:
@crankyoldgit 
Thanks for your effort.
I will take a look in what i have to help out.

mike2nl, 2019-03-16:
@crankyoldgit 
my server was faster to find the specsa for it then i with searching, haha.
Here it is. Look at page 6 there it starts.
I hope it helps...

BTW: LEGO is a Registrated Trademark ;-)

Doc: [LEGO_Power_Functions_RC_v120.pdf](https://github.com/markszabo/IRremoteESP8266/files/2974525/LEGO_Power_Functions_RC_v120.pdf)

crankyoldgit, 2019-03-16:
After quickly skimming the protocol spec, I can now see why the author of the previous implementation didn't write a decoder for it. I'm not sure this library could easily decode it AND still auto detect all the other different protocols it supports.

For the time being, I'll look at just converting the existing code over "as is".

Do you have a capture of the real remote from IRrecvDumpV2 that you can supply?

mike2nl, 2019-03-16:
@crankyoldgit 
Thanks for the information.
No i have no capture because my ir shield for the wemos d1 mini v3.1.0 i have not received. I think it will be next weekend. Then i can do a test with it.

crankyoldgit, 2019-03-20:
@mike2nl I've added lego support in the following branch: https://github.com/markszabo/IRremoteESP8266/tree/lego

Please try it out and let me know how you go.

mike2nl, 2019-03-20:
@crankyoldgit 
wow, yes will test it on friday evening or statyrday, because the ir shiled for the Wemos D1 mini arrives on friday afternoon.
Cool work!

closed 2019-03-21 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

PR #655 (2019-03-21): Add LEGO(TM) Power Functions IR protocol.
Add support for sending & receiving LEGO Power Functions IR messages.
* Unit tests
  - Output compared to IRremote library's old function. Appears compatible.
* Written from just the spec, so should be correct.
* Examples & tools code updated.

Fixes #641
commit 2cfc2d7d (2019-03-21): Add LEGO(TM) Power Functions IR protocol. (#655)

Add support for sending & receiving LEGO Power Functions IR messages.
* Unit tests
  - Output compared to IRremote library's old function. Appears compatible.
* Written from just the spec, so should be correct.
* Examples & tools code updated.

Fixes #641