crankyoldgit/IRremoteESP8266 #642: Problem in last 4 bytes . (Electra IR encoding)

AsimZulfiqar67, 2019-03-16:
### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

### Expected behavior
I have done reverse engineering on Electra encoding and have found the bits for the temp, swing, mode, fan but I am unable to figure out the last 4 bytes, which I think are the checksum values? can anyone help me find out what are those values and how they are calculated?

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

Timestamp : 000025.190
Encoding  : ELECTRA_AC
Code      : C309070004000400000000A01E (104 bits)
Library   : v2.5.5

Raw Timing[211]:
   +  9200, -  4288,    +   738, -  1506,    +   730, -  1522,    +   704, -   412, 
   +   658, -   454,    +   648, -   460,    +   644, -   466,    +   638, -  1606, 
   +   632, -  1612,    +   622, -   488,    +   608, -   510,    +   582, -   534, 
   +   564, -   544,    +   552, -  1690,    +   544, -   564,    +   540, -   566, 
   +   542, -  1700,    +   542, -   566,    +   540, -   566,    +   540, -   568, 
   +   542, -   566,    +   538, -   568,    +   538, -  1704,    +   542, -  1700, 
   +   542, -  1700,    +   544, -   564,    +   548, -   560,    +   544, -   564, 
   +   542, -   564,    +   540, -   568,    +   548, -   562,    +   536, -   570, 
   +   546, -   562,    +   542, -   564,    +   542, -   564,    +   542, -   564, 
   +   544, -   564,    +   542, -   566,    +   544, -  1700,    +   538, -   570, 
   +   544, -   564,    +   544, -   564,    +   544, -   564,    +   546, -   560, 
   +   542, -   566,    +   542, -   566,    +   546, -   562,    +   542, -   566, 
   +   540, -   566,    +   544, -   564,    +   546, -   562,    +   546, -   562, 
   +   546, -   560,    +   548, -   560,    +   542, -  1700,    +   548, -   560, 
   +   544, -   562,    +   542, -   566,    +   542, -   564,    +   544, -   564, 
   +   548, -   560,    +   546, -   562,    +   546, -   562,    +   548, -   560, 
   +   542, -   566,    +   542, -   564,    +   542, -   566,    +   542, -   564, 
   +   544, -   564,    +   544, -   564,    +   546, -   562,    +   546, -   560, 
   +   540, -   568,    +   544, -   562,    +   542, -   566,    +   544, -   564, 
   +   548, -   560,    +   546, -   562,    +   546, -   560,    +   548, -   560, 
   +   548, -   558,    +   546, -   562,    +   542, -   564,    +   542, -   566, 
   +   540, -   568,    +   540, -   568,    +   544, -   564,    +   544, -   564, 
   +   546, -   560,    +   546, -  1698,    +   546, -   562,    +   542, -  1702, 
   +   542, -   564,    +   546, -   562,    +   544, -   564,    +   538, -   568, 
   +   544, -   562,    +   542, -   564,    +   546, -   562,    +   546, -   562, 
   +   548, -  1696,    +   544, -  1698,    +   548, -  1696,    +   550, -  1694, 
   +   546, -   562,    +   544

uint16_t rawData[211] = {9200, 4288,  738, 1506,  730, 1522,  704, 412,  658, 454,  648, 460,  644, 466,  638, 1606,  632, 1612,  622, 488,  608, 510,  582, 534,  564, 544,  552, 1690,  544, 564,  540, 566,  542, 1700,  542, 566,  540, 566,  540, 568,  542, 566,  538, 568,  538, 1704,  542, 1700,  542, 1700,  544, 564,  548, 560,  544, 564,  542, 564,  540, 568,  548, 562,  536, 570,  546, 562,  542, 564,  542, 564,  542, 564,  544, 564,  542, 566,  544, 1700,  538, 570,  544, 564,  544, 564,  544, 564,  546, 560,  542, 566,  542, 566,  546, 562,  542, 566,  540, 566,  544, 564,  546, 562,  546, 562,  546, 560,  548, 560,  542, 1700,  548, 560,  544, 562,  542, 566,  542, 564,  544, 564,  548, 560,  546, 562,  546, 562,  548, 560,  542, 566,  542, 564,  542, 566,  542, 564,  544, 564,  544, 564,  546, 562,  546, 560,  540, 568,  544, 562,  542, 566,  544, 564,  548, 560,  546, 562,  546, 560,  548, 560,  548, 558,  546, 562,  542, 564,  542, 566,  540, 568,  540, 568,  544, 564,  544, 564,  546, 560,  546, 1698,  546, 562,  542, 1702,  542, 564,  546, 562,  544, 564,  538, 568,  544, 562,  542, 564,  546, 562,  546, 562,  548, 1696,  544, 1698,  548, 1696,  550, 1694,  546, 562,  544};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x09, 0x07, 0x00, 0x04, 0x00, 0x04, 0x00, 0x00, 0x00, 0x00, 0xA0, 0x1E};

	

Timestamp : 000233.587
Encoding  : ELECTRA_AC
Code      : C309070004000400000400A019 (104 bits)
Library   : v2.5.5

Raw Timing[211]:
   +  9178, -  4314,    +   716, -  1542,    +   662, -  1586,    +   650, -   460, 
   +   642, -   466,    +   636, -   472,    +   630, -   478,    +   620, -  1624, 
   +   612, -  1632,    +   600, -   508,    +   584, -   524,    +   570, -   550, 
   +   548, -   560,    +   552, -  1692,    +   546, -   566,    +   542, -   566, 
   +   542, -  1694,    +   546, -   568,    +   538, -   570,    +   540, -   568, 
   +   536, -   566,    +   546, -   564,    +   544, -  1696,    +   546, -  1694, 
   +   548, -  1700,    +   542, -   564,    +   542, -   568,    +   540, -   566, 
   +   548, -   558,    +   548, -   558,    +   546, -   564,    +   544, -   564, 
   +   544, -   564,    +   548, -   560,    +   548, -   560,    +   544, -   566, 
   +   540, -   564,    +   542, -   564,    +   544, -  1698,    +   546, -   564, 
   +   540, -   568,    +   542, -   566,    +   538, -   566,    +   546, -   562, 
   +   546, -   564,    +   546, -   556,    +   552, -   558,    +   546, -   564, 
   +   538, -   570,    +   542, -   566,    +   540, -   568,    +   542, -   566, 
   +   546, -   558,    +   544, -   562,    +   546, -  1696,    +   546, -   568, 
   +   542, -   568,    +   540, -   564,    +   542, -   566,    +   542, -   566, 
   +   540, -   564,    +   546, -   560,    +   548, -   560,    +   546, -   562, 
   +   542, -   564,    +   546, -   564,    +   540, -   566,    +   546, -   562, 
   +   548, -   558,    +   550, -   560,    +   548, -   562,    +   544, -   560, 
   +   546, -   564,    +   540, -   568,    +   540, -   566,    +   546, -   552, 
   +   552, -   560,    +   542, -   568,    +   540, -  1702,    +   542, -   560, 
   +   544, -   568,    +   542, -   564,    +   546, -   562,    +   542, -   558, 
   +   550, -   564,    +   544, -   566,    +   542, -   564,    +   540, -   566, 
   +   552, -   552,    +   546, -  1700,    +   550, -   556,    +   548, -  1686, 
   +   556, -   564,    +   546, -   556,    +   554, -   556,    +   544, -   566, 
   +   544, -   566,    +   542, -   562,    +   550, -   562,    +   540, -   560, 
   +   550, -  1690,    +   552, -  1696,    +   548, -   564,    +   544, -   560, 
   +   548, -  1698,    +   542

uint16_t rawData[211] = {9178, 4314,  716, 1542,  662, 1586,  650, 460,  642, 466,  636, 472,  630, 478,  620, 1624,  612, 1632,  600, 508,  584, 524,  570, 550,  548, 560,  552, 1692,  546, 566,  542, 566,  542, 1694,  546, 568,  538, 570,  540, 568,  536, 566,  546, 564,  544, 1696,  546, 1694,  548, 1700,  542, 564,  542, 568,  540, 566,  548, 558,  548, 558,  546, 564,  544, 564,  544, 564,  548, 560,  548, 560,  544, 566,  540, 564,  542, 564,  544, 1698,  546, 564,  540, 568,  542, 566,  538, 566,  546, 562,  546, 564,  546, 556,  552, 558,  546, 564,  538, 570,  542, 566,  540, 568,  542, 566,  546, 558,  544, 562,  546, 1696,  546, 568,  542, 568,  540, 564,  542, 566,  542, 566,  540, 564,  546, 560,  548, 560,  546, 562,  542, 564,  546, 564,  540, 566,  546, 562,  548, 558,  550, 560,  548, 562,  544, 560,  546, 564,  540, 568,  540, 566,  546, 552,  552, 560,  542, 568,  540, 1702,  542, 560,  544, 568,  542, 564,  546, 562,  542, 558,  550, 564,  544, 566,  542, 564,  540, 566,  552, 552,  546, 1700,  550, 556,  548, 1686,  556, 564,  546, 556,  554, 556,  544, 566,  544, 566,  542, 562,  550, 562,  540, 560,  550, 1690,  552, 1696,  548, 564,  544, 560,  548, 1698,  542};  // ELECTRA_AC
uint8_t state[13] = {0xC3, 0x09, 0x07, 0x00, 0x04, 0x00, 0x04, 0x00, 0x00, 0x04, 0x00, 0xA0, 0x19};

#### Example code used
I am using IRrecvDumpV2.ino to read raw data.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Other useful information
_More information is always welcome. Be verbose._

**A019** & **A01E** are the values , i need to know about and i want to know how they will be calculated. I have tried the bit reversal method but i cannot get the answer what i expected.

AsimZulfiqar67, 2019-03-16:
C3020000040004000004008026 // 16
C31200000400040000040000D6 // 17
C30A00000400040000040000CE //18
C31A00000400040000040000DE //19
C30600000400040000040000C1 //20
C31600000400040000040000D1 // 21
C30E00000400040000040000C9 // 22
C31E00000400040000040000D9 //23
C30100000400040000040000C5 //24

C3110000040004000004008035 // 25 set
C3110700040004000004008031 //25 on
C311070004000400000000A00E //25 off

C30900000400040000040000CD // 26 set
C309070004000400000000A01E //26 off
C309070004000400000400A019 //26 on

C31900000400040000040000DD // 27 set

C3E00000050000000004006009 // auto

These are some other values for different temperature and modes, you can notice last 4 bytes are always different.

crankyoldgit, 2019-03-16:
I'm pretty sure the problem is we have gotten the bit ordering wrong on this protocol.
It makes more sense when you reverse the order of the bits in each byte.
Instead of MSB order (0xC309070004000400000400A019) you look at it in LSB order [per byte] (0xC390E000200020000020000598) then 0xC3+0x90+0xE0+0x00+0x20+0x00+0x20+0x00+0x00+0x20+0x00+0x00+0x05 = 0x(2)98. i.e. the last byte.

AsimZulfiqar67, 2019-03-16:
I do not understand how you changed (0xC309070004000400000400A019) to (0xC390E000200020000020000598). 

I tried the method you told and answers are not matching , can you please explain this ?

crankyoldgit, 2019-03-17:
Try the code in this branch: https://github.com/markszabo/IRremoteESP8266/tree/Issue642
It will use the _correct_ bit ordering now for the state.
I've also added checksum verification.

Any data collected with older versions of Electra support will be wrong. You'll need to re-capture or convert the arrays according.

Hopefully it will make your temperature analysis a little more easier.

None of this however will explain why the second-last byte is changing. You're still on your own with that.

AsimZulfiqar67, 2019-03-19:
Thank you so much :) I got it working and decoded everything :)

closed 2019-03-19 by AsimZulfiqar67

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.