crankyoldgit/IRremoteESP8266 #643: Raw timestamps work, Sending the command has no effect

DasVoelkel, 2019-03-16:
master today

### Expected behavior
copied the raw and command from an infrared signal.
this worked as expected and as it should.
 but when sending the IR signal the device reacts if i sent the signal via the raw timestamps.
but my device does not react if i use the command string.
Encoding  : PANASONIC
Code      : 40040538BC81 (48 bits)
this doesn't work. the time stamps as uint16 array work perfectly every time

### Actual behavior
the signal is ignored by my device, i am not sure where the error lies here since the raw timestamps achieve the desired effect. (but they occupy so much data, that it is more problamatic to process further.

### Steps to reproduce the behavior
scan a parasonic device, the raw timestamps work for immitating the remote.
```
char *p;
int number = strtol( String("40040538BC81").c_str(), &p, 16 );
 irsend.send(5 (parasonic),number);
```
With the raw timestamps:
```
std::vector<uint16_t> L = [3498,1690,478,392,478,1264,476,394,476,396,484,386,484,388,482,388,480,390,478,392,478,394,476,394,476,396,474,396,484,1258,480,390,480,390,478,392,478,394,476,394,476,396,484,386
,484,1258,480,390,478,1264,476,394,506,364,484,1258,482,1260,478,1264,476,394,476,394,486,386,484,1258,480,390,486,1256,478,1264,476,1266,472,1270,480,390,478,392,478,1264,476,396,478,396,480,388,482,396,472,390
,480,390,478,1264,476];
uint16_t tmp[L.size()];

        std::copy(L.begin(), L.end(), tmp);

        irsend.sendRaw(tmp,L.size(),38);
```

are the raw timestamps, and works

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes_

### Has this library/code previously worked as expected for you?
_Yes. If "Yes", which version last worked for you?_ the current one (downloaded today, not vie the arduino library function)Everything up until now worked just as expected

### Other useful information
AC is on, i added all headers i could find to apply to as many devices as possible.

crankyoldgit, 2019-03-16:
Your problem is most likely here:
```
int number = strtol( String("").c_str(), &p, 16 );
```

You've made a number of programming/data type & size errors.

1. `int` is `int16_t` i.e. a _signed_ **16** bit integer on the Arduino platform.
2. `strtol()` results in a `long`. i.e. a _signed_ **32** bit integer on the Arduino platform.
3.  `0x40040538BC81` is a **48** bit value. You need a `long long`, preferably a `unsigned long long` (aka `uint64_t`) to store an integer that big.

Basically, you are trying to shove 48 bits, into 32 bits, and then in 16 bits. You've lost 2/3rds of the information.

Next problem:
`irsend.send(5 (parasonic),number);` is incorrect, you also need to pass the number of bits.
See the arguments required here: https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L499

If you try the following you'll probably find it works fine:
```
irsend.sendPanasonic64(0x40040538BC81);
```
or
```
irsend,send(PANASONIC, 0x40040538BC81, kPanasonicBits);
```

DasVoelkel, 2019-03-17:
thanks a lot!
.ok so i have something wrong with converting the number back.
I had my suspicion because it just wouldn't work for me. 
originally i save the cmd as a string:
40040538BC81, if i create a uint64_t and shove each of the values into it with bit shift one at a time, would that work or would i lose data again?

crankyoldgit, 2019-03-18:
Sadly the Arduino library doesn't have a `strtoull()` function. (ull == unsigned long long == uint64_t)
I had to write my own. Feel free to steal my routine from here: https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L1337

crankyoldgit, 2019-03-20:
Marking this closed for now.

closed 2019-03-20 by crankyoldgit

DasVoelkel, 2019-03-22:
@crankyoldgit ok, it works now, I just converted the numbers with a union and managed to recreate the value correctly.
For some reason it will still not react, but I only have a remote control which identifies as "unknown".
So i am not sure if i am sending it correctly, or how i send that at all.

crankyoldgit, 2019-03-22:
If you can please supply the test dump for the `UNKNOWN` message I may be able to help you further.