crankyoldgit/IRremoteESP8266 #645: IRMQTTServer Enhancement - allow more than one GPIO Pin to be used for IR Sending

sheppy99, 2019-03-16:
I use IRMQTTServer to send IR. Is it possible to expand it to use more than one IR Emitter?
Perhaps the command sent to it could add an extra parameter on the end if a different PIN to default is wanted for sending?

crankyoldgit, 2019-03-17:
I'll see what I can do.
I think allowing the MQTT message to set the GPIO is dangerous.
How about we have between 1-4 MQTT topics which can be assigned to a GPIO at compile time. Would that suffice?

sheppy99, 2019-03-17:
Absolutely. I’ve got 4 separate IR channels currently including the Daikin2

crankyoldgit, 2019-04-02:
@sheppy99 what are the four GPIO's you are using?

sheppy99, 2019-04-02:
Its a new project so entirely up to you.
For less "why doesn't this work" questions I'd suggest avoiding the serial Pins (GPIO 1 & 3 , the SOnOff Relay Pin (GPIO12), the SOnOff LED Pin (GPIO13) and those that screw up booting if held in the wrong state (GPIO 0, 2, 15)
This leaves GPIO 4, 5, 14, 16
With care you can of course use all pins which I have done a few times :-)

crankyoldgit, 2019-04-02:
Thanks.

You're the one requesting/wanting the feature, I might as well make it default to what you are using. ;-)

sheppy99, 2019-04-02:
That’s an easy one to answer. I’m only using one so far which is the default in MQTTServer, I can’t remember which pin that is without looking at the sketch. I made the board the Huzzah is plugged into have all pins available as experience tells me I’ll usually expand things

crankyoldgit, 2019-04-02:
FYI, you're going to be the one who gets to test the multi-channel functionality. ;-) I'm not going to hook up extra LEDs for this. :-P

sheppy99, 2019-04-02:
All good! Very happy to help. 
Incidentally so far I've used OpenHAB to manufacture a valid 53,ABCDEF1234 formatted packet which is sent to the ESP to be output as IR. For my purposes the current way works well. However if it makes sense to use it a different way to help with testing can you point me towards any documentation? So far a lot of your decoding work is only used if I use it to receive IR

crankyoldgit, 2019-04-04:
PR #645 has been merged into master. This should fix this feature request.
After testing, please let me if it doesn't and I'll re-open this.

closed 2019-04-04 by crankyoldgit

sheppy99, 2019-04-04:
Thanks very much!

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.