crankyoldgit/IRremoteESP8266 #646: IRMQTTServer using last known WiFi AP instead of strongest AP

sheppy99, 2019-03-16:
I use IRMQTTServer to send IR. I have 2 AP's with the same SSID. This allows devices to seamlessly move to the strongest AP as they are moved around the house. Unfortunately IRMQTTServer only connects to the first AP it has seen, and despite reconfiguring, it won't connect to its nearest AP.
Is there a way of either specifying a MAC Address it should use or getting it to use the strongest one?

crankyoldgit, 2019-03-17:
If you can find me some nice/simple example code elsewhere I can look at adding it. Otherwise this will take a while, as it's really a low priority for _example code_ ;-)

sheppy99, 2019-03-17:
All good, it’s something Tasmota added a while ago which is useful. It has 2 things that can be enabled, one option scans for a stronger signal every 44 minutes and the other scans for the strongest after a reboot.

scropion86, 2019-03-18:
here is the commit for this feature in tasmota repo [LINK](https://github.com/gemu2015/Sonoff-Tasmota/commit/444659cad5d3c436902e349b8d8b59200145e931)

and the issue [LINK](https://github.com/arendst/Sonoff-Tasmota/issues/4704)

i hope this can make it easier to you

crankyoldgit, 2019-03-18:
@scropion86 Sadly, that not the change/code necessary to do this. It's far more complicated than that particular commit.

scropion86, 2019-03-18:
@crankyoldgit  i agree it can't be that simple but [that file](https://github.com/arendst/Sonoff-Tasmota/blob/06a9fbd792372905059782c478e1ca09317309c5/sonoff/support_wifi.ino) has the logic

crankyoldgit, 2019-03-18:
@sheppy99 I had a quick look over the WifiManager code, which that example code uses. As far as I understand it, it should connect to a different AP as long it has the same SSID (and passphrase obviously).
Maybe it sorts the BSSIDs (think of it as the MAC address for the base station) internally. No idea.

It looks like you can filter out low-quality base stations (APs) with: `wifiManager.setMinimumSignalQuality(percent);`
See: https://github.com/tzapu/WiFiManager#filter-networks
No idea if that is at setup/config time or not. You'll probably need to [visit the "/reset" URL](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L45) to get back to that mode, or fully erase the flash on the chip

sheppy99, 2019-03-31:
In the end I used the Erase WiFi Config option on Arduino IDE, turned Off the AP I didn't want it to attach to, then reflashed and reconfigured. Now it seems to be sticking on the wanted AP and ignoring the other AP even though I've reenabled it.
Kind of a nuclear option but hopefully the many disconnections have gone. Its been 14 hours now without one and I'll update if it remains stable after a few days.

crankyoldgit, 2019-04-01:
Did you try the instructions listed in the comments of the code? e.g. https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L45

```
 * If you need to reset the WiFi settings, visit:
 *   http://<your_esp8266's_ip_address>/reset
```

sheppy99, 2019-04-01:
I did and it reconnected to the same wrong Access Point which is weaker in that location, but was stronger when I first installed the ESP

crankyoldgit, 2019-04-01:
Thanks for the feedback. I thought it would/should go back to the initial setup configuration. i.e. the ESP hosts it's own AP etc for config. Did that not happen? If not, that's a bug.

sheppy99, 2019-04-01:
It did go back to the select the wireless network AP mode, but when I selected it, it reconnected to the weaker AP - I have 2 AP’s with different channels and the same SSID

crankyoldgit, 2019-04-01:
Oh phew. No bug then. But interesting behaviour. Perhaps if you had turned off the undesired AP while it was re-setting up etc it may have worked okay too, rather than resorting to wiping the flash etc.

sheppy99, 2019-04-01:
googling suggests it may store the first AP it finds and then tries to reconnect to it regardless of its signal strength. I’ve got a couple of WEMO Insights and one of those seems to be working the same way

crankyoldgit, 2019-04-22:
I think I found the issue.
I've basically added: `wifiManager.setRemoveDuplicateAPs(false);` per https://github.com/tzapu/WiFiManager#filter-networks

I think that should allow you to select the correct AP on first boot/after a full wipe etc.

But yes, it seems it's sticky to the AP initially chosen. I've added everything I can easily do to allow you to steer it towards the AP you want, without re-inventing the wheel.

closed 2019-04-22 by crankyoldgit

sheppy99, 2019-04-22:
Nice one, can’t test for a few weeks however

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

PR #679 (2019-04-22): Add Wifi filtering options to IRMQTTServer.
- Add the ability to filter on Wifi signal strength.
- Show all Wifi networks in WifiManager config interface. (configurable)
  * Should allow users to select the AP/SSID they want when there are duplicates.
- Block Serial debug output if we know a IR gpio is using the Serial TX Gpio.
- Reorder and group up configuration options into logical sections.
- Add details to the /info page.

Limited testing on my personal device. i.e. Things still work. Debug disables when it should.

For #645
Fixes #646
commit c68c863d (2019-04-22): Add Wifi filtering options to IRMQTTServer. (#679)

- Add the ability to filter on Wifi signal strength.
- Show all Wifi networks in WifiManager config interface. (configurable)
  * Should allow users to select the AP/SSID they want when there are duplicates.
- Block Serial debug output if we know a IR gpio is using the Serial TX Gpio.
- Reorder and group up configuration options into logical sections.
- Add details to the /info page.

Limited testing on my personal device. i.e. Things still work. Debug disables when it should.

For #645
Fixes #646