crankyoldgit/IRremoteESP8266 #651: Invert kRecvPin  - Low Signal instead of IR

hoernche, 2019-03-18:
Hey guys, 

I have some issue I cannot solve by myself: 

I took an old remote and removed the IR-LED. The chip on the remote is an ADAM24p20g, it pulls the LED output low - I connected the pin directly with the esp8266 to a digital input, where I used pinMode INPUT_PULLUP. 

--> I want to use the IRremoteESP8266 without IR-LED and without IR-Receiver. 

My problem now: Of course now, the results are useless. I need to invert the kRecvPin and I would like to do it in globally in the void setup, if possible.

Or do I need to chance sth. in the library? 

Do you guys have any recommendations?  

Thanks and best Regards! 

Lukas

crankyoldgit, 2019-03-18:
Given what I understand of your setup, I don't think you can use this library. The library expects a demodulated signal, not a raw IR led signal. You need a IR demodulator.

You shouldn't need to _"invert the kRecvPin"_. The library works by detecting a change on the input GPIO, not if it is HIGH or LOW, just if it CHANGEs.

hoernche, 2019-03-18:
Thank's a lot for your answer, now I understand what I did wrong. 
Somehow I don't like to put the IR Led and the IR-Receiver face-to-face on one PCB, I need to find a way to decode the output of the Adam24p20g directly... I understand the IRremoteESP8266 wouldn't help me here...

Do you have any recommendations how to tackle this problem? 

Cheers, 

Lukas

crankyoldgit, 2019-03-18:
You are going to need to demodulate the signal some how if you want to use this library. You could adapt the library, or use a small uController to demodulate the signal, or there is probably some IC designed just to do this but without an IR photo-diode. I don't know. Most people using this library use a $2 IR demodulating component and use their remote as normal.

Or you could do it all manually with a logic probe/oscilloscope or similar.

closed 2019-03-26 by crankyoldgit