crankyoldgit/IRremoteESP8266 #657: Repeating the same code when remote button is hold

GMFalka, 2019-03-30:
Hello / Szia Márk!

Is it possible to send the same IRCode / message when i keep holding the button on the remote?

Reason: Remote volume buttons :)

I saw some mentions about "bool repeat" in the code. I can JerryRig it if you can help me where should i enable/disable the repeat filter.

Regards / Köszi,
Falka

crankyoldgit, 2019-03-30:
Yes.
Most, if not all protocols support _repeat_.
See: https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_NEC.cpp#L27

e.g.
```irsend.sendNEC(code_you_want_sent, kNECBits, nr_of_times_you_want_it_to_repeat);  // 0 repeats typically means send it once, 3 repeats would send it 4 times.```

Some protocols have a special _repeat_ message. e.g. NEC. If you use the `repeat` parameter, the code knows what to do for each protocol.

closed 2019-03-30 by crankyoldgit

GMFalka, 2019-03-30:
Thanks for the quick response!

Is this repeat function means, if press the remote button once, but the code will actually send x amount of time?
Because what im looking for is to send the same ir code as long as i hold the button. 

If it is, then the question issue can be closed :)

Thanks!

crankyoldgit, 2019-03-30:
Depending on the protocol, yes.
For example, say a single press/message is 100ms long ... and if you repeat 9 times, you'll send a message 10 times. So, it's like holding it down for 1000ms (or 1 second).

Some protocols however have a different code for when a button is released. It all depends on the remote/protocol.