crankyoldgit/IRremoteESP8266 #659: Not working with the sendRC5() method in IRsendDemo

qianli2017, 2019-04-01:
### Version/revison of the library used
IRromotESP8266  Ver 2.5.6 lib.

IRremoteESP8266: IRrecvDumpV2, Version 0.4 July, 2018      (master)

IRremoteESP8266: IRsendDemo, Version 1.1 January, 2019.

NodeMCU 1.0E

### Expected behavior
I can control  an old CRT TV(TCL brand) by sending IR on a NodeMCU using IRremoteESP8266 lib. 

### Actual behavior
I have an old CRT TV(TCL brand)，I use my Android phone to send IR(from a software of HUAWEI) and can work well. So I use it for decoding IR for my TV, then the result is IR RC5 protocol. I captured all the data. 

Next, I use the IRSendDemo to send IR to my TV with two ways. 
One is that irsend.sendRaw(data, 21, 36) or irsend.sendRaw(data2, 23, 36), then works well;
The second way is that using irsend.sendRC5(0x16, 12) or irsend.sendRC5(0x11, 12),
but the TV has NO response.  

So, I use two NodeMCU with SmartIR module, one is for sending IR(IRsendDemo), the other is for receiving IR(IRrecvDumpV2). For capturing IR data, I send IR with my Android phone and NodeMCU with a module SmartIR. I get the data of two ways, but I found the data is very similar and is the same result after decoding. BUT  there are different performance, sendRaw() can work , but sendRC5() can NOT work. 

I think maybe there is a little bug for send RC5 IR method. I‘m confused by this problem.
Looking forward to your resolution, THANKS.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
[Android phone send IR]
uint16_t rawData[21] = {878, 818,  1696, 816,  912, 764,  858, 816,  912, 764,  858, 816,  914, 762,  858, 1656,  1720, 1630,  914, 762,  1750};  // RC5 16
uint32_t address = 0x0;
uint32_t command = 0x16;
uint64_t data = 0x16;

[NodeMCU send IR with sendRC5()]
uint16_t rawData[21] = {808, 900,  1622, 896,  840, 846,  812, 870,  836, 844,  842, 840,  838, 844,  838, 1680,  1624, 1736,  836, 844,  1652};  // RC5 16
uint32_t address = 0x0;
uint32_t command = 0x16;
uint64_t data = 0x16;

### Steps to reproduce the behavior
If you have an Andorid phone(HUAWEI honor6) which can send IR from its head.

#### Example code used
if(n < 6){
      //irsend.sendRC5(0x80D, 12, 1);
      irsend.sendRaw(rawData, 23, 36);
  }else if(n < 12){
      //irsend.sendRC5(0x10, 12, 1);
      irsend.sendRaw(rawData2, 21, 36);
  }

crankyoldgit, 2019-04-02:
Thanks for the report. Here are a couple of things I immediately notice.
1. You probably should be using Zero(0) repeats. i.e. `irsend.sendRC5(0x10, 12, 0);` instead of `irsend.sendRC5(0x10, 12, 1);`. Zero means the command is sent only once, One means it's sent twice. i.e. One repeat. Note: You can shorted it to `irsend.sendRC5(0x10, 12);` and maybe even `irsend.sendRC5(0x10);`.
2. The timings are very close, so it should be close enough to be the same. You can try tweaking the value `kRc5T1` at https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RC5_RC6.cpp#L25 to try to match it closer to what your TV wants/needs. I'm not sure this will make a difference.
3. It could be a duty cycle issue. `sendRC5()` uses a 25% duty cycle, while `sendRaw()` uses 50%. Try changing: `enableIROut(36, 25);` to `enableIROut(36, 50);` at https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RC5_RC6.cpp#L76

Let me know your results after trying those.

qianli2017, 2019-04-05:
Thanks for your good suggestions.  I cost my spare time to test the three ways.
Following your second and third suggestion, I resolve this problem. 

1.  I found the IR remote control of my TCL CRT TV, then I open it and found the chip of IR sender which is AS1213. I got the datasheet pdf of AS1213 by internet. I found the carrier frequency is 38kHz and duty cycle is 1/3.

2. So I changed the duty cycle:  enableIROut(36, 33), and tweaked the value kRc5T1 is 842 according ratio of 36kHz compared with 38kHz.
   kRc5T1 at https://github.com/markszabo/IRremoteESP8266/blob/master/src/ir_RC5_RC6.cpp#L25

Then the problem is resolved, the TV gives the right response after every IR transmission.

Thank you so much again!

crankyoldgit, 2019-04-05:
@qianli2017 Thanks for the feedback. Glad its working for you.
Were you able to isolate which of the `enableIROut(38, 33)` or `tweaked the value kRc5T1 is 842` worked for you, or did it require both of them for reliable operation?

qianli2017, 2019-04-05:
Yes. I did several combinational test of carrier freq, duty cycle and kRc5T1.
The list is as follows:
enableIROut(freq, duty)   kRc5T1   success
1.  enableIROut(38, 25)     889        35%
2.  enableIROut(38, 33)     889        10%
3.  enableIROut(36, 25)     842        90%
4.  enableIROut(36, 33)     842        95%
5.  enableIROut(38, 25)     842        95%
6.  enableIROut(38, 33)     842        100%    

The success rate is estimated by myself, and is not precise.
It seems that kRc5T1 is the most fatal and sensitive one in the three factors.

crankyoldgit, 2019-04-05:
Thanks for that. That's quite interesting. It seems it's quite different from the RC5 specification. Glad you found some setting that works for your specific device.

closed 2019-04-05 by crankyoldgit