crankyoldgit/IRremoteESP8266 #660: Mitsubishi Heavy Industry Air Conditioner Code from ToniA add please

tonyfitzs, 2019-04-01:
### Version/revison of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
version": "2.5.5"

#### Example code used
I love what you're trying to do with you IRremoteESP8266 project, I have a Mitsubishi Heavy Industry Air Conditioner that I would like to control with my home automation using your library. I have found another library that is able to read my remote and report back the correct codes, this library can be found on github here https://github.com/ToniA/arduino-heatpumpir, unfortunately, this library is designed to work on an Arduino Mega and I prefer to use an ESP or even a Sonoff running tasmota to control my AC 
I have read on your wiki that if we are looking to have a protocol implemented and we find a project that already exists that we should tell you. so with that in mind I would like to ask you how I might go about getting your library to work with the code set that is attached as this is the library that works with my AC. 
### Other useful information
I'm happy to do some of the work but I'm not a Profesional programmer and I have already spent several hours trying to work out how to use the code created by ToniA to work with your code. If you can point me in the right direction I would be most grateful.

crankyoldgit, 2019-04-02:
I'll try to take a look at it soon. In the mean time, can you please capture a few Raw signals from `IRrecvDumpV2` and include them here, and please add what modes are set on the remote. That will help me verify what code I port across is working etc.

I assume this (https://github.com/ToniA/arduino-heatpumpir/blob/master/MitsubishiHeavyHeatpumpIR.cpp) is the code you're referring to. Correct?
If so, what make/model number is your specific unit? e.g. What's on the label of the head unit, and what does that other code detect it as?

tonyfitzs, 2019-04-07:
Hi David,

Sorry for the delay in replying to you, I was away on a business trip. So what I have done here in an attempt to help is provided you with the data from the IRrecvDumpV2 as well as the data from the Raw-IR-decoder-for-Arduino which was created by the same author as the arduino-heatpumpir and which uses the same set of library files. I hope this is helpful and I am able to help push this project forward.

I have a Mitsubishi Heavy Industry SRK20ZMXA-S and a SRK50ZMXA-S that I would like to control, they both use the same remote control (RLA502A700B), I have attached the library file that is included with the Raw-IR-decoder-for-Arduino set, as I know this is decoding the codes corectly, the link to the original can be found here. https://github.com/ToniA/Raw-IR-decoder-for-Arduino please let me know if you need more codes or what eles I can do to help.

Below is the Dump from IRrecvDumpV2

The states are

POWER ON
MODE HEAT
Temperature: 24
FAN 4

Timestamp : 000040.054
Encoding  : UNKNOWN
Code      : A650F2C1 (154 bits)
Library   : v2.5.2

Raw Timing[307]:
   +  3136, -  1638,    +   364, -   428,    +   366, -  1224,    +   362, -   432,
   +   364, -   430,    +   364, -  1226,    +   362, -   432,    +   364, -  1224,
   +   366, -   428,    +   366, -   430,    +   366, -  1224,    +   362, -  1228,
   +   362, -  1228,    +   362, -   432,    +   364, -  1224,    +   364, -   432,
   +   364, -  1226,    +   364, -  1224,    +   366, -  1226,    +   364, -   428,
   +   364, -   430,    +   364, -   430,    +   364, -   432,    +   366, -  1226,
   +   364, -  1224,    +   364, -   430,    +   364, -  1226,    +   364, -   428,
   +   364, -  1224,    +   368, -  1224,    +   364, -   428,    +   364, -   430,
   +   366, -   430,    +   364, -  1158,    +   430, -   432,    +   366, -  1222,
   +   366, -   430,    +   366, -   430,    +   364, -  1226,    +   364, -  1224,
   +   364, -  1224,    +   364, -  1224,    +   366, -  1224,    +   364, -   430,
   +   364, -   430,    +   364, -  1228,    +   362, -  1226,    +   364, -  1226,
   +   366, -  1222,    +   366, -   430,    +   364, -   430,    +   364, -  1224,
   +   366, -  1224,    +   364, -   430,    +   364, -   430,    +   364, -   432,
   +   364, -   430,    +   364, -   428,    +   364, -   430,    +   364, -   430,
   +   366, -  1226,    +   362, -  1154,    +   434, -  1228,    +   364, -  1226,
   +   362, -  1226,    +   364, -  1226,    +   364, -  1228,    +   362, -  1226,
   +   362, -   432,    +   364, -   430,    +   364, -   428,    +   364, -   430,
   +   364, -   430,    +   364, -  1228,    +   362, -  1228,    +   362, -   432,
   +   364, -  1224,    +   368, -  1224,    +   364, -  1226,    +   362, -  1226,
   +   364, -  1226,    +   366, -   428,    +   366, -   430,    +   364, -  1224,
   +   364, -   430,    +   366, -   430,    +   366, -   430,    +   364, -   430,
   +   364, -   430,    +   364, -  1226,    +   364, -  1226,    +   366, -  1224,
   +   366, -  1224,    +   366, -  1226,    +   364, -  1224,    +   366, -  1224,
   +   366, -  1224,    +   366, -   428,    +   364, -   430,    +   366, -   428,
   +   364, -   430,    +   364, -   430,    +   366, -   428,    +   364, -   430,
   +   364, -   432,    +   364, -  1226,    +   364, -  1226,    +   364, -  1226,
   +   364, -  1228,    +   364, -  1222,    +   370, -  1222,    +   362, -  1228,
   +   362, -  1226,    +   362, -   430,    +   364, -   430,    +   364, -   430,
   +   364, -   432,    +   364, -   428,    +   364, -   432,    +   364, -   428,
   +   364, -   430,    +   366, -  1226,    +   362, -  1224,    +   364, -  1226,
   +   364, -  1226,    +   364, -  1226,    +   362, -  1226,    +   366, -  1224,
   +   366, -  1224,    +   364, -   430,    +   364, -   432,    +   364, -   428,
   +   364, -   432,    +   364, -   428,    +   364, -   430,    +   366, -   430,
   +   364, -   430,    +   364, -  1226,    +   362, -  1226,    +   364, -  1224,
   +   366, -  1226,    +   362, -  1228,    +   364, -  1224,    +   366, -  1224,
   +   364, -   430,    +   364, -   432,    +   364, -   428,    +   364, -   430,
   +   364, -   430,    +   364, -   430,    +   366, -   430,    +   364, -   430,
   +   338, -  1252,    +   362

uint16_t rawData[307] = {3136, 1638,  364, 428,  366, 1224,  362, 432,  364, 430,  364, 1226,  362, 432,  364, 1224,  366, 428,  366, 430,  366, 1224,  362, 1228,  362, 1228,  362, 432,  364, 1224,  364, 432,  364, 1226,  364, 1224,  366, 1226,  364, 428,  364, 430,  364, 430,  364, 432,  366, 1226,  364, 1224,  364, 430,  364, 1226,  364, 428,  364, 1224,  368, 1224,  364, 428,  364, 430,  366, 430,  364, 1158,  430, 432,  366, 1222,  366, 430,  366, 430,  364, 1226,  364, 1224,  364, 1224,  364, 1224,  366, 1224,  364, 430,  364, 430,  364, 1228,  362, 1226,  364, 1226,  366, 1222,  366, 430,  364, 430,  364, 1224,  366, 1224,  364, 430,  364, 430,  364, 432,  364, 430,  364, 428,  364, 430,  364, 430,  366, 1226,  362, 1154,  434, 1228,  364, 1226,  362, 1226,  364, 1226,  364, 1228,  362, 1226,  362, 432,  364, 430,  364, 428,  364, 430,  364, 430,  364, 1228,  362, 1228,  362, 432,  364, 1224,  368, 1224,  364, 1226,  362, 1226,  364, 1226,  366, 428,  366, 430,  364, 1224,  364, 430,  366, 430,  366, 430,  364, 430,  364, 430,  364, 1226,  364, 1226,  366, 1224,  366, 1224,  366, 1226,  364, 1224,  366, 1224,  366, 1224,  366, 428,  364, 430,  366, 428,  364, 430,  364 [... cut ...]
      case 0x00:
        Serial.println(F("ON"));
        break;
      case 0x80:
        Serial.println(F("OFF"));
        break; 
    }   

    // Clean and alergen
    Serial.print(F("Clean or Alergen: "));
    switch (bytes[5] & 0x60) {
      case 0x00:
        Serial.println(F("CLEAN"));
        break;
      case 0x20:
        Serial.println(F("ALERGEN"));
        break; 
      case 0x60:
        Serial.println(F("OFF"));
        break;         
    } 
    return true;
  }  
  return false;
}

crankyoldgit, 2019-04-07:
Thanks for all of that. I'll see what I can do shortly.

tonyfitzs, 2019-04-07:
Thank you

Sent from my SAMSUNG Galaxy S7 edge on the Telstra Mobile Network

crankyoldgit, 2019-04-08:
@tonyfitzs I've just created a branch which should support your A/Cs. If you care to try the `mitsub-heavy` branch (https://github.com/markszabo/IRremoteESP8266/tree/mitsub-heavy) it may work. At the least you can provide me with feedback on what does or doesn't work.

tonyfitzs, 2019-04-08:
Thanks heaps, I will give it a try tonight after work

Sent from my SAMSUNG Galaxy S7 edge on the Telstra Mobile Network

tonyfitzs, 2019-04-10:
Hi David,

Thanks for all your efforts so far, I have spent several hours trying to get this to work, but for some stupid reason I am not able to load the library, I keep getting the error ir_MitsubishiHeavy.h: No such file or directory I'm not sure why. I have attached the Arduino code I have been testing with.

Also are you able to tell me how I work out how to set the different modes? I am using the TurnOnMitsubishiAC example as a base to get this working, and I can't seem to figure out how to define the different modes, not that it matters much at the moment as I can't even get the thing to load on my ESP.

crankyoldgit, 2019-04-10:
> Thanks for all your efforts so far, I have spent several hours trying to get this to work, but for some stupid reason I am not able to load the library, I keep getting the error ir_MitsubishiHeavy.h: No such file or directory I'm not sure why.

That probably is the result of how you've downloaded (and where you've put) the code from that branch.
`ir_MitsubishiHeavy.h` currently only exists in the branch: https://github.com/markszabo/IRremoteESP8266/tree/mitsub-heavy

It's been a while since I used a custom branch with the Arduino IDE (assuming that's what you are using) but I think you need to download the ZIP file from that link, and remove the old "normal" files from and put them in place of them. It should then be able to find it.

> I have attached the Arduino code I have been testing with. Also are you able to tell me how I work out how to set the different modes? I am using the TurnOnMitsubishiAC example as a base to get this [...]

No code attached that I could see.
But here is roughly how you change modes etc:
https://github.com/markszabo/IRremoteESP8266/blob/mitsub-heavy/test/ir_MitsubishiHeavy_test.cpp#L301

crankyoldgit, 2019-04-10:
@sheppy99 may be better at explaining how to download and use custom branches, as he does it frequently.

tonyfitzs, 2019-04-10:
Thanks David,

I will try again doing as you have sugested, but that is exactly what I did. I should note that when I compile any of the examples they work fine it's only when I change the code to include the Mitsubishi file you created that I get the error.

I will keep trying

Sent from my SAMSUNG Galaxy S7 edge on the Telstra Mobile Network

https://github.com/markszabo/IRremoteESP8266/blob/mitsub-heavy/test/ir_MitsubishiHeavy_test.cpp#L301

crankyoldgit, 2019-04-10:
My _guess_ is there is a release version of the library the Arduino IDE is _still_ finding or using.

tonyfitzs, 2019-04-11:
Doh.. (Face plant into palm) yes of course, how stupid of me. As soon as you said that it dawned on me what the problem is. I have now successfully uploaded it and will do some tests and get back to you laiter today or on the weekend.

tonyfitzs, 2019-04-11:
Hi David,

I'm getting closer, but I'm struggling a little with testing as I can't work out how to set up the sketch to send the commands, I have used the TurnOnMitsubishiAC.ino (see below) as a base, but it appears that the library for the Mitsubishi is a lot different than the one for the Mitsubishi heavy, for example when I compile the code the Compiler barks with the error 'class IRMitsubishiHeavy152Ac' has no member named 'send', its because in the code below in the loop section there is a command   “ac.send();” which appears to send the settings that are configured in the setup, I'm not to sure how that is references but it works.

Are you able to help me understand how to send commands? If you use a different method other than Arduino to compile code to an Arduino then I'm happy to try it.

void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send. See ir_Mitsubishi.cpp for all the options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting desired state for A/C.");
  ac.on();
  ac.setFan(1);
  ac.setMode(kMitsubishiAcCool);
  ac.setTemp(26);
  ac.setVane(kMitsubishiAcVaneAuto);
}

void loop() {
  // Now send the IR signal.
#if SEND_MITSUBISHI_AC
  Serial.println("Sending IR command to A/C ...");
  ac.send();
#endif  // SEND_MITSUBISHI_AC
  printState();
  delay(5000);
}

crankyoldgit, 2019-04-11:
Try the following should work:

```
/* Copyright 2019 David Conran
*
* An IR LED circuit *MUST* be connected to the ESP8266 on a pin
* as specified by kIrLed below.
*
* TL;DR: The IR LED needs to be driven by a transistor for a good result.
*
* Suggested circuit:
*     https://github.com/markszabo/IRremoteESP8266/wiki#ir-sending
*
* Common mistakes & tips:
*   * Don't just connect the IR LED directly to the pin, it won't
*     have enough current to drive the IR LED effectively.
*   * Make sure you have the IR LED polarity correct.
*     See: https://learn.sparkfun.com/tutorials/polarity/diode-and-led-polarity
*   * Typical digital camera/phones can be used to see if the IR LED is flashed.
*     Replace the IR LED with a normal LED if you don't have a digital camera
*     when debugging.
*   * Avoid using the following pins unless you really know what you are doing:
*     * Pin 0/D3: Can interfere with the boot/program mode & support circuits.
*     * Pin 1/TX/TXD0: Any serial transmissions from the ESP8266 will interfere.
*     * Pin 3/RX/RXD0: Any serial transmissions to the ESP8266 will interfere.
*   * ESP-01 modules are tricky. We suggest you use a module with more GPIOs
*     for your first time. e.g. ESP-12 etc.
*/
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_MitsubishiHeavy.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRMitsubishiHeavy152Ac ac(kIrLed);  // Set the GPIO used for sending messages.

void printState() {
  // Display the settings.
  Serial.println("Mitsubishi Heavy A/C remote is in the following state:");
  Serial.printf("  %s\n", ac.toString().c_str());
  // Display the encoded IR sequence.
  unsigned char* ir_code = ac.getRaw();
  Serial.print("IR Code: 0x");
  for (uint8_t i = 0; i < kMitsubishiHeavy152StateLength; i++)
    Serial.printf("%02X", ir_code[i]);
  Serial.println();
}

void setup() {
  ac.begin();
  Serial.begin(115200);
  delay(200);

  // Set up what we want to send. See ir_MitsubishiHeavy.(cpp|h) for all the
  // options.
  Serial.println("Default state of the remote.");
  printState();
  Serial.println("Setting desired state for A/C.");
  ac.setPower(true);  // Turn it on.
  ac.setFan(kMitsubishiHeavy152FanMed);  // Medium Fan
  ac.setMode(kMitsubishiHeavyCool);  // Cool mode
  ac.setTemp(26);  // Celsius
  ac.setSwingVertical(kMitsubishiHeavy152SwingVAuto);  // Swing vertically
  ac.setSwingHorizontal(kMitsubishiHeavy152SwingHMiddle);  // Swing Horizontally
}

void loop() {
  // Now send the IR signal.
  Serial.println("Sending IR command to A/C ...");
  ac.send();
  printState();
  delay(5000);
}
```

crankyoldgit, 2019-04-11:
@tonyfitzs I've added that example code to the branch we've been using. So, if you re-download the branch, it should appear for you as an example. I've compiled it, so I know it work at least that much. ;-)

> If you use a different method other than Arduino to compile code to an Arduino then I'm happy to try it.

I personally use [PlatformIO](https://platformio.org/platformio-ide) & [Atom](https://atom.io/) (& [git](https://git-scm.com/)) on Linux. They meet my developer needs more than the Arduino IDE does, but it should work just as well with this library. I'm just not as familiar with day-to-day operation of the Arduino IDE as typically only use it when someone reports a problem specific to it.

**TL;DR**: My setup is quite advanced and more complicated because I need to juggle multiple versions of the library at once and have a lot of automation for testing and building etc. It's far from a typical arduino user's env.

tonyfitzs, 2019-04-14:
Hi David,

I have finally had the chance to test this again, now I'm getting a new error (sort of), and while I have tried to fix it. I'm not sure where the variable that is missing is supposed to be declared, it looks like this should be imported with the library, but for whatever reason, it doesn't seem to be working.

I have tried compiling the code you provided, and I am getting the following errors.

Arduino: 1.8.8 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, DOUT, 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from \TurnOnMitsubishiHeavyAC\TurnOnMitsubishiHeavyAC.ino:31:0:

\\libraries\IRremoteESP8266\src/ir_MitsubishiHeavy.h:187:24: error: 'kMitsubishiHeavy152StateLength' was not declared in this scope

   uint8_t remote_state[kMitsubishiHeavy152StateLength];

                        ^

\libraries\IRremoteESP8266\src/ir_MitsubishiHeavy.h:174:31: error: 'kMitsubishiHeavy152StateLength' was not declared in this scope

       const uint16_t length = kMitsubishiHeavy152StateLength);

                               ^

\libraries\IRremoteESP8266\src/ir_MitsubishiHeavy.h:251:24: error: 'kMitsubishiHeavy152StateLength' was not declared in this scope

   uint8_t remote_state[kMitsubishiHeavy152StateLength];

                        ^

\libraries\IRremoteESP8266\src/ir_MitsubishiHeavy.h:238:31: error: 'kMitsubishiHeavy88StateLength' was not declared in this scope

       const uint16_t length = kMitsubishiHeavy88StateLength);

                               ^

\TurnOnMitsubishiHeavyAC.ino: In function 'void printState()':

TurnOnMitsubishiHeavyAC:48:27: error: 'kMitsubishiHeavy152StateLength' was not declared in this scope

   for (uint8_t i = 0; i < kMitsubishiHeavy152StateLength; i++)

                           ^

\TurnOnMitsubishiHeavyAC.ino: In function 'void loop()':

TurnOnMitsubishiHeavyAC:79:6: error: 'class IRMitsubishiHeavy152Ac' has no member named 'send'

   ac.send();

      ^

exit status 1
'kMitsubishiHeavy152StateLength' was not declared in this scope

crankyoldgit, 2019-04-14:
Can you check if "IRremoteESP8266.h" exists and contains `kMitsubishiHeavy152StateLength`. It should, or again, for some reason you are still trying to compile against a _release_ version of the library, or it's still finding a release copy. Maybe you need to remove the Library via `Library Manager`?

tonyfitzs, 2019-04-14:
Got ya, I will check it out.

Sent from my SAMSUNG Galaxy S7 edge on the Telstra Mobile Network

tonyfitzs, 2019-04-14:
You rock!

It works. I will test it out with a bunch more modes and let you know the outcome but so far so good.

Thanks for you patience, I'm sure it's frustrating dealing with people who make stupid errors. I greatly appreciate your efforts.

crankyoldgit, 2019-04-14:
> I'm sure it's frustrating dealing with people who make stupid errors.

Haha. I'm not going to lie. ;-)

Glad it's working for you (so far).

closed 2019-04-15 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

commit 55f44249 (2019-04-15): First draft of support for Mitsubishi Heavy Industries A/Cs. (#665)

* First draft of support for Mitsubishi Heavy Industries A/Cs.
- Full support (send/receive/decode/class) for 152 bit (ZMS) protocol.
- Limited support (send/receive) for 88 bit (ZJS) protocol.
- Unit test coverage for all added code.
  * Can accept and fully decode supplied real examples.
- Updated example code to handle new A/C types.
* Add full support for Mitsubishi Heavy Industries 88bit/ZJS.
- Add IRMitsubishiHeavy88Ac class.
- Unit tests for the class.
- Minor tweaks to existing IRMitsubishiHeavy152Ac class.
- Update I [... cut]