crankyoldgit/IRremoteESP8266 #662: Add seperate status topic to IRMQTTServer

sheppy99, 2019-04-04:
Currently IRMQTTServer sends both its last command and also all status messages such as MQTT re connection and reboot messages on topic ir_server/sent. It would be useful to have these status messages on a separate topic to allow logic elsewhere to log failures and resend messages if necessary.

crankyoldgit, 2019-04-04:
PR #633 will address this and some other minor improvements.

crankyoldgit, 2019-04-10:
@sheppy99 Any testing results?

sheppy99, 2019-04-10:
Seems to be working, I've only had 9 dropouts and most of them were down to rebooting my OpenHAB box which also hosts Mosquitto. The 3 normal dropouts I did have seemed to create the right log entries on the correct topic. The Daikin2 sending on the default topic is proving reliable too. Unfortunately I'm going to be time constrained for the next 6 weeks so I can't test the extra channels for a while.

closed 2019-04-10 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

PR #663 (2019-04-10): Improve MQTT topic usage for feedback messages.
- Move log messages to their own MQTT topic. (`ir_server/log`)
- Set up a Last Will & Testament topic (`ir_server/status`)
- Change MQTT QoS to 1, which should hopefully handle messages lost during disconnects.

Tested log and LWT functionality on my personal device. Everything seems to work as normal.

Fixes #662
Ref #661
commit 3e3ef26c (2019-04-10): Improve MQTT topic usage for feedback messages. (#663)

- Move log messages to their own MQTT topic. (ir_server/log)
- Set up a Last Will & Testament topic (ir_server/status)
- Change MQTT QoS to 1, which should hopefully handle messages lost during disconnects.

Tested log and LWT functionality on my personal device. Everything seems to work as normal.

Fixes #662
Ref #661