crankyoldgit/IRremoteESP8266 #666: Does this library works for ESP32? What library can I use?

GiampaMan, 2019-04-09:
Hi! I'm a beginner.

I'm making a project using ESP32 and I need to send some IR commands. My problem is that I'm not founding any information about IR libraries compatibles with ESP32. Does this library works for ESP32? In case not, which one can I use? If is not created yet, do you have plans to do that library? that could be awesome and useful.

I will appreciate all your help.

Thanks!

crankyoldgit, 2019-04-09:
> Does this library works for ESP32?

No.

> In case not, which one can I use?

The library you want to use is: https://github.com/z3t0/Arduino-IRremote
It supports the ESP32 allegedly. @marcmerlin did the porting of it to the ESP32 (https://github.com/marcmerlin/Arduino-IRremote).

> If is not created yet, do you have plans to do that library?

I've been meaning to port this library over to ESP32 for 2 years. I've had zero need or incentive and thus made zero progress. It's unlikely to happen in the near future.

closed 2019-04-09 by crankyoldgit

GiampaMan, 2019-04-19:
Hi! Sorry about late response:

Thanks for all your help and answering my questions. 

Giampaolo.

dony71, 2019-05-20:
I try https://github.com/marcmerlin/Arduino-IRremote and integrate it in espeasy plugin
Compile it successfully, however while execution get this error below
Any idea why?
================================================================
IR Init
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x401bf2f4: bad00bad bad00bad bad00bad
Guru Meditation Error: Core  1 panic'ed (IlleGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopP        4)
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Unknown reason)
Guru Meditation Error: Core  1 panic'ed (Double exception)

marcmerlin, 2019-05-21:
you'll want to go to http://z3t0.github.io/Arduino-IRremote/ which is the canonical place for the ESP32 library. Try the latest version and file a bug there if there is a problem.

crankyoldgit, 2019-06-19:
FYI @GiampaMan & @marcmerlin As of [v2.6.2](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.6.2) this library now supports the ESP32. It isn't using `rmt` either, but it's a first step.

marcmerlin, 2019-06-20:
Cool, thanks @crankyoldgit .
Say, a crazy idea: would it make sense for you to merge back with the original code?

crankyoldgit, 2019-06-20:
@marcmerlin  You keep asking that. :)
No, It would not be a sane thing to do. The library is far to bigger than will fit on the typical microcontrollers that they support. Plus, it's far from a simple merge back.
And then there is the politics involved.
All I can do is try to offer basically the same or similar API.

But hey, feel free to try.

marcmerlin, 2019-06-23:
Haha, that's not me, it was the other marcmerlin guy :)
But more seriously, I haven't had looked at this lib ever since I needed it on ESP8266. I use the other one on ESP32 since I contributed that code there. What is the benefit of this lib on ESP32 since you mention it's bigger (and I assume does more)?

crankyoldgit, 2019-06-24:
This supports [65+ different protocols](https://github.com/markszabo/IRremoteESP8266/blob/master/SupportedProtocols.md). Almost all of them with decoders. 20+ of which are A/C protocols with full decode & construction of the settings and emulation of the A/C remotes behaviours, as well as a common framework to try to glue them all together in a "common" way.
It also has much large capture buffers (needed for A/C protocols) than `IRremote` supports. It also supports double buffering so decoding can happen while IR capture continues.
More configurable options/parameters ...

and (though nothing to do with prog/memory space) pretty much fully covered with extensive Unit Tests! :)

TL;DR: When fully optioned up, it takes up 100's of kilobytes of prog space. That doesn't leave much space on the older uControllers. Of course it can be paired down a LOT too.  ;-)

Why use this vs `IRremote` for your Burning Man suits? Probably no reason to for you. You don't need most of the functionality. But, for controlling and decoding every IR device in your house from a single device, probably yes.

tk97tk, 2021-03-16:
So, i have read that the library works with the esp32 nowadays. But i can not get it to work. With the esp8226 it is working perfectly fine.

crankyoldgit, 2021-03-16:
It does. If you have a problem, please file a new issue so we can get details of your configuration.