crankyoldgit/IRremoteESP8266 #667: No use of the F macro in the code base

jimmys01, 2019-04-09:
The code does use a lot of static strings that they do not use the F() macro
I am curius, is there a particular reason why not ?

crankyoldgit, 2019-04-09:
Short answer: 
In most general operations, the library doesn't use any strings. e.g. sending and receiving. Hence it doesn't need to use the `F()` macro. (see https://playground.arduino.cc/Learning/Memory/)

Longer answer:
`F()` and the small amount of SRAM it saves is most beneficial to _ATMega_ where there is radically less ram than the ESP line of chips **AND** to very large projects. Again, typically in most usage (i.e. outside of this library's example code) there is little to no strings used by this library. The SRAM space is save only when that string-based code is executed.

The library's code examples on the other had do use strings extensively. However, they are not situations that require great memory optimisation. They fit comfortably in the ESP8266's memory. There is currently no memory issues or pressure to optimise that code just yet.

An argument could be made to use the `F()` macros in the A/C class objects in their `.toString()` methods, but again, other than the example code e.g. `IRrecvDumpV2` nothing probably uses that method.

_IRMQTTServer_ currently isn't under any memory(SRAM) pressure, when it does, then yes. It could probably benefit from it. It's by far the biggest program/code included in the library that uses strings.

I could be completely wrong however, and if so, please point it out and I'll do something about it.
Till then, I'll not use `F()` as it makes the code easier to read/understand without it. If there is some project that is using a string-heavy part of this library and you want it optimised, just point to the area and I'll see what I can do.

Does that answer your question?

crankyoldgit, 2019-04-09:
@jimmys01 I've added a work-around to support the use of the `F()` macro in the Unit Tests that we use.
Thus, you can feel free to submit a PR with any F("String") changes you want/need/desire.

crankyoldgit, 2019-04-10:
Ref #670

closed 2019-04-12 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

commit f33450da (2019-04-12): Add F() Macros (#670)

Fixes #667