crankyoldgit/IRremoteESP8266 #668: Temp Up and Temp Down Commands is not working

mazeemsk, 2019-04-10:
Hi,
I am trying to send temp_up and temp_down commands but its is not working.....I got the code using IRrecvDumpV2 for temp up is

18:00:47.022 -> Encoding  : HAIER_AC
18:00:47.022 -> Code      : A51666180C802000E5 (72 bits)
18:00:47.022 -> Mesg Desc.: Command: 6 (Temp Up), Mode: 0 (AUTO), Temp: 17C, Fan: 0 (AUTO), Swing: 1 (Up), Sleep: Off, Health: Off, Current Time: 6:24, On Timer: Off, Off Timer: Off
18:00:47.055 -> Library   : v2.5.5
18:00:47.055 -> 
18:00:47.055 -> Raw Timing[149]:
18:00:47.055 ->    +  3076, -  2998,    +  3076, -  4244,    +   640, -  1648,    +   594, -   542, 
18:00:47.055 ->    +   592, -  1664,    +   592, -   674,    +   592, -   540,    +   592, -  1666, 
18:00:47.089 ->    +   592, -   540,    +   592, -  1794,    +   594, -   540,    +   620, -   510, 
18:00:47.089 ->    +   610, -   540,    +   594, -  1776,    +   610, -   544,    +   592, -  1646, 
18:00:47.089 ->    +   610, -  1648,    +   610, -   672,    +   590, -   526,    +   610, -  1662, 
18:00:47.089 ->    +   620, -  1652,    +   590, -   670,    +   596, -   540,    +   592, -  1662, 
18:00:47.123 ->    +   594, -  1646,    +   610, -   674,    +   590, -   540,    +   594, -   540, 
18:00:47.123 ->    +   594, -   542,    +   618, -  1764,    +   610, -  1646,    +   610, -   540, 
18:00:47.123 ->    +   594, -   538,    +   594, -   672,    +   592, -   538,    +   594, -   540, 
18:00:47.123 ->    +   594, -   546,    +   588, -   670,    +   620, -  1632,    +   610, -  1662, 
18:00:47.123 ->    +   594, -   540,    +   592, -   670,    +   596, -  1662,    +   594, -   540, 
18:00:47.157 ->    +   592, -   540,    +   596, -   668,    +   594, -   538,    +   622, -   524, 
18:00:47.157 ->    +   594, -   538,    +   596, -   668,    +   596, -   540,    +   592, -   542, 
18:00:47.157 ->    +   590, -  1662,    +   594, -   672,    +   594, -   538,    +   596, -   538, 
18:00:47.157 ->    +   590, -   554,    +   596, -   668,    +   596, -   536,    +   598, -   538, 
18:00:47.191 ->    +   596, -   538,    +   596, -   670,    +   594, -   538,    +   594, -   540, 
18:00:47.191 ->    +   594, -   538,    +   620, -   686,    +   592, -  1646,    +   610, -  1646, 
18:00:47.191 ->    +   612, -  1646,    +   610, -   698,    +   590, -   540,    +   594, -  1662, 
18:00:47.191 ->    +   596, -   540,    +   594, -  1660,    +   622
18:00:47.191 -> 
18:00:47.225 -> uint16_t rawData[149] = {3076, 2998,  3076, 4244,  640, 1648,  594, 542,  592, 1664,  592, 674,  592, 540,  592, 1666,  592, 540,  592, 1794,  594, 540,  620, 510,  610, 540,  594, 1776,  610, 544,  592, 1646,  610, 1648,  610, 672,  590, 526,  610, 1662,  620, 1652,  590, 670,  596, 540,  592, 1662,  594, 1646,  610, 674,  590, 540,  594, 540,  594, 542,  618, 1764,  610, 1646,  610, 540,  594, 538,  594, 672,  592, 538,  594, 540,  594, 546,  588, 670,  620, 1632,  610, 1662,  594, 540,  592, 670,  596, 1662,  594, 540,  592, 540,  596, 668,  594, 538,  622, 524,  594, 538,  596, 668,  596, 540,  592, 542,  590, 1662,  594, 672,  594, 538,  596, 538,  590, 554,  596, 668,  596, 536,  598, 538,  596, 538,  596, 670,  594, 538,  594, 540,  594, 538,  620, 686,  592, 1646,  610, 1646,  612, 1646,  610, 698,  590, 540,  594, 1662,  596, 540,  594, 1660,  622};  // HAIER_AC
18:00:47.293 -> uint8_t state[9] = {0xA5, 0x16, 0x66, 0x18, 0x0C, 0x80, 0x20, 0x00, 0xE5};
 
and for temp down is

18:00:04.059 -> Encoding  : HAIER_AC
18:00:04.059 -> Code      : A50766180C802000D6 (72 bits)
18:00:04.059 -> Mesg Desc.: Command: 7 (Temp Down), Mode: 0 (AUTO), Temp: 16C, Fan: 0 (AUTO), Swing: 1 (Up), Sleep: Off, Health: Off, Current Time: 6:24, On Timer: Off, Off Timer: Off
18:00:04.092 -> Library   : v2.5.5
18:00:04.092 -> 
18:00:04.092 -> Raw Timing[149]:
18:00:04.092 ->    +  3078, -  3000,    +  3074, -  4264,    +   592, -  1676,    +   570, -   564, 
18:00:04.092 ->    +   596, -  1662,    +   568, -   698,    +   594, -   536,    +   596, -  1662, 
18:00:04.126 ->    +   568, -   566,    +   566, -  1826,    +   562, -   566,    +   594, -   590, 
18:00:04.126 ->    +   528, -   568,    +   566, -   696,    +   570, -   564,    +   568, -  1690, 
18:00:04.126 ->    +   566, -  1688,    +   568, -  1892,    +   494, -   570,    +   564, -  1692, 
18:00:04.126 ->    +   590, -  1678,    +   566, -   698,    +   568, -   568,    +   566, -  1688, 
18:00:04.159 ->    +   568, -  1688,    +   568, -   700,    +   566, -   568,    +   566, -   564, 
18:00:04.159 ->    +   568, -   564,    +   596, -  1810,    +   566, -  1686,    +   570, -   564, 
18:00:04.159 ->    +   570, -   566,    +   566, -   744,    +   520, -   566,    +   566, -   566, 
18:00:04.159 ->    +   568, -   568,    +   562, -   698,    +   594, -  1676,    +   566, -  1686, 
18:00:04.159 ->    +   570, -   566,    +   568, -   696,    +   566, -  1688,    +   570, -   566, 
18:00:04.194 ->    +   568, -   568,    +   564, -   700,    +   564, -   564,    +   596, -   548, 
18:00:04.194 ->    +   572, -   564,    +   570, -   696,    +   568, -   566,    +   568, -   566, 
18:00:04.194 ->    +   570, -  1684,    +   570, -   696,    +   568, -   566,    +   566, -   566, 
18:00:04.194 ->    +   594, -   552,    +   566, -   696,    +   568, -   566,    +   566, -   566, 
18:00:04.227 ->    +   568, -   566,    +   564, -   700,    +   566, -   564,    +   568, -   568, 
18:00:04.227 ->    +   564, -   570,    +   592, -   706,    +   570, -  1688,    +   568, -  1686, 
18:00:04.227 ->    +   568, -   568,    +   566, -  1844,    +   568, -   566,    +   568, -  1692, 
18:00:04.227 ->    +   564, -  1692,    +   564, -   566,    +   596
18:00:04.227 -> 
18:00:04.262 -> uint16_t rawData[149] = {3078, 3000,  3074, 4264,  592, 1676,  570, 564,  596, 1662,  568, 698,  594, 536,  596, 1662,  568, 566,  566, 1826,  562, 566,  594, 590,  528, 568,  566, 696,  570, 564,  568, 1690,  566, 1688,  568, 1892,  494, 570,  564, 1692,  590, 1678,  566, 698,  568, 568,  566, 1688,  568, 1688,  568, 700,  566, 568,  566, 564,  568, 564,  596, 1810,  566, 1686,  570, 564,  570, 566,  566, 744,  520, 566,  566, 566,  568, 568,  562, 698,  594, 1676,  566, 1686,  570, 566,  568, 696,  566, 1688,  570, 566,  568, 568,  564, 700,  564, 564,  596, 548,  572, 564,  570, 696,  568, 566,  568, 566,  570, 1684,  570, 696,  568, 566,  566, 566,  594, 552,  566, 696,  568, 566,  566, 566,  568, 566,  564, 700,  566, 564,  568, 568,  564, 570,  592, 706,  570, 1688,  568, 1686,  568, 568,  566, 1844,  568, 566,  568, 1692,  564, 1692,  564, 566,  596};  // HAIER_AC
18:00:04.330 -> uint8_t state[9] = {0xA5, 0x07, 0x66, 0x18, 0x0C, 0x80, 0x20, 0x00, 0xD6};

and I am trying to send is as

#include <IRsend.h>
#include <ir_Haier.h>
const uint16_t kIrLed = 4;
IRsend ac(kIrLed);
uint8_t temp_down[9] = {0xA5, 0x16, 0x66, 0x18, 0x0C, 0x80, 0x20, 0x00, 0xE5};
uint8_t temp_up[9] = {0xA5, 0x07, 0x66, 0x18, 0x0C, 0x80, 0x20, 0x00, 0xD6};
ac.sendHaierAC(temp_dn);
ac.sendHaierAC(temp_dn);

its just produce beeps but command not executed....but by same procedure i can send ON/OFF commands and it is working where am I missing.

crankyoldgit, 2019-04-10:
I'm not 100% sure what you are describing so, apologies if I get it wrong. i.e. How do you know the commands are not executing/producing the desired change on the A/C?

Looking at those codes etc, they look like they are a result of you pressing the "temp up" and "temp down" buttons on the remote.

i don't own the make/model of the A/C so I can't say for certain, but I think the code is reporting what button was last pressed AND then setting the rest of the state of the A/C to what is in the message. etc. e.g. `temp_up` is setting it to 17C auto etc, and `temp_down` to 16C auto etc.
They are _not_ codes to increase the temp by one degree and down by one degree.
Also, I think the A/C unit needs to be ON before those command sequences will work.

Another way to control the devices is via the class method. See https://github.com/markszabo/IRremoteESP8266/blob/master/test/ir_Haier_test.cpp#L356 for some examples. You can adapt some of the other existing A/C examples to suit. Just be aware you need to use the `.send()` method on the class to send the constructed state, and that Haier appears odd in that doesn't include the on/off setting as part of the normal state that is sent. You need to send "on" commands like:
```
IRHaierAC ac(kIrPin);
ac.on();
ac.send();
delay(sometimelater);
ac.setMode(blah);
ac.setTemp(22);
ac.send();
```

mazeemsk, 2019-04-11:
What if I have to increase the temp by one degree and down by one degree ?? Can I use 

IRHaierAC ac(kIrPin);

for increasing

ac.setcommand(kHaierAcCmdTempUp);
ac.send();

for decreasing
ac.setcommand(kHaierAcCmdTempDown);
ac.send();

is it the right way to increase and decrease temperature by 1 degree by using class method??.

crankyoldgit, 2019-04-11:
No.

I believe the correct way would be:
```
IRHaierAC ac(kIrPin);

ac.on();
ac.send();  // Turn the device on. Haier is different in that it needs to send an independent power on mesg.

delay(sometime later etc);

ac.setMode(kHaierAcCool);
ac.setTemp(25);  // Set the temp to 25C.
ac.send();  // Set the unit into Cool mode and 25C.

delay(more time has passed);

// I want to increase the temp by one degree (i.e. 26) when I know what I want the temp to be.
ac.setTemp(26);
ac.send();  // Specifically send a new temp message.

delay(some time passes);

// I want to increase the temp by one degree when I don't care what the temp is.
// i.e. The `ac` class object remembers that I previously had the temp set to 26C.
// Note: If in the mean time you set the A/C to say "19C" via the actual remote, it will NOT change
//           it to 20C. The remote (or the class) has it's own concept of what the previous temp is/was.
//           e.g. `ac.getTemp()` gets what the previous requested temp was from the "remote", not the A/C.
ac.setTemp(ac.getTemp() + 1);  // Will set the temp to 26 (from `ac.getTemp()`) + 1 which is 27.
ac.send();  // Send the new temp.

delay(omg_its_gotten_too_warm);

// Lower what ever the last set temp was by one degree.
ac.setTemp(ac.getTemp() -1 1);  // Will set the temp to 27 (from `ac.getTemp()`) - 1 which is 26.
ac.send();  // Send the new temp.
// etc etc.
```

Now, I'm not sure if that works perfectly etc with the A/C Unit, as I don't have one to test against.
What I do know is, there is NO magic `state[]` or raw message that you can repeat to just "adjust the temp by one degree" (up or down) that I am aware of. i.e. It's not like a typical IR TV remote that has a simple "vol up" or "vol down" message.

Almost all A/C remotes send the entire configuration in each message. i.e. The entire state of the remote control. Not "just change one little aspect please".

mazeemsk, 2019-04-14:
Hello,
I tried this method and it gives me the error.
'class IRHaierAC' has no member named 'on'
'class IRHaierAC' has no member named 'off'

how can resolve this???

crankyoldgit, 2019-04-14:
Haier A/C is a weird one it seems. It doesn't have `.on()`/ `.off()`/`.setPower(bool)` methods.

Try the following:
```
IRHaierAC ac(kIrPin);

ac.setCommand(kHaierAcCmdOn);
ac.send();  // Send the power on command.
delay(some_reasonable_time_passes);
ac.setCommand(kHaierAcCmdOff);
ac.send();  // Send the power off command.
```

Let us know how that goes please.

mazeemsk, 2019-04-14:
Yes it is working by ac.setCommand(kHaierAcCmdOn); 
but now temp up and down is not working as I am trying your method.

ac.setCommand(kHaierAcCmdOn);
          ac.send();  // Send the power on command.
          delay(3000);
          ac.setTemp(25); 
          ac.send();  // Send the new temp.
         delay(sometime);
           ac.setTemp(ac.getTemp() + 1); 
           ac.send();  // Send the new temp.
          delay(sometime);
           ac.setTemp(ac.getTemp() - 1);  
            ac.send();  // Send the new temp

crankyoldgit, 2019-04-15:
@mazeemsk I've added a test case to try to emulate what you are doing. See the commit https://github.com/markszabo/IRremoteESP8266/commit/9d189fd9d29b355c1c91bd5aaa8967c902e539a7

As far as I can tell, the code is doing what you're trying to do.
i.e. Turn on the unit, _send_, then set the temp to _25C_, _send_, then _raise_ the temp by 1 _(26C)_, _send_, then _lower_ the temp by 1 _(25C again)_, and _send_.

The net result is the unit will stay at _25C_.

Each send, the command appears to be valid and do the correct things. See the `char expected_blah[]` text in the test.

If you believe it isn't, in order for me to fix it I'll need you to please capture (via `IRrecvDumpV2`) from the real remote the same situations you are trying to do from the code. i.e. **ALL** the settings on the remote need to be the same as the text. 
e.g.
Have the remote **off**, then press **on** so it produces:
`Command: 1 (On), Mode: 0 (AUTO), Temp: 25C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:00, On Timer: Off, Off Timer: Off`

Then press **Temp Up**:
and it then should produce:
`Command: 6 (Temp Up), Mode: 0 (AUTO), Temp: 26C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:00, On Timer: Off, Off Timer: Off`

Please provide the `state[]` and `rawData[]` both of those situations, and label them accordingly.

Also, you might as well include a link (or paste) the entire code you are using so we can see if there might be some other problem.

mazeemsk, 2019-04-15:
Okay I tried to send the commands via MQTT but the results are same the complete code I am using is

```
/***************************************************
 ****************************************************/
#include "Adafruit_MQTT.h"
#include "Adafruit_MQTT_Client.h"
#include <ESP8266WiFi.h>
#include <ESP_EEPROM.h>
#include <IRremoteESP8266.h>
#include <ir_Haier.h>
/************************* Pin Definition *********************************/       
IRHaierAC ac(4);
IPAddress ip(192, 168, 1, 90);            // IP address of the server
IPAddress gateway(192,168,1,1);           // gateway of your network
IPAddress subnet(255,255,255,0);          // subnet mask of your network
/************************* WiFi Access Point *********************************/
#define WLAN_SSID       "Home"
#define WLAN_PASS       "11223344"
/************************* Adafruit.io Setup *********************************/
#define AIO_SERVER      "192.168.1.200" //IP address of RPi
#define AIO_SERVERPORT  1883                   // use 8883 for SSL
#define AIO_USERNAME    ""
#define AIO_KEY         ""
/************ Global State (you don't need to change this!) ******************/
// Create an ESP8266 WiFiClient class to connect to the MQTT server.
WiFiClient esp4client;
// or... use WiFiFlientSecure for SSL
//WiFiClientSecure client;
const char MQTT_CLIENTID[] PROGMEM  = "testir";
// Setup the MQTT client class by passing in the WiFi client and MQTT server and login details.
Adafruit_MQTT_Client mqtt(&esp4client, AIO_SERVER, AIO_SERVERPORT,MQTT_CLIENTID, AIO_USERNAME, AIO_KEY);
/****************************** Feeds ***************************************/
// Notice MQTT paths for AIO follow the form: <username>/feeds/<feedname>
// Setup a feed called 'onoff' for subscribing to changes.
Adafruit_MQTT_Subscribe acremoteon = Adafruit_MQTT_Subscribe(&mqtt, AIO_USERNAME "/feeds/acon");
Adafruit_MQTT_Subscribe acremoteoff = Adafruit_MQTT_Subscribe(&mqtt, AIO_USERNAME "/feeds/acoff");
Adafruit_MQTT_Subscribe acremotetempup = Adafruit_MQTT_Subscribe(&mqtt, AIO_USERNAME "/feeds/actempup");
Adafruit_MQTT_Subscribe acremotetempdn = Adafruit_MQTT_Subscribe(&mqtt, AIO_USERNAME "/feeds/actempdown");
/*************************** Sketch Code ************************************/
void setup() {
  Serial.begin(115200);
  ac.begin();
  delay(10);
  Serial.println(); Serial.println();
  // Connect to WiFi access point.
  Serial.println(); Serial.println();
  Serial.print("Connecting to ");
  Serial.println(WLAN_SSID);
  WiFi.config(ip, gateway, subnet);      // forces to use the fix IP 
  WiFi.begin(WLAN_SSID, WLAN_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.println("WiFi connected");
  Serial.println("IP address: "); Serial.println(WiFi.localIP());
 
  // Setup MQTT subscription for onoff feed.
  mqtt.subscribe(&acremoteon);
  mqtt.subscribe(&acremoteoff);
  mqtt.subscribe(&acremotetempup);
  mqtt.subscribe(&acremotetempdn);
}

// Bug workaround for Arduino 1.6.6, it seems to need a function declaration
// for some reason (only affects ESP8266, likely an arduino-builder bug).
void MQTT_connect();
//Servo myservo;
[... 66 lines cut ...]
  uint8_t retries = 3;
  while ((ret = mqtt.connect()) != 0) { // connect will return 0 for connected
    Serial.println(mqtt.connectErrorString(ret));
    Serial.println("Retrying MQTT connection in 5 seconds...");
    mqtt.disconnect();
    delay(5000);  // wait 5 seconds
    retries--;
    if (retries == 0) {
      // basically die and wait for WDT to reset me
      while (1);
    }
  }
  Serial.println("MQTT Connected!");
}

```

The Results are:

1. When I press Power ON it turned it ON and after 3 Seconds set the temp to 25.
2. When I press Power OFF it turned it OFF.
3. When I press Temp UP AC beeps but screen does not shows 26(while controlling by remote it beeps and shows 26 for sometime when pressed temp up and when pressed temp up again it shows 27 for sometime  then shows room temperature again).
4. Same as number 3 happens for Temp Down.

sorry for my bad English.

crankyoldgit, 2019-04-15:
> If you believe it isn't, in order for me to fix it I'll need you to please capture (via `IRrecvDumpV2`) from the real remote the same situations you are trying to do from the code. i.e. **ALL** the [...]

Can you please do the above, I'm guessing something we got something wrong on how to emulate it.

mazeemsk, 2019-04-16:
Here it is...
Power ON:
```
11:21:55.972 -> Timestamp : 000073.866
11:21:55.972 -> Encoding  : HAIER_AC
11:21:55.972 -> Code      : A59120080CC020004A (72 bits)
11:21:55.972 -> Mesg Desc.: Command: 1 (On), Mode: 0 (AUTO), Temp: 25C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:08, On Timer: Off, Off Timer: Off
11:21:56.005 -> Library   : v2.5.6
11:21:56.005 -> 
11:21:56.005 -> Raw Timing[149]:
11:21:56.005 ->    +  3052, -  3024,    +  3050, -  4284,    +   564, -  1708,    +   564, -   568, 
11:21:56.005 ->    +   568, -  1688,    +   564, -   700,    +   568, -   564,    +   568, -  1688, 
11:21:56.005 ->    +   566, -   570,    +   560, -  1822,    +   570, -  1690,    +   592, -   552, 
11:21:56.039 ->    +   570, -   562,    +   568, -  1826,    +   558, -   570,    +   564, -   570, 
11:21:56.039 ->    +   564, -   566,    +   568, -  1822,    +   562, -   572,    +   564, -   564, 
11:21:56.039 ->    +   594, -  1674,    +   568, -   696,    +   568, -   564,    +   570, -   566, 
11:21:56.039 ->    +   566, -   564,    +   570, -   694,    +   572, -   562,    +   568, -   616, 
11:21:56.039 ->    +   512, -   570,    +   592, -   686,    +   566, -  1692,    +   560, -   568, 
11:21:56.073 ->    +   566, -   568,    +   566, -   694,    +   570, -   568,    +   562, -   566, 
11:21:56.073 ->    +   570, -   566,    +   566, -   698,    +   594, -  1676,    +   568, -  1688, 
11:21:56.073 ->    +   570, -   568,    +   560, -   750,    +   514, -  1690,    +   570, -  1694, 
11:21:56.073 ->    +   562, -   564,    +   566, -   696,    +   570, -   616,    +   540, -   606, 
11:21:56.107 ->    +   514, -   572,    +   562, -   702,    +   566, -   568,    +   566, -   566, 
11:21:56.107 ->    +   566, -  1684,    +   570, -   696,    +   566, -   568,    +   566, -   564, 
11:21:56.107 ->    +   596, -   552,    +   570, -   698,    +   560, -   618,    +   514, -   572, 
11:21:56.107 ->    +   564, -   566,    +   568, -   704,    +   560, -   570,    +   564, -   566, 
11:21:56.107 ->    +   566, -   566,    +   594, -   708,    +   564, -   568,    +   568, -  1692, 
11:21:56.141 ->    +   560, -   574,    +   558, -   726,    +   566, -  1738,    +   514, -   572, 
11:21:56.141 ->    +   562, -  1690,    +   564, -   574,    +   588
11:21:56.141 -> 
11:21:56.141 -> uint16_t rawData[149] = {3052, 3024,  3050, 4284,  564, 1708,  564, 568,  568, 1688,  564, 700,  568, 564,  568, 1688,  566, 570,  560, 1822,  570, 1690,  592, 552,  570, 562,  568, 1826,  558, 570,  564, 570,  564, 566,  568, 1822,  562, 572,  564, 564,  594, 1674,  568, 696,  568, 564,  570, 566,  566, 564,  570, 694,  572, 562,  568, 616,  512, 570,  592, 686,  566, 1692,  560, 568,  566, 568,  566, 694,  570, 568,  562, 566,  570, 566,  566, 698,  594, 1676,  568, 1688,  570, 568,  560, 750,  514, 1690,  570, 1694,  562, 564,  566, 696,  570, 616,  540, 606,  514, 572,  562, 702,  566, 568,  566, 566,  566, 1684,  570, 696,  566, 568,  566, 564,  596, 552,  570, 698,  560, 618,  514, 572,  564, 566,  568, 704,  560, 570,  564, 566,  566, 566,  594, 708,  564, 568,  568, 1692,  560, 574,  558, 726,  566, 1738,  514, 572,  562, 1690,  564, 574,  588};  // HAIER_AC
11:21:56.209 -> uint8_t state[9] = {0xA5, 0x91, 0x20, 0x08, 0x0C, 0xC0, 0x20, 0x00, 0x4A};
```

and Temp UP

```
11:22:54.692 -> Timestamp : 000132.599
11:22:54.692 -> Encoding  : HAIER_AC
11:22:54.692 -> Code      : A5A620090CC0200060 (72 bits)
11:22:54.692 -> Mesg Desc.: Command: 6 (Temp Up), Mode: 0 (AUTO), Temp: 26C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:09, On Timer: Off, Off Timer: Off
11:22:54.726 -> Library   : v2.5.6
11:22:54.726 -> 
11:22:54.726 -> Raw Timing[149]:
11:22:54.726 ->    +  3048, -  3026,    +  3046, -  4284,    +   596, -  1672,    +   570, -   572, 
11:22:54.760 ->    +   558, -  1690,    +   568, -   696,    +   568, -   568,    +   560, -  1696, 
11:22:54.760 ->    +   566, -   568,    +   564, -  1826,    +   558, -  1694,    +   592, -   550, 
11:22:54.760 ->    +   570, -  1690,    +   564, -   700,    +   566, -   566,    +   570, -  1688, 
11:22:54.760 ->    +   566, -  1690,    +   566, -   702,    +   564, -   566,    +   566, -   568, 
11:22:54.760 ->    +   592, -  1674,    +   570, -   698,    +   566, -   564,    +   568, -   568, 
11:22:54.794 ->    +   564, -   568,    +   566, -   696,    +   568, -   566,    +   568, -   572, 
11:22:54.794 ->    +   556, -   570,    +   592, -   690,    +   560, -  1694,    +   560, -   570, 
11:22:54.794 ->    +   566, -   566,    +   566, -  1822,    +   564, -   566,    +   568, -   566, 
11:22:54.794 ->    +   566, -   566,    +   566, -   696,    +   596, -  1678,    +   564, -  1688, 
11:22:54.794 ->    +   570, -   566,    +   564, -   702,    +   564, -  1690,    +   564, -  1692, 
11:22:54.828 ->    +   566, -   570,    +   562, -   704,    +   560, -   570,    +   592, -   550, 
11:22:54.828 ->    +   568, -   568,    +   560, -   704,    +   562, -   566,    +   568, -   566, 
11:22:54.828 ->    +   566, -  1690,    +   566, -   702,    +   562, -   566,    +   568, -   564, 
11:22:54.828 ->    +   596, -   554,    +   564, -   698,    +   566, -   564,    +   568, -   564, 
11:22:54.862 ->    +   568, -   564,    +   568, -   702,    +   560, -   566,    +   568, -   566, 
11:22:54.862 ->    +   568, -   568,    +   592, -   714,    +   562, -   572,    +   560, -  1694, 
[... cut ...] 1688,  570, 566,  564, 702,  564, 1690,  564, 1692,  566, 570,  562, 704,  560, 570,  592, 550,  568, 568,  560, 704,  562, 566,  568, 566,  566, 1690,  566, 702,  562, 566,  568, 564,  596, 554,  564, 698,  566, 564,  568, 564,  568, 564,  568, 702,  560, 566,  568, 566,  568, 568,  592, 714,  562, 572,  560, 1694,  562, 1692,  562, 724,  566, 564,  568, 568,  564, 568,  564, 570,  592};  // HAIER_AC
11:22:54.964 -> uint8_t state[9] = {0xA5, 0xA6, 0x20, 0x09, 0x0C, 0xC0, 0x20, 0x00, 0x60};

```

mazeemsk, 2019-04-16:
Temp UP at Time 00:00

```
11:32:25.478 -> Timestamp : 000703.365
11:32:25.478 -> Encoding  : HAIER_AC
11:32:25.478 -> Code      : A5A620000CC0200057 (72 bits)
11:32:25.478 -> Mesg Desc.: Command: 6 (Temp Up), Mode: 0 (AUTO), Temp: 26C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:00, On Timer: Off, Off Timer: Off
11:32:25.478 -> Library   : v2.5.6
11:32:25.478 -> 
11:32:25.512 -> Raw Timing[149]:
11:32:25.512 ->    +  3046, -  3032,    +  3070, -  4266,    +   588, -  1676,    +   568, -   570, 
11:32:25.512 ->    +   564, -  1736,    +   520, -   694,    +   568, -   566,    +   566, -  1688, 
11:32:25.512 ->    +   568, -   568,    +   564, -  1820,    +   570, -  1686,    +   596, -   554, 
11:32:25.512 ->    +   566, -  1688,    +   566, -   698,    +   566, -   564,    +   570, -  1686, 
11:32:25.546 ->    +   570, -  1690,    +   566, -   698,    +   568, -   564,    +   570, -   566, 
11:32:25.546 ->    +   592, -  1674,    +   568, -   694,    +   570, -   564,    +   568, -   564, 
11:32:25.546 ->    +   568, -   564,    +   570, -   696,    +   568, -   564,    +   568, -   568, 
11:32:25.546 ->    +   564, -   564,    +   596, -   688,    +   562, -   564,    +   570, -   564, 
11:32:25.546 ->    +   566, -   566,    +   568, -   698,    +   566, -   568,    +   564, -   566, 
11:32:25.580 ->    +   568, -   566,    +   566, -   700,    +   590, -  1676,    +   568, -  1688, 
11:32:25.580 ->    +   568, -   564,    +   570, -   696,    +   568, -  1688,    +   568, -  1692, 
11:32:25.580 ->    +   564, -   562,    +   570, -   696,    +   568, -   570,    +   590, -   552, 
11:32:25.580 ->    +   570, -   568,    +   566, -   700,    +   562, -   614,    +   520, -   564, 
11:32:25.580 ->    +   568, -  1686,    +   570, -   694,    +   572, -   568,    +   562, -   564, 
11:32:25.614 ->    +   594, -   554,    +   568, -   694,    +   570, -   564,    +   568, -   564, 
11:32:25.614 ->    +   568, -   566,    +   568, -   698,    +   566, -   566,    +   568, -   568, 
11:32:25.614 ->    +   564, -   566,    +   592, -   706,    +   570, -   564,    +   570, -  1686, 
11:32:25.614 ->    +   570, -   564,    +   568, -  1842,    +   570, -   564,    +   570, -  1690, 
11:32:25.648 ->    +   568, -  1688,    +   566, -  1688,    +   594
11:32:25.648 -> 
11:32:25.648 -> uint16_t rawData[149] = {3046, 3032,  3070, 4266,  588, 1676,  568, 570,  564, 1736,  520, 694,  568, 566,  566, 1688,  568, 568,  564, 1820,  570, 1686,  596, 554,  566, 1688,  566, 698,  566, 564,  570, 1686,  570, 1690,  566, 698,  568, 564,  570, 566,  592, 1674,  568, 694,  570, 564,  568, 564,  568, 564,  570, 696,  568, 564,  568, 568,  564, 564,  596, 688,  562, 564,  570, 564,  566, 566,  568, 698,  566, 568,  564, 566,  568, 566,  566, 700,  590, 1676,  568, 1688,  568, 564,  570, 696,  568, 1688,  568, 1692,  564, 562,  570, 696,  568, 570,  590, 552,  570, 568,  566, 700,  562, 614,  520, 564,  568, 1686,  570, 694,  572, 568,  562, 564,  594, 554,  568, 694,  570, 564,  568, 564,  568, 566,  568, 698,  566, 566,  568, 568,  564, 566,  592, 706,  570, 564,  570, 1686,  570, 564,  568, 1842,  570, 564,  570, 1690,  568, 1688,  566, 1688,  594};  // HAIER_AC
11:32:25.716 -> uint8_t state[9] = {0xA5, 0xA6, 0x20, 0x00, 0x0C, 0xC0, 0x20, 0x00, 0x57};

````

mazeemsk, 2019-04-16:
Here it is...
Power ON:
```
11:21:55.972 -> Timestamp : 000073.866
11:21:55.972 -> Encoding  : HAIER_AC
11:21:55.972 -> Code      : A59120080CC020004A (72 bits)
11:21:55.972 -> Mesg Desc.: Command: 1 (On), Mode: 0 (AUTO), Temp: 25C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:08, On Timer: Off, Off Timer: Off
11:21:56.005 -> Library   : v2.5.6
11:21:56.005 -> 
11:21:56.005 -> Raw Timing[149]:
11:21:56.005 ->    +  3052, -  3024,    +  3050, -  4284,    +   564, -  1708,    +   564, -   568, 
11:21:56.005 ->    +   568, -  1688,    +   564, -   700,    +   568, -   564,    +   568, -  1688, 
11:21:56.005 ->    +   566, -   570,    +   560, -  1822,    +   570, -  1690,    +   592, -   552, 
11:21:56.039 ->    +   570, -   562,    +   568, -  1826,    +   558, -   570,    +   564, -   570, 
11:21:56.039 ->    +   564, -   566,    +   568, -  1822,    +   562, -   572,    +   564, -   564, 
11:21:56.039 ->    +   594, -  1674,    +   568, -   696,    +   568, -   564,    +   570, -   566, 
11:21:56.039 ->    +   566, -   564,    +   570, -   694,    +   572, -   562,    +   568, -   616, 
11:21:56.039 ->    +   512, -   570,    +   592, -   686,    +   566, -  1692,    +   560, -   568, 
11:21:56.073 ->    +   566, -   568,    +   566, -   694,    +   570, -   568,    +   562, -   566, 
11:21:56.073 ->    +   570, -   566,    +   566, -   698,    +   594, -  1676,    +   568, -  1688, 
11:21:56.073 ->    +   570, -   568,    +   560, -   750,    +   514, -  1690,    +   570, -  1694, 
11:21:56.073 ->    +   562, -   564,    +   566, -   696,    +   570, -   616,    +   540, -   606, 
11:21:56.107 ->    +   514, -   572,    +   562, -   702,    +   566, -   568,    +   566, -   566, 
11:21:56.107 ->    +   566, -  1684,    +   570, -   696,    +   566, -   568,    +   566, -   564, 
11:21:56.107 ->    +   596, -   552,    +   570, -   698,    +   560, -   618,    +   514, -   572, 
11:21:56.107 ->    +   564, -   566,    +   568, -   704,    +   560, -   570,    +   564, -   566, 
11:21:56.107 ->    +   566, -   566,    +   594, -   708,    +   564, -   568,    +   568, -  1692, 
11:21:56.141 ->    +   560, -   574,    +   558, -   726,    +   566, -  1738,    +   514, -   572, 
11:21:56.141 ->    +   562, -  1690,    +   564, -   574,    +   588
11:21:56.141 -> 
11:21:56.141 -> uint16_t rawData[149] = {3052, 3024,  3050, 4284,  564, 1708,  564, 568,  568, 1688,  564, 700,  568, 564,  568, 1688,  566, 570,  560, 1822,  570, 1690,  592, 552,  570, 562,  568, 1826,  558, 570,  564, 570,  564, 566,  568, 1822,  562, 572,  564, 564,  594, 1674,  568, 696,  568, 564,  570, 566,  566, 564,  570, 694,  572, 562,  568, 616,  512, 570,  592, 686,  566, 1692,  560, 568,  566, 568,  566, 694,  570, 568,  562, 566,  570, 566,  566, 698,  594, 1676,  568, 1688,  570, 568,  560, 750,  514, 1690,  570, 1694,  562, 564,  566, 696,  570, 616,  540, 606,  514, 572,  562, 702,  566, 568,  566, 566,  566, 1684,  570, 696,  566, 568,  566, 564,  596, 552,  570, 698,  560, 618,  514, 572,  564, 566,  568, 704,  560, 570,  564, 566,  566, 566,  594, 708,  564, 568,  568, 1692,  560, 574,  558, 726,  566, 1738,  514, 572,  562, 1690,  564, 574,  588};  // HAIER_AC
11:21:56.209 -> uint8_t state[9] = {0xA5, 0x91, 0x20, 0x08, 0x0C, 0xC0, 0x20, 0x00, 0x4A};
```
and Temp UP

```
11:32:25.478 -> Timestamp : 000703.365
11:32:25.478 -> Encoding  : HAIER_AC
11:32:25.478 -> Code      : A5A620000CC0200057 (72 bits)
11:32:25.478 -> Mesg Desc.: Command: 6 (Temp Up), Mode: 0 (AUTO), Temp: 26C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:00, On Timer: Off, Off Timer: Off
11:32:25.478 -> Library   : v2.5.6
11:32:25.478 -> 
11:32:25.512 -> Raw Timing[149]:
11:32:25.512 ->    +  3046, -  3032,    +  3070, -  4266,    +   588, -  1676,    +   568, -   570, 
11:32:25.512 ->    +   564, -  1736,    +   520, -   694,    +   568, -   566,    +   566, -  1688, 
11:32:25.512 ->    +   568, -   568,    +   564, -  1820,    +   570, -  1686,    +   596, -   554, 
11:32:25.512 ->    +   566, -  1688,    +   566, -   698,    +   566, -   564,    +   570, -  1686, 
11:32:25.546 ->    +   570, -  1690,    +   566, -   698,    +   568, -   564,    +   570, -   566, 
11:32:25.546 ->    +   592, -  1674,    +   568, -   694,    +   570, -   564,    +   568, -   564, 
11:32:25.546 ->    +   568, -   564,    +   570, -   696,    +   568, -   564,    +   568, -   568, 
11:32:25.546 ->    +   564, -   564,    +   596, -   688,    +   562, -   564,    +   570, -   564, 
11:32:25.546 ->    +   566, -   566,    +   568, -   698,    +   566, -   568,    +   564, -   566, 
11:32:25.580 ->    +   568, -   566,    +   566, -   700,    +   590, -  1676,    +   568, -  1688, 
11:32:25.580 ->    +   568, -   564,    +   570, -   696,    +   568, -  1688,    +   568, -  1692, 
11:32:25.580 ->    +   564, -   562,    +   570, -   696,    +   568, -   570,    +   590, -   552, 
11:32:25.580 ->    +   570, -   568,    +   566, -   700,    +   562, -   614,    +   520, -   564, 
11:32:25.580 ->    +   568, -  1686,    +   570, -   694,    +   572, -   568,    +   562, -   564, 
11:32:25.614 ->    +   594, -   554,    +   568, -   694,    +   570, -   564,    +   568, -   564, 
11:32:25.614 ->    +   568, -   566,    +   568, -   698,    +   566, -   566,    +   568, -   568, 
11:32:25.614 ->    +   564, -   566,    +   592, -   706,    +   570, -   564,    +   570, -  1686, 
[... cut ...] 88,  568, 564,  570, 696,  568, 1688,  568, 1692,  564, 562,  570, 696,  568, 570,  590, 552,  570, 568,  566, 700,  562, 614,  520, 564,  568, 1686,  570, 694,  572, 568,  562, 564,  594, 554,  568, 694,  570, 564,  568, 564,  568, 566,  568, 698,  566, 566,  568, 568,  564, 566,  592, 706,  570, 564,  570, 1686,  570, 564,  568, 1842,  570, 564,  570, 1690,  568, 1688,  566, 1688,  594};  // HAIER_AC
11:32:25.716 -> uint8_t state[9] = {0xA5, 0xA6, 0x20, 0x00, 0x0C, 0xC0, 0x20, 0x00, 0x57};
```

mazeemsk, 2019-04-16:
Here it is...
Power ON:
```
11:40:39.921 -> Timestamp : 001197.816
11:40:39.921 -> Encoding  : HAIER_AC
11:40:39.921 -> Code      : A59120000CC0200042 (72 bits)
11:40:39.921 -> Mesg Desc.: Command: 1 (On), Mode: 0 (AUTO), Temp: 25C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:00, On Timer: Off, Off Timer: Off
11:40:39.921 -> Library   : v2.5.6
11:40:39.921 -> 
11:40:39.955 -> Raw Timing[149]:
11:40:39.955 ->    +  3052, -  3022,    +  3052, -  4284,    +   598, -  1676,    +   568, -   570, 
11:40:39.955 ->    +   562, -  1688,    +   570, -   698,    +   562, -   570,    +   566, -  1686, 
11:40:39.955 ->    +   570, -   566,    +   566, -  1822,    +   560, -  1696,    +   588, -   554, 
11:40:39.955 ->    +   568, -   566,    +   564, -  1826,    +   564, -   570,    +   564, -   566, 
11:40:39.989 ->    +   568, -   564,    +   568, -  1820,    +   568, -   568,    +   564, -   566, 
11:40:39.989 ->    +   590, -  1728,    +   514, -   700,    +   564, -   568,    +   568, -   564, 
11:40:39.989 ->    +   568, -   566,    +   562, -   700,    +   568, -   568,    +   562, -   566, 
11:40:39.989 ->    +   568, -   570,    +   590, -   686,    +   566, -   568,    +   566, -   568, 
11:40:40.023 ->    +   562, -   574,    +   558, -   704,    +   558, -   570,    +   566, -   566, 
11:40:40.023 ->    +   568, -   566,    +   568, -   694,    +   596, -  1672,    +   570, -  1692, 
11:40:40.023 ->    +   560, -   572,    +   562, -   700,    +   560, -  1694,    +   566, -  1686, 
11:40:40.023 ->    +   572, -   564,    +   568, -   698,    +   566, -   564,    +   596, -   554, 
11:40:40.023 ->    +   562, -   568,    +   568, -   696,    +   568, -   566,    +   566, -   566, 
11:40:40.057 ->    +   566, -  1688,    +   568, -   700,    +   560, -   568,    +   564, -   572, 
11:40:40.057 ->    +   588, -   560,    +   560, -   700,    +   566, -   568,    +   562, -   568, 
11:40:40.057 ->    +   570, -   566,    +   566, -   698,    +   568, -   564,    +   568, -   566, 
11:40:40.057 ->    +   566, -   570,    +   590, -   706,    +   568, -   566,    +   566, -  1690, 
11:40:40.057 ->    +   566, -   568,    +   562, -   724,    +   568, -   564,    +   570, -   568, 
11:40:40.091 ->    +   560, -  1690,    +   566, -   568,    +   594
11:40:40.091 -> 
11:40:40.091 -> uint16_t rawData[149] = {3052, 3022,  3052, 4284,  598, 1676,  568, 570,  562, 1688,  570, 698,  562, 570,  566, 1686,  570, 566,  566, 1822,  560, 1696,  588, 554,  568, 566,  564, 1826,  564, 570,  564, 566,  568, 564,  568, 1820,  568, 568,  564, 566,  590, 1728,  514, 700,  564, 568,  568, 564,  568, 566,  562, 700,  568, 568,  562, 566,  568, 570,  590, 686,  566, 568,  566, 568,  562, 574,  558, 704,  558, 570,  566, 566,  568, 566,  568, 694,  596, 1672,  570, 1692,  560, 572,  562, 700,  560, 1694,  566, 1686,  572, 564,  568, 698,  566, 564,  596, 554,  562, 568,  568, 696,  568, 566,  566, 566,  566, 1688,  568, 700,  560, 568,  564, 572,  588, 560,  560, 700,  566, 568,  562, 568,  570, 566,  566, 698,  568, 564,  568, 566,  566, 570,  590, 706,  568, 566,  566, 1690,  566, 568,  562, 724,  568, 564,  570, 568,  560, 1690,  566, 568,  594};  // HAIER_AC
11:40:40.159 -> uint8_t state[9] = {0xA5, 0x91, 0x20, 0x00, 0x0C, 0xC0, 0x20, 0x00, 0x42};

```
and Temp UP

```
11:32:25.478 -> Timestamp : 000703.365
11:32:25.478 -> Encoding  : HAIER_AC
11:32:25.478 -> Code      : A5A620000CC0200057 (72 bits)
11:32:25.478 -> Mesg Desc.: Command: 6 (Temp Up), Mode: 0 (AUTO), Temp: 26C, Fan: 0 (AUTO), Swing: 0 (Off), Sleep: Off, Health: Off, Current Time: 00:00, On Timer: Off, Off Timer: Off
11:32:25.478 -> Library   : v2.5.6
11:32:25.478 -> 
11:32:25.512 -> Raw Timing[149]:
11:32:25.512 ->    +  3046, -  3032,    +  3070, -  4266,    +   588, -  1676,    +   568, -   570, 
11:32:25.512 ->    +   564, -  1736,    +   520, -   694,    +   568, -   566,    +   566, -  1688, 
11:32:25.512 ->    +   568, -   568,    +   564, -  1820,    +   570, -  1686,    +   596, -   554, 
11:32:25.512 ->    +   566, -  1688,    +   566, -   698,    +   566, -   564,    +   570, -  1686, 
11:32:25.546 ->    +   570, -  1690,    +   566, -   698,    +   568, -   564,    +   570, -   566, 
11:32:25.546 ->    +   592, -  1674,    +   568, -   694,    +   570, -   564,    +   568, -   564, 
11:32:25.546 ->    +   568, -   564,    +   570, -   696,    +   568, -   564,    +   568, -   568, 
11:32:25.546 ->    +   564, -   564,    +   596, -   688,    +   562, -   564,    +   570, -   564, 
11:32:25.546 ->    +   566, -   566,    +   568, -   698,    +   566, -   568,    +   564, -   566, 
11:32:25.580 ->    +   568, -   566,    +   566, -   700,    +   590, -  1676,    +   568, -  1688, 
11:32:25.580 ->    +   568, -   564,    +   570, -   696,    +   568, -  1688,    +   568, -  1692, 
11:32:25.580 ->    +   564, -   562,    +   570, -   696,    +   568, -   570,    +   590, -   552, 
11:32:25.580 ->    +   570, -   568,    +   566, -   700,    +   562, -   614,    +   520, -   564, 
11:32:25.580 ->    +   568, -  1686,    +   570, -   694,    +   572, -   568,    +   562, -   564, 
11:32:25.614 ->    +   594, -   554,    +   568, -   694,    +   570, -   564,    +   568, -   564, 
11:32:25.614 ->    +   568, -   566,    +   568, -   698,    +   566, -   566,    +   568, -   568, 
11:32:25.614 ->    +   564, -   566,    +   592, -   706,    +   570, -   564,    +   570, -  1686, 
[... cut ...] 88,  568, 564,  570, 696,  568, 1688,  568, 1692,  564, 562,  570, 696,  568, 570,  590, 552,  570, 568,  566, 700,  562, 614,  520, 564,  568, 1686,  570, 694,  572, 568,  562, 564,  594, 554,  568, 694,  570, 564,  568, 564,  568, 566,  568, 698,  566, 566,  568, 568,  564, 566,  592, 706,  570, 564,  570, 1686,  570, 564,  568, 1842,  570, 564,  570, 1690,  568, 1688,  566, 1688,  594};  // HAIER_AC
11:32:25.716 -> uint8_t state[9] = {0xA5, 0xA6, 0x20, 0x00, 0x0C, 0xC0, 0x20, 0x00, 0x57};
```

crankyoldgit, 2019-04-16:
Thanks for that @mazeemsk It was really helpful.
I think I've found the problem.

Can you please download and try the code in this branch: https://github.com/markszabo/IRremoteESP8266/tree/Issue668

PR #671 should fix it.

Let me know if it does or doesn't.

crankyoldgit, 2019-04-22:
@mazeemsk Have you had any luck or a chance to test it?

mazeemsk, 2019-04-23:
@crankyoldgit Its working now as expected.....Will you please explain in a simple way what was the issue?? for my understanding if such cases occurs in future.

crankyoldgit, 2019-04-23:
The crux of the issue was there was some missing data in the initial data/state that the class constructed. When I added that we were able to construct me the exact messages you captured. Your captures helped find that missing data.

closed 2019-04-23 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

mazeemsk, 2019-04-30:
Downloaded Thanks David...

crankyoldgit, 2019-06-09:
FYI, The changes mentioned above have been include in v2.6.1 of this library, which has just been released.

commit 61528bf8 (2019-04-16): Add missing info from initial Haier A/C state.
* Add hidden state information to HaierAC's initial state based on real world data.
* Add/adjust unit tests to catch this.

Should fix #668
commit d89d1721 (2019-04-23): [BUG] Fix artificial vs. real state creation on HaierAC. (#671)

* Unit test to help debug Haier Temp Up/Down.
* Add missing info from initial Haier A/C state.
* Add hidden state information to HaierAC's initial state based on real world data.
* Add/adjust unit tests to catch this.

Should fix #668