crankyoldgit/IRremoteESP8266 #669: custom mqtt parameters.

keerthikumaran, 2019-04-10:
Hi,
I am working with the custom parameters for mqtt credentials like the WiFi manger.
I need help in adding the MQTT SERVER, PORT, USER NAME AND PASSWORD as a autoconnect custom FS Parameters.
I have used the same code as given and the libraries too..

crankyoldgit, 2019-04-10:
I'm assuming you are trying to add this to the _IRMQTTServer_ example code. It just uses the standard `WifiManager` library.

Have you tried looking at the example code for WifiManager on how to do this?
e.g.
https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnectWithFSParameters/AutoConnectWithFSParameters.ino

keerthikumaran, 2019-04-15:
I added WIFIMANGER library code accordingly and the json file is not opening to write and save the parameters for the future usage. MQTT connects for the first time with the autoconnectFSparameters, and once when the nodemcu is reset the MQTT parameters are not assigned but the SSID and PASSWORD is read and the mcu gets connected to the wifi.

crankyoldgit, 2019-04-15:
I'm not sure. Perhaps ask on the WifiManager issues page(s)?

As their example code uses SPIFFS, did you add a partition for it in the build/target options in the Arduino IDE?

keerthikumaran, 2019-04-16:
Ok @crankyoldgit will ask WifiManager page.

Ya I added 1M SPIFFS. And I am clearing the memory every time I upload the code.

crankyoldgit, 2019-04-16:
Did you try the example code I liked earlier? i.e. Just to test you've got all the correct build settings.

Sadly, I don't personally have experience with saving stuff to the eeprom/flash on the ESP8266. But, please let me know what you find out, as I'm keen to add that functionality at some point.

keerthikumaran, 2019-04-16:
I have tried the wifimanger code alone and it works fine when it comes to the integration part the json file is not opening to write the parameters in it.
I have attached the screenshot of the serial output screen.

[image]

TEXT ON SERIAL SCREEN:

*WM: Handle root
*WM: WiFi save
*WM: Parameter
*WM: server
*WM: 192.168.43.1
*WM: Parameter
*WM: port
*WM: 1883
*WM: Parameter
*WM: auth
*WM: hello
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Should save config
connected...yeey :)
saving config
opening file to write.
{"mqtt_server":"192.168.43.1","mqtt_port":"1883","auth_token":"hello"}local ip
192.168.43.131
*WM: freeing allocated params!
0000199.959: MDNS responder started
0000199.960: HTTP server started
0000199.964: client mqtt not connected, trying to connect
0000199.964: Attempting MQTT connection to 192.168.43.1:1883topic : hello... 
0000200.146: connected.
0000200.147: Subscription OK to 
0000200.148: successful client mqtt connection
0000222.881: Incoming IR message sent to MQTT: 3,31CE00FF,32
0000224.194: Incoming IR message sent to MQTT: 3,FFFFFFFFFFFFFFFF,0
rl l⸮⸮| ⸮c⸮N⸮$⸮p⸮r⸮⸮⸮{⸮"⸮c⸮⸮Nn⸮ dNN⸮⸮⸮Bp⸮cd rlp⸮N⸮⸮l⸮⸮#n⸮|d⸮⸮p⸮⸮oo⸮l⸮⸮ d ⸮no$ or⸮⸮⸮ob⸮$s⸮⸮o#⸮l⸮2d⸮p⸮⸮⸮{$⸮⸮o⸮
⸮⸮H⸮뫹ѥ⸮⸮⸮FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"192.168.43.1","mqtt_port":"1883","auth_token":"hello"}
parsed json
load json config ... SUCESS
custom_mqtt_server
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: auth
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.43.131
connected...yeey :)
local ip
192.168.43.131
*WM: freeing allocated params!
0000008.496: MDNS responder started
0000008.497: HTTP server started
0000008.498: client mqtt not connected, trying to connect
0000008.498: Attempting MQTT connection to 192.168.43.1:1883topic : hello... 
0000008.648: connected.
0000008.649: Subscription OK to 
0000008.650: successful client mqtt connection

crankyoldgit, 2019-04-16:
Do you have a link to the code you are using?

keerthikumaran, 2019-04-16:
no @crankyoldgit I dont have a link for my code.

crankyoldgit, 2019-04-20:
Without code to test/debug, I can't really help you.

keerthikumaran, 2019-04-22:
this is my code.

[mqtt_irblaster.zip](https://github.com/markszabo/IRremoteESP8266/files/3102252/mqtt_irblaster.zip)

keerthikumaran, 2019-04-23:
I figured out where I was wrong.
Thanks for your help.

PR #680 (2019-04-24): Move MQTT server connection parameters to WifiManager settings.
This should allow a fairly standard build (except for GPIOs) for most people.
- Add mqtt server, port, username & password for mqtt/http and hostname, prefix to WifiManager settings.
  * Saves a `/config.json` file into the SPIFFS to store these over reboots.
- Move hard-coded examples to their own page.
- Lots of hacks to try to reduce heap fragmentation/memory use.
- Move user config settings to `IRMQTTServer.h` file.
- General code cleanup and grouping.
- Bump version. Almost ready for release.
- Bucket loads of debugging added [... cut]
commit 53587851 (2019-04-24): Move MQTT server connection and other parameters to WifiManager settings. (#680)

This should allow a fairly standard build (except for GPIOs) for most people.
- Add mqtt server, port, username & password for mqtt/http and hostname, prefix to WifiManager settings.
  * Saves a `/config.json` file into the SPIFFS to store these over reboots.
- Move hard-coded examples to their own page.
- Lots of hacks to try to reduce heap fragmentation/memory use.
- Move user config settings to `IRMQTTServer.h` file.
- General code cleanup and grouping.
- Bump version. Almost ready for release.
- Bucket loads  [... cut]
closed 2019-04-23 by keerthikumaran