crankyoldgit/IRremoteESP8266 #673: Compile error with compile flag DEBUG at ESP8266 (Arduino)

timostr, 2019-04-16:
### Version/revison of the library used
v2.5.6

### Expected behavior
Enabling -DDEBUG at Arduino should allow compile without errors

### Actual behavior
Compile error: more than one instance of overloaded function "HardwareSerial::println" matches the argument list: -- function "Print::println(char)" -- function "Print::println(unsigned char, int = 10)" -- function "Print::println(int, int = 10)" -- function "Print::println(unsigned int, int = 10)" -- function "Print::println(long, int = 10)" -- function "Print::println(unsigned long, int = 10)" -- function "Print::println(double, int = 2)" -- argument types are: (uint64_t) -- object type is: HardwareSerial
Correction: R132: ir_MWM.cpp  change DPRINTLN(data); -->  DPRINTLN((uint8) data & 0xFF);

_What steps did you do, and what did or didn't actually happen?_
1. Compile with -DDEBUG using esp8266 (Arduino)

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_
Simple compile with -DDEBUG with Arduino (ESP8266)

### I have followed the steps in the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/markszabo/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes

### Has this library/code previously worked as expected for you?
_Yes - not sure must be version from last year

crankyoldgit, 2019-04-16:
Thanks for the report. Sadly it didn't show under my linux-based unit-tests. But yes, when compiled fully under the Arduino compilers it does berk.

#674 should fix it.

closed 2019-04-16 by crankyoldgit

crankyoldgit, 2019-04-30:
FYI, this has been included in the newly released v2.6.0 of the library.

PR #674 (2019-04-16): Fix compile problem when DEBUG is defined.
Fixes #673
commit cb524ad0 (2019-04-16): Fix compile problem when DEBUG is defined. (#674)

Fixes #673